MichaIng/DietPi #6070: Home Assistant upgrade failing for segmentation error and never recovering

fra87, 2023-01-16:
### Creating a bug report/issue

#### Required Information
- DietPi version | `8.12.1`
- Distro version | `bullseye`
- Kernel version | `5.15.80-rockchip64 #22.11.1 SMP PREEMPT Wed Nov 30 11:12:47 UTC 2022 aarch64 GNU/Linux`
- SBC model | `ROCK64 (aarch64)`
- Power supply used | Meanwell LPV-100-24 (24V 4.2A) + 5V 10A step down
- SD card used | Samsung microSDHC UHS-I Card EVO Plus (32GB)

#### Additional Information (if applicable)
- Software title | Home Assistant
- Was the software title installed freshly or updated/migrated? Trying to be upgraded
- Can this issue be replicated on a fresh installation of DietPi? I don't think so

#### Steps to reproduce
1. Have an old Home Assistant installation (I'm not sure how old, I suspect it needs to have at most Python 3.9)
2. Try to upgrade to v2013.1 version (I did it towards v2013.1.4

#### Expected behaviour
- The HA is upgraded

#### Actual behaviour
- Upgrade failed with a segmentation error
- Successive upgrades fail (but not with the segmentation error), and HA refuses to launch

#### Workaround
- Reinstall home assistant with the command `dietpi-software reinstall 157`; all the HA data is kept

#### Extra details
The complete error message I received when upgrading the first time was

```
Installing collected packages: homeassistant
  Attempting uninstall: homeassistant
    Found existing installation: homeassistant 2023.1.1
    Uninstalling homeassistant-2023.1.1:
      Successfully uninstalled homeassistant-2023.1.1
/home/homeassistant/.pyenv/pyenv.d/exec/pip-rehash/pip: riga 20: 406230 Errore di segmentazione "$PYENV_COMMAND_PATH" "$@"
```

When I tried to upgrade it again (I tried two more times) the error I received in the installation was

`WARNING: Ignoring invalid distribution -omeassistant (/home/homeassistant/.pyenv/versions/3.9.15/lib/python3.9/site-packages)`

When I tried to start HA, systemctl recorded this as failure reason:

```
gen 12 15:27:10 rock64 systemd[1]: Started Home Assistant (DietPi).
gen 12 15:27:11 rock64 Home Assistant[406920]: /home/homeassistant/homeassistant-start.sh: 3: exec: hass: not found
gen 12 15:27:11 rock64 systemd[1]: home-assistant.service: Main process exited, code=exited, status=127/n/a
gen 12 15:27:11 rock64 systemd[1]: home-assistant.service: Failed with result 'exit-code'.
gen 12 15:27:11 rock64 systemd[1]: home-assistant.service: Consumed 1.168s CPU time.
```

I then reinstalled the HA software (it took roughly one hour), but now I have it with Python 3.10. I just had to manually reinstall numpy, since it was not present anymore

I don't think this is a real issue to be solved, but I think it was better to document it.
I saw also that some people were having this issue on the forum as well ([here](https://dietpi.com/forum/t/home-assistant-2022-12-and-python-3-10-how-to-update/15202/9))

MichaIng, 2023-01-16:
Did you upgrade to DietPi v8.12 before or after you faced this issue? Because in that version we raised the implied Python version to v3.10, required by Home Assistant since first January release. So you shouldn't have faced the issue when updating DietPi first and reinstalling HA afterwards.

fra87, 2023-01-16:
I upgraded to 8.12 before, then started the update of HA without reinstalling it.

This means that after `dietpi-update` I ran `/home/homeassistant/homeassistant-update.sh`, but not `dietpi-software reinstall 157`, and it failed as written above.

In order to fix it I had to reinstall it, thus updating also Python

Joulinar, 2023-01-17:
I guess a reinstall was needed first due to new requirement of Python v3.10.

MichaIng, 2023-01-17:
Ah yes this is expected then. We could add a hint to our HA docs update tab, encourage users to read HA changelogs before updating, and if a newer Python version is required, do a `reinstall` to upgrade Python itself as well.

MichaIng, 2024-05-07:
Done with: https://github.com/MichaIng/DietPi-Docs/pull/995

closed 2024-05-07 by MichaIng