MichaIng/DietPi #6071: Dietpi goes into semi-online mostly-offline state please help

JustRedTTG, 2023-01-18:
### Creating a bug report/issue

#### Required Information
- DietPi version, up to date
- Distro version, raspberry
- Kernel version | `uname -a`
- SBC model rpi3 (2GB I think) 
- Power supply used, the connector is broken, using the power pins directly with 5V 3.5A according to power source
- USB used: 4GB no idea what type

#### Additional Information (if applicable)
- Software title, THE WHOLE SYSTEM IDK
- it's probably next cloud since it's always happened with next cloud from the beginning
- Can this issue be replicated on a fresh installation of DietPi? Probably? Idk? 

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. I suppose stress the system? 
2. Try to connect to ssh or something

#### Expected behaviour

- it should connect! 

#### Actual behaviour

- loads indefinitely

#### Extra details

- it's still on the network, just not actually responding to anything and it doesn't seem to be working at all

Forced solution is to unplug and plug back in (not suitable when away and using cloud flare tunnel to connect to web interface) 

Has a 1T hard disk connected directly (no external power to the hard disk) for next cloud and site and dietpi user storage, since the boot usb doesn't have much space

Joulinar, 2023-01-18:
> Has a 1T hard disk connected directly (no external power to the hard disk)

You should think of using an external PSU for your HDD. Especially RPI device have issues providing enough power for connected USB disk. Therefore, we highly recommend using an external PSU.

MichaIng, 2023-01-18:
More something like [this](https://duckduckgo.com/?q=USB+HDD+docking+station&iax=images&ia=images) with an own PSU 😄. If you start soldering, this is your own responsibility 😉.

JustRedTTG, 2023-01-18:
> More something like [this](https://duckduckgo.com/?q=USB+HDD+docking+station&iax=images&ia=images) with an own PSU 😄. If you start soldering, this is your own responsibility 😉.

Actually it's a usb hard disk, one of those consumer ones

MichaIng, 2023-01-18:
You should be able to take out the actual HDD from its case, or isn't it possible to open it without breaking it? Another solution is a self-powered USB hub.

What does the kernel say?
```sh
dmesg -l 0,1,2,3
```

JustRedTTG, 2023-01-18:
> You should be able to take out the actual HDD from its case, or isn't it possible to open it without breaking it? Another solution is a self-powered USB hub.
> 
> What does the kernel say?
> ```sh
> dmesg -l 0,1,2,3
> ```

[image]

Joulinar, 2023-01-18:
as you can see, some undervoltage message. It's highly recommended to get a dedicated PSU/ powered USB hub for your HDD.

JustRedTTG, 2023-01-18:
Fixed low power using diagram mentioned above but not with usb hub rather just the usb cable going to the HDD
[image]
I will report if issues still occur!

JustRedTTG, 2023-01-19:
It still does it, happened twice, I saw somewhere that it might be a static IP issue but the pi is still on the network you just can't do anything with it, no ssh and the cloudflare tunnel also breaks

MichaIng, 2023-01-19:
Since you mention the issue happens when Nextcloud becomes active, which uses that USB drive, I'm still pretty sure that it is that one which causes the issue. Especially since it is a HDD which requires even more power compared to an SSD. And we see a lot of such cases with 2.5" SSDs already.

JustRedTTG, 2023-01-19:
I fixed the power issue with external power, so no longer an issue, and also I mentioned that this has been an issue from the start and the only thing I had at the start was nextcloud

JustRedTTG, 2023-01-19:
tell me if I can grab some logs with a command, it just happened yet again

Joulinar, 2023-01-19:
You would need to make logs persistent to be able to check them after reboot.

[code]
dietpi-software uninstall 103 # uninstalls DIetPi-RAMlog
mkdir /var/log/journal # triggers systemd-journald logs to disk
reboot # required to finalise the RAMlog uninstall
[/code]

Then you can check system logs via:
[code]journalctl[/code]

which will then show as well logs from previous boot sessions. To limit the size, you can additionally e.g. apply the following:
[code]
mkdir -p /etc/systemd/journald.conf.d
cat << '_EOF_' > /etc/systemd/journald.conf.d/99-custom.conf
[Journal]
SystemMaxFiles=2
MaxFileSec=7day
_EOF_
[/code]

This will limit logs to 14 days split across two journal files, so that with rotation you will always have between 7 and 14 days of logs available.

JustRedTTG, 2023-01-20:
New revelation, but I think the server keeps having this issue exactly at 9:06 cause I checked my messages with my mom which is sometimes still home around that time and both times it was exactly 9:06 I asked her if she's home so she can restart the power to the server. Not sure about it honestly, since it happens when I'm home too but I guess I'll check when it happens again when I get home and just see what the intervals and stuff is. Anyone got any idea what it might be from this, I'll also check the logs when I get home

JustRedTTG, 2023-01-20:
```Jan 20 06:26:59 redspi.local gunicorn[39299]: ValueError: invalid literal for int() with base 10: 'Баскетбол' <-- something I was doing
\/ \/ \/ Below the server has done the thingy and is no longer acessable \/ \/ \/
Jan 20 06:30:01 redspi.local CRON[74135]: pam_unix(cron:session): session opened for user www-data(uid=33) by (uid=0)
Jan 20 06:30:01 redspi.local CRON[74136]: (www-data) CMD (php /var/www/nextcloud/cron.php)
Jan 20 06:30:07 redspi.local CRON[74135]: pam_unix(cron:session): session closed for user www-data
Jan 20 06:35:01 redspi.local CRON[74674]: pam_unix(cron:session): session opened for user www-data(uid=33) by (uid=0)
Jan 20 06:35:01 redspi.local CRON[74675]: (www-data) CMD (php /var/www/nextcloud/cron.php)
Jan 20 06:35:07 redspi.local CRON[74674]: pam_unix(cron:session): session closed for user www-data
Jan 20 06:39:01 redspi.local CRON[75128]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
Jan 20 06:39:01 redspi.local CRON[75129]: (root) CMD (  [ -x /usr/lib/php/sessionclean ] && if [ ! -d /run/systemd/system ]; then /usr/lib/php/sessionclean; fi)
Jan 20 06:39:01 redspi.local CRON[75128]: pam_unix(cron:session): session closed for user root
Jan 20 06:39:02 redspi.local systemd[1]: Starting Clean php session files...
Jan 20 06:39:03 redspi.local systemd[1]: phpsessionclean.service: Succeeded.
Jan 20 06:39:03 redspi.local systemd[1]: Finished Clean php session files.
Jan 20 06:40:01 redspi.local CRON[75267]: pam_unix(cron:session): session opened for user www-data(uid=33) by (uid=0)
Jan 20 06:40:01 redspi.local CRON[75268]: (www-data) CMD (php /var/www/nextcloud/cron.php)
Jan 20 06:40:07 redspi.local CRON[75267]: pam_unix(cron:session): session closed for user www-data
Jan 20 06:45:01 redspi.local CRON[75825]: pam_unix(cron:session): session opened for user www-data(uid=33) by (uid=0)
Jan 20 06:45:01 redspi.local CRON[75826]: (www-data) CMD (php /var/www/nextcloud/cron.php)
Jan 20 06:45:07 redspi.local CRON[75825]: pam_unix(cron:session): session closed for user www-data
Jan 20 06:50:01 redspi.local CRON[76364]: pam_unix(cron:session): session opened for user www-data(uid=33) by (uid=0)
Jan 20 06:50:01 redspi.local CRON[76365]: (www-data) CMD (php /var/www/nextcloud/cron.php)
Jan 20 06:50:07 redspi.local CRON[76364]: pam_unix(cron:session): session closed for user www-data
Jan 20 06:55:01 redspi.local CRON[76910]: pam_unix(cron:session): session opened for user www-data(uid=33) by (uid=0)
Jan 20 06:55:01 redspi.local CRON[76912]: (www-data) CMD (php /var/www/nextcloud/cron.php)
Jan 20 06:55:06 redspi.local CRON[76910]: pam_unix(cron:session): session closed for user www-data
Jan 20 07:00:01 redspi.local CRON[77453]: pam_unix(cron:session): session opened for user www-data(uid=33) by (uid=0)
Jan 20 07:00:01 redspi.local CRON[77454]: (www-data) CMD (php /var/www/nextcloud/cron.php)
-- Boot 728fc35e085e4c7b8fd2bc51d0c153e2 --
/\ /\ /\ Booted like 4 hours later /\ /\ /\
```

JustRedTTG, 2023-01-20:
not too sure if that's what caused it but the rest of the log seems normal, I found this error from kernel after boot:
```
Jan 20 11:31:49 redspi.local kernel: sd 1:0:0:0: [sdb] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=DRIVER_OK cmd_age=0s
Jan 20 11:31:49 redspi.local kernel: sd 1:0:0:0: [sdb] tag#0 Sense Key : 0x2 [current]
Jan 20 11:31:49 redspi.local kernel: sd 1:0:0:0: [sdb] tag#0 ASC=0x3a ASCQ=0x0
Jan 20 11:31:49 redspi.local kernel: sd 1:0:0:0: [sdb] tag#0 CDB: opcode=0x2a 2a 00 73 cf 69 e0 00 00 38 00
Jan 20 11:31:49 redspi.local kernel: blk_update_request: I/O error, dev sdb, sector 1942972896 op 0x1:(WRITE) flags 0x800 phys_seg 7 prio class 0
Jan 20 11:31:49 redspi.local kernel: Aborting journal on device sdb2-8.
Jan 20 11:31:49 redspi.local kernel: sd 1:0:0:0: [sdb] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=DRIVER_OK cmd_age=0s
Jan 20 11:31:49 redspi.local kernel: sd 1:0:0:0: [sdb] tag#0 Sense Key : 0x2 [current]
Jan 20 11:31:49 redspi.local kernel: sd 1:0:0:0: [sdb] tag#0 ASC=0x3a ASCQ=0x0
Jan 20 11:31:49 redspi.local kernel: sd 1:0:0:0: [sdb] tag#0 CDB: opcode=0x2a 2a 00 73 cf 68 00 00 00 08 00
Jan 20 11:31:49 redspi.local kernel: blk_update_request: I/O error, dev sdb, sector 1942972416 op 0x1:(WRITE) flags 0x800 phys_seg 1 prio class 0
Jan 20 11:31:49 redspi.local kernel: Buffer I/O error on dev sdb2, logical block 1310720, lost sync page write
Jan 20 11:31:49 redspi.local kernel: JBD2: Error -5 detected when updating journal superblock for sdb2-8.```
full log: https://pastebin.com/LcgYxuwN

MichaIng, 2023-01-20:
Logs of the two boot sessions can mix, because the system time is up to 1 hour too early after a crash, until network time sync has completed. You can use the following to show logs for a specific boot ID (the one shown on your last line):
```sh
journalctl -b 728fc35e085e4c7b8fd2bc51d0c153e2
```
Find and use the boot ID of the previous session where the crash happened.

MichaIng, 2023-01-20:
> but the rest of the log seems normal

The end of your log shows a complete database failure loop on `/dev/sdb2`, starting with a kernel I/O error:
```sh
Jan 20 11:30:34 redspi.local kernel: usb 1-1.4: reset high-speed USB device number 5 using dwc_otg
Jan 20 11:31:37 redspi.local kernel: sd 1:0:0:0: [sdb] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=DRIVER_OK cmd_age=94s
Jan 20 11:31:37 redspi.local kernel: sd 1:0:0:0: [sdb] tag#0 Sense Key : 0x2 [current]
Jan 20 11:31:37 redspi.local kernel: sd 1:0:0:0: [sdb] tag#0 ASC=0x3a ASCQ=0x0
Jan 20 11:31:37 redspi.local kernel: sd 1:0:0:0: [sdb] tag#0 CDB: opcode=0x2a 2a 00 73 af 42 78 00 00 08 00
Jan 20 11:31:37 redspi.local kernel: blk_update_request: I/O error, dev sdb, sector 1940865656 op 0x1:(WRITE) flags 0x800 phys_seg 1 prio class 0
Jan 20 11:31:37 redspi.local kernel: EXT4-fs warning (device sdb2): ext4_end_bio:348: I/O error 10 writing to inode 27992 starting block 242608208)
Jan 20 11:31:37 redspi.local kernel: Buffer I/O error on device sdb2, logical block 1047375
```
Looks like there is still issues with this drive, probably bad physical connection or something like that. The first error is about raw disk I/O, not the filesystem, so an issue on hardware level, not on software level, I'd say.

MichaIng, 2023-02-12:
Let us know if the issue persists, referring to above comment.

closed 2023-03-18 by MichaIng