MichaIng/DietPi #6072: Can't update

anuehnen, 2023-01-18:
#### Details:
- Date           | Wed Jan 18 09:19:20 CET 2023
- Bug report     | 3e919177-79b4-49a1-9e68-f38f6fa38ce5
- DietPi version | v8.12.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Meveric
- Hardware       | Odroid N2 (aarch64) (ID=15)
- Kernel version | `Linux media2 4.9.277-arm64 #1 SMP PREEMPT Fri Mar 11 00:57:59 CET 2022 aarch64 GNU/Linux`
- Distro         | buster (ID=5)
- Command        | `apt-get -y upgrade`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
The following packages have been kept back:
  linux-image-arm64-odroid-n2
Attempt to access disallowed ▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒).▒▒▒# s▒ߙ▒r▒t▒t▒▒ng▒▒f▒r▒▒▒ t▒▒▒▒f▒▒▒
                                                                                                                                                            یta▒▒a▒f▒▒▒sߋ▒e▒na▒▒sߐ▒ ▒w▒▒d▒tab▒s▒s▒' in a restricted hash at /usr/share/perl5/Debconf/Config.pm line 43, <DEBCONF_CONFIG> chunk 1.
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up snmpd (5.7.3+dfsg-5+deb10u4) ...
Attempt to access disallowed ▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒).▒▒▒# s▒ߙ▒r▒t▒t▒▒ng▒▒f▒r▒▒▒ t▒▒▒▒f▒▒▒
                                                                                                                                                            یta▒▒a▒f▒▒▒sߋ▒e▒na▒▒sߐ▒ ▒w▒▒d▒tab▒s▒s▒' in a restricted hash at /usr/share/perl5/Debconf/Config.pm line 43, <DEBCONF_CONFIG> chunk 1.
dpkg: error processing package snmpd (--configure):
 installed snmpd package post-installation script subprocess returned error exit status 255
Errors were encountered while processing:
 snmpd
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

MichaIng, 2023-01-18:
`snmpd` is causing issues. I'm not familiar with this package, you did install it manually, or it was pulled as dependency by some software you installed. Is this strange looking grey part some progress bar or was there some additional output? Probably the service log contains further info:
```sh
journalctl -u snmpd
```

MichaIng, 2023-02-12:
Closing this issue due to outstanding reply. Feel free to reopen if the problem persists.

closed 2023-02-12 by MichaIng