MichaIng/DietPi #6073: [Help request] How to route most traffic trough ETH0 even when WG0 connection to VPN provider is active?

GvY85, 2023-01-18:
Hi,

I am looking for a solution for the following problem.
Currently I have 2 devices running DietPi
1. Device with Pihole, PiVPN for my mobile devices, Unbound, Samba and Rsync. This is my DNS server/Adblockeer/Backup device
2. Device with qBitttorrent, Wireguard client for VPN, SimpleDLNA. This I use to download stuff via a VPN and serve it to the home network over DLNA.

Now what I want is to migrate the 2nd device over to the 1st device since it is powerfull enough and it will save me 20W 24/7.
However, since I then need a VPN that would (I assume) mean that all traffic from this combined device and this also from my mobile devices and from Unbound will be routed trough the Wireguard VPN Client correct?

How can I make it so that all traffic goes trough ETH0 and only qBittorrent uses WG0?\
In qBittorrent I ca just bind it to WG0 so thats good but how do I make sure all other traffic ignores the WG0 interface and keeps using ETH0 after connecting to my VPN provider?
Also, will PiVPN still work as a server to my mobile devices with a client running on the same device as well?  

I read something about Namespaces and/or Virtual Networkstack or something?

Thanks a lot!

MichaIng, 2023-01-18:
Indeed, you need to implement an `ip rule` with higher priority than WireGuards rules to override the gateway/route for packets originating from the PiVPN network. Do you use WireGuard as well with PiVPN?

GvY85, 2023-01-18:
Yes, I use Wireguard as well with PiVPN. Basically what I want is for all processes to ignore the fact that there is a WG0 connection and keep routing everything trough ETH0 as they did before making the connection to the VPN provider.

So it seems I need to make the OS 'ignore' the fact that there now is a VPN connection and/or set ETH0 to have a higher priority or some sort it seems?

But my Wireguard client config has no post-up/down rules or anything so what happens when you connect as a Wireguard client that force all its traffic now from ETH0 to WG0 and how can I prevent that from happening?
Since qBittorrent has an internal option to bind to a specific interface that seems to be the best option? Ignore the fact that there is a WG0 connection and just use the internal bind option of qBittorrent to bind it to WG0 while all other traffic keeps using ETH0.

Or am I making a mistake here?

MichaIng, 2023-01-18:
If you want only qBittorrent to use the VPN, then indeed this is best. In your `wg0.conf`, add e.g.
```bash
Table = 42
```
to the `[Interface]` block. With a custom routing table applied, `wg-quick` does not add any rules to route traffic through it, but you can do this manually then via `PostUp`/`PostDown`:
```
PostUp = ip -4 rule add uidrange 999-999 table 42
PostUp = ip -4 rule add uidrange 999-999 table 42
PostUp = ip -4 rule add uidrange 999-999 table 42

GvY85, 2023-01-18:
Thanks! And what about the `AllowedIPs = 0.0.0.0/0, ::0/0` that is in the `[Peer]` section?

MichaIng, 2023-01-18:
`AllowedIPs` defines the target IP range/mask for connections/packets which should be routed through the tunnel. WireGuard adds respective routes to the table (`42`). So if anything is forced to use this table, it will be routed through the VPN.

GvY85, 2023-01-18:
> `AllowedIPs` defines the target IP range/mask for connections/packets which should be routed through the tunnel. WireGuard adds respective routes to the table (`42`). So if anything is forced to use this table, it will be routed through the VPN.

Ok, so still all traffic get routed trough WG0 but **only** for that specific `uid`?

MichaIng, 2023-01-18:
> I knew 42 was the answer to all my troubles ;)

😄 any table name works which is not already in use:
```sh
ip r l table all | grep -Po 'table \K[^ ]+' | sort -u
```

> Ok, so still all traffic gets routed trough WG0 but only for that specific `uid`?

Yes. Ah, problematic is that it might apply as well for connections from LAN and (Pi)VPN, so you might not be able to visit the qBittorrent web interface anymore. To solve this:
```sh
ip -4 rule add uidrange 999-999 table main suppress_prefixlength 0
ip -6 rule add uidrange 999-999 table main suppress_prefixlength 0
```
When calling these after `ip -[46] rule add uidrange 999-999 table 42`, those rules are added with higher priority (probably better to define it explicitly). They again force the user to use the main table IF the prefix length of the matching route is not `0`, i.e. all routes but the default (`0.0.0.0/0` resp. `::/0`) route. That way network internal routes from main table are still effective, so if a packet goes to e.g. `192.168.1.12`, the LAN route from main table to `192.168.1.0/24` still applies, allowing answers to LAN. Same for communication within the PiVPN network.

GvY85, 2023-01-18:
Thanks, I will try this out somewhere in the next few days.

MichaIng, 2023-01-18:
This is btw the same logic WireGuard uses, if you do not change `Table` and let it setting up the rules:
```console
# ip rule
0:	from all lookup local
32764:	from all lookup main suppress_prefixlength 0
32765:	not from all fwmark 0xca6c lookup 51820
32766:	from all lookup main
32767:	from all lookup default
```
It marks packages, which are for communicating with the VPN server itself, with this `0xca6c` (actually hexadecomal for `51820` 😄), and forces all traffic which does not match this mark and being from `root` user to lookup via table `51820` (used by default by WireGuard). Then it adds the higher priority rule to force lookups from main table for all routes but the default routes (to remote/WAN hosts).

GvY85, 2023-01-18:
So basically it is going to look like this:

in **`wg0.conf`**
add `Table = 42` to the  `[Interface]` part

and add 
```
PostUp = ip -4 rule add uidrange 999-999 table 42 
PostUp = ip -6 rule add uidrange 999-999 table 42
PostUp = ip -4 rule add uidrange 999-999 table main suppress_prefixlength 0
PostUp = ip -6 rule add uidrange 999-999 table main suppress_prefixlength 0
PostDown = ip -4 rule del uidrange 999-999 table 42
PostDown = ip -6 rule del uidrange 999-999 table 42
PostDown = ip -4 rule add uidrange 999-999 table main suppress_prefixlength 0
PostDown = ip -6 rule add uidrange 999-999 table main suppress_prefixlength 0
```

with `999 `being the UID of the qBittorrent process and `42 `just being a free table.

Right?

MichaIng, 2023-01-18:
Yes that should be it, if I haven't forgotten anything 😉.

GvY85, 2023-01-21:
Hi,

Tested it on the current device (so without PiVPN etc) and that works a charm!
`dig ipleak.net` and `dietpi-banner` return my ISP IP address and a test from `https://bash.ws/torrent-leak-test` returns the VPN's IP.

Next step is to integrate everything on the single device with also PiVPN and see if that interferes.

GvY85, 2023-01-22:
Tried to do the same on the 1st device with PiVPN and Pihole + Unbound and as soon as I connect to the WG VPN Server Pihole + Unbound stops working so I cannot resolve any DNS  anymore. Not on other devices connected and not on my local device.

A `wg-quick down 'connection to VPN provider'` restores everything again back to normal so I guess something is missing in the `PostUp/PostDown`?

What I did:
Copied the (on that system working) WG VPN provider config from device 2 to device 1 and ran `wg-quick up 'vpn provider profile`'.In order for that to work I had to `enable IPV6` in `dietpi-config`.
I then immediately loose DNS resolving until I close the connection with `wg-quick down 'vpn provider profile`'.

My WG config:
`[Interface]
PrivateKey = (redacted)
Address = 10.65.51.10/32,fc00:bbbb:bbbb:bb01::2:3309/128
DNS = 192.168.178.3
MTU = 1372
Table = 42
PostUp = ip -4 rule add uidrange 998-998 table 42
PostUp = ip -6 rule add uidrange 998-998 table 42
PostUp = ip -4 rule add uidrange 998-998 table main suppress_prefixlength 0
PostUp = ip -6 rule add uidrange 998-998 table main suppress_prefixlength 0
PostDown = ip -4 rule del uidrange 998-998 table 42
PostDown = ip -6 rule del uidrange 998-998 table 42
PostDown = ip -4 rule add uidrange 998-998 table main suppress_prefixlength 0
PostDown = ip -6 rule add uidrange 998-998 table main suppress_prefixlength 0

[Peer]
PublicKey = (redacted)
AllowedIPs = 0.0.0.0/0,::0/0
Endpoint = (redacted)
`

The `DNS = 192.168.178.3` is now its own ip. Tried changing it to `127.0.0.1` but no avail.

GvY85, 2023-01-22:
Hmm, I found that if I set the `DNS = 192.168.178.3`  to a public DNS server it works with WG enabled. But that would mean everything will bypass Pihole + Unbound right?

For clarification:
I have a FRITZ!Box and that serves `192.168.178.3` (the Pihole + Unbound + PiVPN device) as DNS server to all devices local and remove via PIVPN.
On the `192.168.178.3` device I have set `127.0.0.1` as DNS.

MichaIng, 2023-01-23:
I fail to understand why `DNS = 192.168.178.3` doesn't work, since a `192.168.178.0/24` route should exist (prefix length = 24) to overrule the VPN-enforcing rule.

`127.*` requests always overrule everything, so it makes sense that this works on the Pi-hole host.

> On the `192.168.178.3` device I have set `127.0.0.1` as DNS.
>
> EDIT: found the solution. Have to set the DNS value in the Wireguard config to `DNS = 127.0.0.1`

Just to understand, as Pi-hole and qBittorrent now runs on the same machine, both statements seem to be the same? And `DNS = 127.0.0.1` on another client should not work anyway. And for PiVPN clients to work, you need to allow all origins in Pi-hole, which then should apply to its host as well.

GvY85, 2023-01-29:
Hmm, seems it does not work entirely.

The DNS = 127.0.0.1 was set in the wireguard config becuase that is the DNS used in dietpi-config, so to prevent it frombeing changed.
But I agree, the 192.168.178.3 should also work because that is its IP and is communicated to all devices in the home network to use as DNS server.

But now it seems that when I use a torrent with a 'new' (not previously resolved i guess) tracker qBittorrent cannot connect to the tracker so I get no data.

MichaIng, 2023-01-29:
> The `DNS = 127.0.0.1` was set in the wireguard config because that is the DNS used in dietpi-config, so to prevent it from being changed.

Again, just to be sure, since there are WireGuard configs on server as well as on client, and you have two WireGuard instances on your server, a server config for your client's to connect and a client config to connect to your VPN provider. In which config exactly did you set this?

If on the server's client config, it should btw change `/etc/resolv.conf` only and has no effect if you use a static IP with this DNS set. I however advice to use a public DNS provider at your server, otherwise you may break your own DNS resolution when doing maintenance. On the server's server config, it doesn't make sense at all to set this.

If unsure, I'd just remove the `DNS` config everywhere. It usually doesn't make sense. If any VPN client is connected a way that all traffic is forced through the VPN, the same is true for requests to public VPN providers. So a DNS leak is ruled out as long as you use `AllowedIPs = 0.0.0.0/0, ::/0` and do not setup the routing table and rules manually to explicitly exclude or include only certain traffic to be tunnelled (like we do with qBittorrent).

> But now it seems that when I use a torrent with a 'new' (not previously resolved i guess) tracker qBittorrent cannot connect to the tracker so I get no data.

Note that if your server is connected to a VPN provider, expectedly through the VPN your server cannot be reached. VPN provides usually do not provide port forwarding, at least not without additional costs. qBittorrent can connect to other peers, to the tracker etc, but other peers cannot connect your qBittorrent. So any connection must be established from your end. But in your case it sounds like your qBittorrent cannot connect?

However, please check my above questions about what is now running on which machine, as I'm not sure whether I understand the setup entirely.

GvY85, 2023-01-29:
Everything is now on the same device:

Pihole: set to only listen to one hop away
Unbound
PiVPN that uses `wg0` config used to connect with my mobile devices to the homenetwork
A Mullvad VPN client config called `nl4-wireguard` that I have edited with the above settings and is 'bound' to qBittorrent.

`nl4-wireguard` looks like:
```
[Interface]
PrivateKey = (private key)
Address = 10.65.51.10/32,fc00:bbbb:bbbb:bb01::2:3309/128
DNS = 127.0.0.1
Table = 42
PostUp = ip -4 rule add uidrange 997-997 table 42
PostUp = ip -6 rule add uidrange 997-997 table 42
PostUp = ip -4 rule add uidrange 997-997 table main suppress_prefixlength 0
PostUp = ip -6 rule add uidrange 997-997 table main suppress_prefixlength 0
PostDown = ip -4 rule del uidrange 997-997 table 42
PostDown = ip -6 rule del uidrange 997-997 table 42
PostDown = ip -4 rule add uidrange 997-997 table main suppress_prefixlength 0
PostDown = ip -6 rule add uidrange 997-997 table main suppress_prefixlength 0

[Peer]
PublicKey = (public key)
AllowedIPs = 0.0.0.0/0,::0/0
Endpoint = 193.32.249.69:54000
```
`997 ` is the `uid `of qBittorrent. The `DNS=` was initially set to the Mullvad DNS server but that becomes than the DNS server for the whole system so I changed it to my local adres. I am aware that if Unbound goes down I loose all.
In the Unbound configuration I added the Mullvad ip as` access-control: 10.65.51.10/32 allow` and  `private-address: 10.65.51.10/32`.
I also opened up ad port in Mullvad and set that in qBittorrent.
I do get some traffic in qBittorrent but the problem is that it cannot connect to any  trackers. However I do see them being resolved in Pihole.

MichaIng, 2023-01-29:
I suggest the following:
- Remove `DNS =` from `nl4-wireguard.conf`, as it is redundant if you apply a static IP anyway.
- Remove `access-control: 10.65.51.10/32 allow` from the Unbound config, which shouldn't be needed as `127.0.0.1` requests bypass the VPN anyway.
- Remove `private-address: 10.65.51.10/32` as this is implied by `private-address: 10.0.0.0/8`, which you all do not want to have as valid response from a public DNS server.

> As soon as I set qBittottent to `listen to all interaces` all trackers connect, I get much more peers and much better downloads.

How was it set before? And is this a setting for the web interface or for incoming BitTorrent requests, or for both? For the web interface, you want it to listen on `eth0` or LAN IP ranges, for incoming BitTorrent requests, if this works with Mullvad (port forwarding supported), you want it to listen on `nl4-wireguard`. If `listen to all interaces` works, since any answers are routed through the VPN anyway, then it is not an issue with the VPN config but with this qBittorrent config.

> edit: hmm, as I try an official Ubuntu torrent that has a HTTPS based tracker, that works. But the UDP and HTTP trackers do not work....

Probably a limitation of Mullvads VPN or port forwarding, to allow HTTPS on TCP only?

GvY85, 2023-01-30:
Thanks for your replies, even if it is unrelated to DietPi, much appreciated!

I will remove the 3 things you mention, those indeed do not make sense.

I always used the qBittorrent setting to bind to the nl4-wireguard interface to make sure all torrent traffic goes via the VPN.
With this setting the web interface keeps working so I guess qBittorrent makes sure its only applies to torrent traffic

But if I understand correctly you propose to set qBittorrrent to listen to all interfaces so that the trackers work (for a reason we do not understand) but because we use the Table 42 thing in the nl4-wireguard config all qBittorrent traffic will go over the VPN anyway correct?

MichaIng, 2023-01-30:
> Thanks for your replies, even if it is unrelated to DietPi, much appreciated!

Also good for me/us, as those are valuable finds here as reference for others and when someone finds time to implement a proper split-tunneling feature with a new script or as `dietpi-software` option 😃.

> With this setting the web interface keeps working so I guess qBittorrent makes sure its only applies to torrent traffic

Jep seems so.

> But if I understand correctly you propose to set qBittorrrent to `listen to all interfaces` so that the trackers work (for a reason we do not understand) but because we use the `Table 42` thing in the `nl4-wireguard` config all qBittorrent traffic will go over the VPN anyway correct?

Exactly. To be more precise: With the routing table and rules, qBittorrent outgoing requests/packets and answers are tunnelled anyway, if not from LAN. The listen/bind feature actually should only affect incoming requests/packets. This makes it stranger, since if all outgoing packets are tunnelled, there should never be an answer coming from anywhere else than the tunnel.

With `bind to the nl4-wireguard interface`, can you check:
```sh
ss -tulpn || grep qbit
```
And what does actually happen if the VPN is not up (yet) and qBittorrent attempts to start. Does it fail?
```sh
journactl -u qbittorrent
ss -tulpn || grep qbit
```

When you stop the VPN, without disabling the rules in `PostDown`, how does it not work? I'm actually not sure what the effect of a rule is if the related table does not exist. Would be great if it then would not use any route but fail to send requests, but obviously that is not the case?

What you want is a killswitch, i.e. firewall rules which deny VPN bypasses. But doing this for a particular user/server only might be tricky, at least I'd need to lookup how to do that without nasty connection/packet marks (possible but complicated). Another idea would be to add `systemctl start/stop qbittorrent` to the WireGuard `PostUp`/`PreDown` settings

Or on systemd level via `wg-quick@nl4-wireguard.service` `ExecStartPost` and `ExecStopPost` or `BindsTo` + `After` and/or `PartOf`. This way qBittorrent is stopped once the VPN is stopped or enters failure state, and optionally even started once the VPN is started: https://www.freedesktop.org/software/systemd/man/systemd.unit.html#BindsTo=

GvY85, 2023-01-31:
This is how the bind interface looks with the VPN up:
[image]

With the VPN down it looks like this and indeed, no traffic goes trough, no torrent work (good).
[image]

With the vpn up and qBittorrent bound to it but no torrents active I get:
`ss -tulpn | grep qbit`
`udp   UNCONN 0      0                        10.65.51.10%nl4-wireguard:56886      0.0.0.0:*    users:(("qbittorrent-nox",pid=6767,fd=16))`
`udp   UNCONN 0      0                                          0.0.0.0:6771       0.0.0.0:*    users:(("qbittorrent-nox",pid=6767,fd=19))`
`udp   UNCONN 0      0      [fc00:bbbb:bbbb:bb01::2:3309]%nl4-wireguard:56886         [::]:*    users:(("qbittorrent-nox",pid=6767,fd=18))`
`udp   UNCONN 0      0                                                *:6771             *:*    users:(("qbittorrent-nox",pid=6767,fd=20))`
`tcp   LISTEN 0      30                       10.65.51.10%nl4-wireguard:56886      0.0.0.0:*    users:(("qbittorrent-nox",pid=6767,fd=14))`
`tcp   LISTEN 0      30     [fc00:bbbb:bbbb:bb01::2:3309]%nl4-wireguard:56886         [::]:*    users:(("qbittorrent-nox",pid=6767,fd=17))`
`tcp   LISTEN 0      50                                               *:1340             *:*    users:(("qbittorrent-nox",pid=6767,fd=30))`

with qbittorrent stopped ( `systemctl stop qbittorrent`)  and vpn down (wg-quick down nl4-wireguard`) and starting qbittorrent again with `systemctl start qbittorrent` I get:
`journalctl -u qbittorrent`
`-- Journal begins at Tue 2023-01-31 07:58:39 CET, ends at Tue 2023-01-31 08:28:46 CET. --`
`(old logs deleted)`
`Jan 31 08:35:17 PiServer systemd[1]: Started qBittorrent (DietPi).`

and :
`ss -tulpn | grep qbit`
`tcp   LISTEN 0      50                   *:1340             *:*    users:(("qbittorrent-nox",pid=7007,fd=24))`

GvY85, 2023-01-31:
I also found out that the `bind interface` in qBittorrent makes it use IPv6 traffic only... so that is a bug there
https://github.com/qbittorrent/qBittorrent/issues/18494

MichaIng, 2023-02-05:
> So it seems then the `table` in `nl4-qbittorrent.conf` stops working and qBittorrent starts to use all interfaces (not good).

This is two different things:
- The routing table is only for outgoing packets. So qBittorrent cannot send a network packet to any remote host through `eth0` but only through `nl4-qbittorrent.conf`.
- The setting from the web interface, based on what the `ss` output shows, defines the addresses/interfaces on which qBittorrent is able to receive packets. I'm actually not sure about this `<ip>%<interface>` output. The interface part could even only be a comment while the relevant part is the IP. And the web interface allows you to select interfaces aside of IPs, because IPs can change (DHCP) while the interfaces usually stay the same. I guess you could instead also just bind to IP `10.65.51.10` and all interfaces instead, to achieve the same effect. Just in theory qBittorrent could tell the tracker through VPN how it is reachable via public IP and `eth0` and it could then receive downloads without ever sending any packet through `eth0`. It shouldn't know the public IP, but who knows. UDP is stateless. Can you disable listening on UDP and see whether the download can also bypass the VPN on TCP?

qBittorrent does not use the table because of the interface or IP binding, but because of the `ip rule` you created for the user. When you remove the `PostDown` from `nl4-qbittorrent.conf` and then disable the VPN, can you show that happened to the rules and the table?
```sh
ip rule
ip r
ip r l table 42
```
As the rule is effective on user basis, can you just check it like that, with VPN up and with VPN down:
```sh
sudo -u qbittorrent curl https://dietpi.com/geoip
```

Ah, btw, when used, the last two `PostDown` rules were meant to `del`ete the rules, while currently they are adding them (which should fail since they exist already):
```
PostDown = ip -4 rule del uidrange 997-997 table main suppress_prefixlength 0
PostDown = ip -6 rule del uidrange 997-997 table main suppress_prefixlength 0
```
But we want to keep all of them anyway and see whether this really enforces qBittorrent to use the VPN or still allows bypassing somehow.

GvY85, 2023-02-05:
With `nl4-wireguard` active:

`wg-quick up nl4-wireguard
`[#] ip link add nl4-wireguard type wireguard`
`[#] wg setconf nl4-wireguard /dev/fd/63`
`[#] ip -4 address add 10.xx.xx.xx/32 dev nl4-wireguard`
`[#] ip -6 address add fc00:bbbb:bbbb:xxxxx::x:xxxx/128 dev nl4-wireguard`
`[#] ip link set mtu 1420 up dev nl4-wireguard`
`[#] ip -6 route add ::/0 dev nl4-wireguard table 42`
`[#] ip -4 route add 0.0.0.0/0 dev nl4-wireguard table 42`
`[#] ip -4 rule add uidrange 997-997 table 42`
`[#] ip -6 rule add uidrange 997-997 table 42`
`[#] ip -4 rule add uidrange 997-997 table main suppress_prefixlength 0`
`[#] ip -6 rule add uidrange 997-997 table main suppress_prefixlength 0`
`[#] systemctl start qbittorrent`

`ip rule`
`0:      from all lookup local`
`32759:  from all uidrange 997-997 lookup main suppress_prefixlength 0`
`32760:  from all uidrange 997-997 lookup 42`
`32761:  from all uidrange 997-997 lookup main suppress_prefixlength 0`
`32762:  from all uidrange 997-997 lookup 42`
`32763:  from all uidrange 997-997 lookup main suppress_prefixlength 0`
`32764:  from all uidrange 997-997 lookup main suppress_prefixlength 0`
`32766:  from all lookup main`
`32767:  from all lookup default`

`ip r`
`default via 192.168.178.1 dev eth0 onlink`
`10.6.0.0/24 dev wg0 proto kernel scope link src 10.6.0.1`
`192.168.178.0/24 dev eth0 proto kernel scope link src 192.168.178.3`

(wg0 is the pivpn)

`ip r l table 42`
`default dev nl4-wireguard scope link`

`sudo -u qbittorrent curl https://dietpi.com/geoip`
`2a03:1b20:x:xxxx:xxxx North Holland NL` (= VPN IPV6 adress)

With `nl4-wireguard` down:

`wg-quick down nl4-wireguard`
`[#] ip link delete dev nl4-wireguard`
`[#] systemctl stop qbittorrent`

`ip rule`
`0:      from all lookup local`
`32759:  from all uidrange 997-997 lookup main suppress_prefixlength 0`
`32760:  from all uidrange 997-997 lookup 42`
`32761:  from all uidrange 997-997 lookup main suppress_prefixlength 0`
`32762:  from all uidrange 997-997 lookup 42`
`32763:  from all uidrange 997-997 lookup main suppress_prefixlength 0`
`32764:  from all uidrange 997-997 lookup main suppress_prefixlength 0`
`32766:  from all lookup main`
`32767:  from all lookup default`

`ip r`
`default via 192.168.178.1 dev eth0 onlink`
`10.6.0.0/24 dev wg0 proto kernel scope link src 10.6.0.1`
`192.168.178.0/24 dev eth0 proto kernel scope link src 192.168.178.3`

`ip r l table 42`
``

`sudo -u qbittorrent curl https://dietpi.com/geoip`
`xx.xx.xx.xx Somewhere in Somewhere` (= my IPV4 ISP address)

GvY85, 2023-02-05:
@Joulinar sorry, i tried adding a comment and acidently deleded your formatting. 
How do i properly inset multi-line code?

Joulinar, 2023-02-05:
````
```
test
```
````

MichaIng, 2023-02-05:
Okay, so the rules stay in place, but since WireGuard removes the routing table when it is shut down, the rules seem to be simply ignored (instead of preventing further traffic).

> So it seems only IPV6 traffic comes trough nl4-wireguard?

To test with VPN up:
```sh
sudo -u qbittorrent curl -4 https://dietpi.com/geoip
```
At eh same time, when the VPN is down, the following should return your ISPs IPv6 address:
```sh
sudo -u qbittorrent curl -6 https://dietpi.com/geoip
```

To check routes and rules for IPv6:
```sh
ip -6 r
ip -6 rule
```

One problem with leaving the rules in place: If you restart the VPN, the same rules are added again, so you have multiple redundant rules. But this practically isn't an issue.

I wonder whether re-adding table 42 with a blackhole route works:
```sh
PreUp = ip r del blackhole 0/0 table 42
PostDown = ip r add blackhole 0/0 table 42
```
So on/before VPN start, table 42 is removed, to allow WireGuard using it. And on/after VPN shutdown, it is added with a blackhole route, so no outgoing packets are possible. With the `suppress_prefixlength 0` rule still in place, it is still possible for qBittorrent to communicate via LAN or hosts within other VPN networks, but not any remote connection.

GvY85, 2023-02-05:
With `nl4-wireguard` up:
```
sudo -u qbittorrent curl -4 https://dietpi.com/geoip
^C
```
no reply 

With `nl4-wireguard` down I get no IPv6 response but that is because I have disabled the IPv6 connection in my router so makes sense.

With `nl4-wireguard`  up:
```
ip -6 r
::1 dev lo proto kernel metric 256 pref medium
fc00:bbbb:bbbb:bb01::2:3309 dev nl4-wireguard proto kernel metric 256 pref medium
fe80::/64 dev eth0 proto kernel metric 256 pref medium
```
```
ip -6 rule
0:      from all lookup local
32764:  from all uidrange 997-997 lookup main suppress_prefixlength 0
32765:  from all uidrange 997-997 lookup 42
32766:  from all lookup main
```
ip -4 r
default via 192.168.178.1 dev eth0 onlink
10.6.0.0/24 dev wg0 proto kernel scope link src 10.6.0.1
192.168.178.0/24 dev eth0 proto kernel scope link src 192.168.178.3
```
```
ip -4 rule
0:      from all lookup local
32764:  from all uidrange 997-997 lookup main suppress_prefixlength 0
32765:  from all uidrange 997-997 lookup 42
32766:  from all lookup main
32767:  from all lookup default
```

With `nl4-wireguard` down:
```
ip -6 r
::1 dev lo proto kernel metric 256 pref medium
fe80::/64 dev eth0 proto kernel metric 256 pref medium
```
```
ip -6 rule
0:      from all lookup local
32764:  from all uidrange 997-997 lookup main suppress_prefixlength 0
32765:  from all uidrange 997-997 lookup 42
32766:  from all lookup main
```
```
ip -4 r
default via 192.168.178.1 dev eth0 onlink
10.6.0.0/24 dev wg0 proto kernel scope link src 10.6.0.1
192.168.178.0/24 dev eth0 proto kernel scope link src 192.168.178.3
```
```
ip -4 rule
0:      from all lookup local
32764:  from all uidrange 997-997 lookup main suppress_prefixlength 0
32765:  from all uidrange 997-997 lookup 42
32766:  from all lookup main
32767:  from all lookup default
```

GvY85, 2023-02-05:
It seems `ip -6 r` does have a `nl4-wireguard` rule but `ip -4 r` does not?

MichaIng, 2023-02-05:
> Any idea why I only get IPv6 with the VPN up?

As you mentioned, because you disabled IPv6 in your router. But I wonder why you cannot connect via IPv4 with VPN up. Is Mullvad IPv6 only somehow, or is the IPv4 address in your config probably wrong? At least, with VPN up, the IPv4 request didn't bypass it.

About UFW, since you use a rule to allow IPv4 loopback traffic, what about IPv6 loopback traffic?
```
Anywhere ALLOW IN ::1/128
```
If not so much experience with UFW, but otherwise it looks good. Of course you can test with if disabled and see whether either qBittorrent performance and/or IPv4 issues are fixed. This would then also indicate probably a wrong IPv4 subnet.

Ahh, btw, now I see: You allow inbound traffic from VPN only on port 56886, but for DHT it listens on port 6771 as well (based on your `ss` output). I think in UFW the rules apply to inbound connections, while incoming packets on connections connections are still allowed right? Because with iptables rules apply individual packets, so that with one rule you can block all incoming packets and with another rule you can allow them for established connections. Could be actually reviewed what UFW does:
```sh
iptables -L
ip6tables -L
```

GvY85, 2023-02-05:
```
Anywhere ALLOW IN ::1/128
```
This makes sense. It is not in because this was a IPv4 only system in a IPv4 only network (I still dont understand IPv6...) until I migrated qBittorrent en the Mullvad VPN to it. Then I enable IPv6.
I have put it in with `ufw allow from ::1/128
`
Disabling UFW does not solve it (tried it before and just now to double check).

> As you mentioned, because you disabled IPv6 in your router. But I wonder why you cannot connect via IPv4 with VPN up

No? IPv6 is disable in the router so the home network is IPv4 and the IPS connection as well. But since Mullvad generated an IPv4 + IPv6 config I had to enable IPv6 on this device. So at least I would expect IPv4 traffic with Mullvad up and down.

I will log in an generate a new Mulvad config and double check the IPv4 adress in it to confirm. I am 100% they do both IPv4 and IPv6.

GvY85, 2023-02-05:
`iptables -L `
```
iptables -L
Chain INPUT (policy DROP)
target     prot opt source               destination
ufw-before-logging-input  all  --  anywhere             anywhere
ufw-before-input  all  --  anywhere             anywhere
ufw-after-input  all  --  anywhere             anywhere
ufw-after-logging-input  all  --  anywhere             anywhere
ufw-reject-input  all  --  anywhere             anywhere
ufw-track-input  all  --  anywhere             anywhere

Chain FORWARD (policy DROP)
target     prot opt source               destination
ufw-before-logging-forward  all  --  anywhere             anywhere
ufw-before-forward  all  --  anywhere             anywhere
ufw-after-forward  all  --  anywhere             anywhere
ufw-after-logging-forward  all  --  anywhere             anywhere
ufw-reject-forward  all  --  anywhere             anywhere
ufw-track-forward  all  --  anywhere             anywhere

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination
ufw-before-logging-output  all  --  anywhere             anywhere
ufw-before-output  all  --  anywhere             anywhere
ufw-after-output  all  --  anywhere             anywhere
ufw-after-logging-output  all  --  anywhere             anywhere
ufw-reject-output  all  --  anywhere             anywhere
ufw-track-output  all  --  anywhere             anywhere

Chain ufw-after-forward (1 references)
target     prot opt source               destination

Chain ufw-after-input (1 references)
target     prot opt source               destination
ufw-skip-to-policy-input  udp  --  anywhere             anywhere             udp dpt:netbios-ns
ufw-skip-to-policy-input  udp  --  anywhere             anywhere             udp dpt:netbios-dgm
ufw-skip-to-policy-input  tcp  --  anywhere             anywhere             tcp dpt:netbios-ssn
ufw-skip-to-policy-input  tcp  --  anywhere             anywhere             tcp dpt:microsoft-ds
ufw-skip-to-policy-input  udp  --  anywhere             anywhere             udp dpt:bootps
ufw-skip-to-policy-input  udp  --  anywhere             anywhere             udp dpt:bootpc
ufw-skip-to-policy-input  all  --  anywhere             anywhere             ADDRTYPE match dst-type BROADCAST

Chain ufw-after-logging-forward (1 references)
target     prot opt source               destination

Chain ufw-after-logging-input (1 references)
target     prot opt source               destination

Chain ufw-after-logging-output (1 references)
target     prot opt source               destination

Chain ufw-after-output (1 references)
target     prot opt source               destination

Chain ufw-before-forward (1 references)
target     prot opt source               destination
ACCEPT     all  --  anywhere             anywhere             ctstate RELATED,ESTABLISHED
ACCEPT     icmp --  anywhere             anywhere             icmp destination-unreachable
ACCEPT     icmp --  anywhere             anywhere             icmp time-exceeded
ACCEPT     icmp --  anywhere             anywhere             icmp parameter-problem
ACCEPT     icmp --  anywhere             anywhere             icmp echo-request
ufw-user-forward  all  --  anywhere             anywhere

Chain ufw-before-input (1 references)
target     prot opt source               destination
ACCEPT     all  --  anywhere             anywhere
ACCEPT     all  --  anywhere             anywhere             ctstate RELATED,ESTABLISHED
ufw-logging-deny  all  --  anywhere             anywhere             ctstate INVALID
DROP       all  --  anywhere             anywhere             ctstate INVALID
[... 294 lines cut ...]

Chain ufw6-user-limit (0 references)
target     prot opt source               destination
REJECT     all      anywhere             anywhere             reject-with icmp6-port-unreachable

Chain ufw6-user-limit-accept (0 references)
target     prot opt source               destination
ACCEPT     all      anywhere             anywhere

Chain ufw6-user-logging-forward (0 references)
target     prot opt source               destination
RETURN     all      anywhere             anywhere

Chain ufw6-user-logging-input (0 references)
target     prot opt source               destination
RETURN     all      anywhere             anywhere

Chain ufw6-user-logging-output (0 references)
target     prot opt source               destination
RETURN     all      anywhere             anywhere

Chain ufw6-user-output (1 references)
target     prot opt source               destination

```

MichaIng, 2023-02-05:
Uff, UFW does really a lot of stuff, split into individual targets. But I see a lot of allowed `ctstate RELATED,ESTABLISHED` traffic, i.e. such on established connections, which makes sense.

Okay understood so far regarding IPv4/IPv6:
- Disabled in router, so no GUA and no IPv6 default route advertised, so even with IPv6 enabled on device level, there is no IPv6 connection to remote hosts possible.
- With VPN up, an IPv6 address and route is assigned by WireGuard, so IPv6 traffic through the VPN is possible.

Still the question why IPv4 through the VPN does not work. UFW at least does not restrict outbound requests/packets at all.

Btw, the port 56886 inbound rules look strange:
```
ACCEPT     tcp  --  pi.hole              anywhere             tcp dpt:56886
ACCEPT     udp  --  pi.hole              anywhere             udp dpt:56886
...
ACCEPT     tcp      pi.hole              anywhere             tcp dpt:56886
ACCEPT     udp      pi.hole              anywhere             udp dpt:56886
```
Did you specify the origin correctly to imply the whole subnet? Like
```
56886                      ALLOW IN    10.65.xx.0/24                    (IPv4 nl4-wireguard open port)
56886                      ALLOW IN    fc00:xxxx:xxxx:xxxx::0/64        (IPv6 nl4-wireguard open port)
```
And as mentioned, you may want to allow port 6771 for DHT as well.

GvY85, 2023-02-05:
> Okay understood so far regarding IPv4/IPv6:
>
> Disabled in router, so no GUA and no IPv6 default route advertised, so even with IPv6 enabled on device level, there is no IPv6 connection to remote [...]

Exactly.  That is the big one, why no IPv4 over the VPN.

I added the UFW rules for `nl4-wireguard`  by using the ip as mentioned in the `nl4-wireguard` config file. So as `10.65.xx.xx/32` and `fc00:bbbb:xxxx:xxxx::x:xxxx/128`. That should be ok right?

Adding 6771 for DHT does not make any sense I think because 6771 is not forwarded by Mullvad so nothing will come through?

MichaIng, 2023-02-05:
> because 6771 is not forwarded by Mullvad so nothing will come through?

Ah okay, makes sense then. Probably you can then simply disable DHT completely.

> I added the UFW rules for `nl4-wireguard` by using the ip as mentioned in the `nl4-wireguard` config file. So as `10.65.xx.xx/32` and `fc00:bbbb:xxxx:xxxx::x:xxxx/128`. That should be ok right?

No that does not make sense since you only allow inbound traffic originating from your own system then, while you want to allow inbound traffic on those ports from remote systems through the VPN. Actually, if the VPN acts as a NAT for remote systems, then you'd even need to allow it for all IPs, not just for those with VPN IPs, i.e. the VPN server. Probably it is somehow possible to tell UFW to allow all inbound traffic originating/coming in from a specific interface, rather than originating from an IP range? With plain `iptables` at least this is possible.

GvY85, 2023-02-06:
Hi, 

Not sure we are exactly on the same page but regardless, that would be a seperate issue from the fact that `nl4-wireguard` does not do any IPv4 right?

I did found something strange. When doing `systemctl status wg-quick@nl4-wireguard` I get:

```
systemctl status wg-quick@nl4-wireguard
● wg-quick@nl4-wireguard.service - WireGuard via wg-quick(8) for nl4/wireguard
     Loaded: loaded (/lib/systemd/system/wg-quick@.service; enabled; vendor preset: enabled)
     Active: active (exited) since Sun 2023-02-05 14:15:07 CET; 18h ago
       Docs: man:wg-quick(8)
             man:wg(8)
             https://www.wireguard.com/
             https://www.wireguard.com/quickstart/
             https://git.zx2c4.com/wireguard-tools/about/src/man/wg-quick.8
             https://git.zx2c4.com/wireguard-tools/about/src/man/wg.8
    Process: 731 ExecStart=/usr/bin/wg-quick up nl4-wireguard (code=exited, status=0/SUCCESS)
   Main PID: 731 (code=exited, status=0/SUCCESS)
        CPU: 50ms

Feb 05 14:15:07 PiServer wg-quick[731]: [#] ip -6 address add fc00:bbbb:bbbb:bb01::2:3309/128 dev nl4-wireguard
Feb 05 14:15:07 PiServer wg-quick[731]: [#] ip link set mtu 1420 up dev nl4-wireguard
Feb 05 14:15:07 PiServer wg-quick[731]: [#] ip -6 route add ::/0 dev nl4-wireguard table 42
Feb 05 14:15:07 PiServer wg-quick[731]: [#] ip -4 route add 0.0.0.0/0 dev nl4-wireguard table 42
Feb 05 14:15:07 PiServer wg-quick[731]: [#] ip -4 rule add uidrange 997-997 table 42
Feb 05 14:15:07 PiServer wg-quick[731]: [#] ip -6 rule add uidrange 997-997 table 42
Feb 05 14:15:07 PiServer wg-quick[731]: [#] ip -4 rule add uidrange 997-997 table main suppress_prefixlength 0
Feb 05 14:15:07 PiServer wg-quick[731]: [#] ip -6 rule add uidrange 997-997 table main suppress_prefixlength 0
Feb 05 14:15:07 PiServer wg-quick[731]: [#] systemctl start qbittorrent
Feb 05 14:15:07 PiServer systemd[1]: Finished WireGuard via wg-quick(8) for nl4/wireguard.

``` 

Looks like, at the top, there is no attempt at a IPv4 connection at all?
While I am certain it is in the config and I tripple checked the IP.

To be sure:
```
cat /etc/wireguard/nl4-wireguard.conf
[Interface]
PrivateKey = (private key)
Address = 10.65.xx.xx/32,fc00:xxxx:xxxx:xxxx::x:xxxx/128
Table = 42
PostUp = ip -4 rule add uidrange 997-997 table 42
PostUp = ip -6 rule add uidrange 997-997 table 42
PostUp = ip -4 rule add uidrange 997-997 table main suppress_prefixlength 0
PostUp = ip -6 rule add uidrange 997-997 table main suppress_prefixlength 0
PostUp = systemctl start qbittorrent
#PostDown = ip -4 rule del uidrange 997-997 table 42
#PostDown = ip -6 rule del uidrange 997-997 table 42
#PostDown = ip -4 rule del uidrange 997-997 table main suppress_prefixlength 0
#PostDown = ip -6 rule del uidrange 997-997 table main suppress_prefixlength 0
PostDown = systemctl stop qbittorrent

[Peer]
PublicKey = (public key)
AllowedIPs = 0.0.0.0/0,::0/0
Endpoint = 193.32.249.69:54000
```

How to debug this?

GvY85, 2023-02-06:
Found something:

I decided to start all over with generating new Mulvad config files and notived that when I do not have the `table 42` stuff in Wireguard seems to add additional rules that are not generated when adding a manual `table`.

```
wg-quick up nl4-wireguard
[#] ip link add nl4-wireguard type wireguard
[#] wg setconf nl4-wireguard /dev/fd/63
[#] ip -4 address add 10.xx.xxx.xx/32 dev nl4-wireguard
[#] ip -6 address add fc00:bbbb:xxxx:xxxx::x:xxxx/128 dev nl4-wireguard
[#] ip link set mtu 1420 up dev nl4-wireguard
[#] wg set nl4-wireguard fwmark 51820
[#] ip -6 route add ::/0 dev nl4-wireguard table 51820
[#] ip -6 rule add not fwmark 51820 table 51820
[#] ip -6 rule add table main suppress_prefixlength 0
[#] ip6tables-restore -n
[#] ip -4 route add 0.0.0.0/0 dev nl4-wireguard table 51820
[#] ip -4 rule add not fwmark 51820 table 51820
[#] ip -4 rule add table main suppress_prefixlength 0
[#] sysctl -q net.ipv4.conf.all.src_valid_mark=1
[#] iptables-restore -n
```
```
wg-quick down nl4-wireguard
[#] ip -4 rule delete table 51820
[#] ip -4 rule delete table main suppress_prefixlength 0
[#] ip -6 rule delete table 51820
[#] ip -6 rule delete table main suppress_prefixlength 0
[#] ip link delete dev nl4-wireguard
[#] iptables-restore -n
[#] ip6tables-restore -n
```

MichaIng, 2023-02-06:
Yes this is expected: If you do not set a custom table, WireGuard sets up rules to enforce the routing table for the whole system. So everything is then tunnelled. As we wanted to tunnel qBittorrent only, we used a custom table, in which case WireGuard does not setup the rules, but we did ourselves, for the qBittorrent user only.

Btw, is IPv4 traffic possible with the default table/rules?

GvY85, 2023-02-06:
Yes, thanks for the recap, I almost got confused after all the trial and error :P 

But yes, the default config allows both IPv4 and IPv6 traffic.
So perhaps we need to add one or two additional rules to the `table`?

MichaIng, 2023-02-06:
Does the table actually exist for both, IPv4 and IPv6, when the VPN is up?
```sh
ip r l table 42
ip -6 r l table 42
```
The routes within the table are managed by WireGuard, we only added the rules about which requests/users use the table. So we never had a look into them. I think you posted and removed a comment where one could see that indeed an IPv4 route was missing, somehow?

GvY85, 2023-02-06:
Hi, that comment was a `systemctl status` run but that just cuts off lines so that was not relevant at all so I deleted it.

```
ip r l table 42
default dev nl4-wireguard scope link
```
```
ip -6 r l table 42
default dev nl4-wireguard metric 1024 pref medium
```

MichaIng, 2023-02-06:
Those look good, consistent at least. There is no exact gateway IP given (`via <IP>`), but that doesn't seem to be required.

GvY85, 2023-02-07:
I have put the output of the default `wg-quick up` config and ours side-by-side to see  if we can find and explain all the differences?

[image]

The differences I noticed (things we do not do in our config):

- `set fwmark`, line 9
- IPv6 `add not fwmark`, line 11
- IPv6 `ip6tables-restore -n, line 13
- IPv4 `add not fwmark`, line 15
- IPv4 `sysctl -q net.ipv4.conf.all.src_valid_mark=1`, line 17
- IPv4 `iptables-restore -n`

MichaIng, 2023-02-07:
All those mark stuff is not needed if the routing table is applied for a specific non-root user only. WireGuard uses this to allow root to contact the VPN server directly, to reestablish the connection when it was dropped and such, if otherwise everything is forced to use the VPN. I think the `src_valid_mark` is related to this.

Not sure why it restores iptables settings since it doesn't change them 🤔.

GvY85, 2023-02-08:
I tried adding the mark stuff and iptables-restore but indeed that does not work at all.

GvY85, 2023-02-08:
FINALLY! Solved it!

Had to change `net.ipv4.conf.default.rp_filter=1` to `net.ipv4.conf.default.rp_filter=2`......

Got the hint from a [random Github post](https://github.com/containers/podman/issues/10923) where they did something with  `ip rule add uidrange` as well.

My gosh I am glad. Thanks for all the time and help @MichaIng !

closed 2023-02-08 by GvY85

MichaIng, 2023-02-08:
Great, very valuable find!

I try to find some background on this: https://sysctl-explorer.net/net/ipv4/rp_filter/
- What is "FIB"?
- How is "the best reverse path the packet" estimated? Can we influence it?

Ah, well explained here: https://serverfault.com/a/932206/577419
- So if a packet was coming in from e.g. `nl4-wireguard` but would be routed back through `eth0`, the filter would discard it.
- Since an incoming packet does not belong to a user, and the kernel is doing the check, I guess reverse routing would be always going through `eth0`, since `nl4-wireguard` is used only by the `qbittorrent` user.
- With default WireGuard table this isn't an issue since all remotely outgoing traffic is routed through the VPN. Since the fwmark there is not set on incoming packets, the bypass for WireGuard itself to contact the server/provider outside of the VPN, does not apply.

So to me it seems that the relaxed rp_filter needs to be applied whenever the default route remains untouched and routing through a VPN (or other network) is only applied to a specific group, like user(s) or fwmark. In all those cases, when a packet is coming in from the VPN/other network, the kernel does the reverse check as root user (at least not the specific user(s)) and without the fwmark, so that it is routed through the default gateway, failing to a failed test.

To keep hardened checks for other interfaces, you should relax it for the VPN only, which needs to be done via PostUp then (as the interface does not exist at boot):
```sh
PostUp = sysctl net.ipv4.conf.nl4-wireguard.rp_filter=1
```

It should be possible to solve this by applying a connection mark to all connections related to incoming packets through `nl4-wireguard`, based on that applying an fwmark to all packets related to such marked connections, via `iptables`, and finally enforcing table 42 for those packets as well:
```sh
iptables -t mangle -A PREROUTING -i nl4-wireguard -m conntrack --ctstate NEW -j CONNMARK --set-mark 42
iptables -t mangle -A OUTPUT -m connmark --mark 42 -j MARK --set-mark 42
ip rule add fwmark 42 table 42
```
1. A packet comes in through `nl4-wireguard`. It is checked whether this is a new connection. If so, it gets a mark `42`.
    - `-m conntrack --ctstate NEW` can be also omitted, so a connection gets marked again on every incoming packet. Not sure whether the check is "heaver" than setting the mark. The mark needs to be set only once for a connection, it remains, but it doesn't hurt to apply it again and again.
3. An outgoing packet is about to be sent. If it has the connection mark `42` it gets the packet mark/fwmark `42`.
4. Since it has the fwmark `42`, it is forced to use the routing table `42`.

I use `42` here for connection mark, packet mark/fwmark as well as routing table, just so it can be seen what belongs together, while those can be freely chosen, of course 😄.

GvY85, 2023-02-09:
`PostUp = sysctl net.ipv4.conf.nl4-wireguard.rp_filter=**2**` ;) is a good tip indeed.

Final config:
```
Table = 42
PostUp = ip -4 rule add uidrange 997-997 table 42
PostUp = ip -6 rule add uidrange 997-997 table 42
PostUp = ip -4 rule add uidrange 997-997 table main suppress_prefixlength 0
PostUp = ip -6 rule add uidrange 997-997 table main suppress_prefixlength 0
PostUp = sysctl net.ipv4.conf.nl4-wireguard.rp_filter=2
PostUp = systemctl start qbittorrent
PostDown = ip -4 rule del uidrange 997-997 table 42
PostDown = ip -6 rule del uidrange 997-997 table 42
PostDown = ip -4 rule del uidrange 997-997 table main suppress_prefixlength 0
PostDown = ip -6 rule del uidrange 997-997 table main suppress_prefixlength 0
PostDown = systemctl stop qbittorrent
```

MichaIng, 2023-02-09:
Looks very good. Does it now also work without interface binding without any packets bypassing the VPN, and without performance decrease?

GvY85, 2023-02-10:
Without perf decrease but it is no kill switch. As long as the VPN is up all `qbitttorrent` user traffic goes via the VPN, regardless of any in-app settings that bind it to the VPN or not.
But with the VPN down (even without `PostDown = ip -4 rule del uidrange 997-997 table 42` etc), if no `bind to interface` is set in qBittorrent, it feels free to use any available interface.
So the `PostDown = systemctl stop qbittorrent` is still needed

MichaIng, 2023-02-10:
Great. Yeah its sad that the rule/routing table is ignored once the interface has been removed, it would be much easier if it would stay in place, forcing qBittorrent packets into nirvana instead. At least, I think if the connection to the VPN server breaks (without WireGuard being shutdown ordinary), qBittorrent won't be able to bypass it, as the interface is still there. So it should be safe to `systemctl stop qbittorrent` on ordinary shutdown to prevent any leaks. Ah one suggestion:
```sh
PreDown = systemctl stop qbittorrent
```
So qBittorrent is stopped before the VPN stopped, closing a little time frame where it can bypass the VPN.

GvY85, 2023-02-10:
Good one, changed the `PostDown` to `Predown`