MichaIng/DietPi #6074: Can't update 2

MurphyDoe0815, 2023-01-18:
#### Details:
- Date           | Wed Jan 18 16:54:40 CET 2023
- DietPi version | v8.12.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | Native PC (x86_64) (ID=21)
- Kernel version | `Linux DietPi 5.10.0-20-amd64 #1 SMP Debian 5.10.158-2 (2022-12-13) x86_64 GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `getent hosts dns9.quad9.net`
- Exit code      | 2
- Software title | DietPi-Update
#### Steps to reproduce:

#### Expected behaviour:
Run now to update DietPi from v8.12.1 to v8.13.2
- ...
#### Actual behaviour:
no update possible

Joulinar, 2023-01-18:
Looks like DNS issue and you are not able to resolve Quad9 global host address. What DNS server you have set with DietPi network settings?

MurphyDoe0815, 2023-01-18:
All right, worked after I turned off pihole. No idea why it suddenly did not work. I have the configuration for a really long time without problems. Pihole is also on google as dns

Thanks for your help

Joulinar, 2023-01-18:
Question is what you have set locally in DietPi system

```
cat /etc/resolv.conf
```

MurphyDoe0815, 2023-01-18:
domain fritz.box
search fritz.box
nameserver 192.168.178.1

Joulinar, 2023-01-18:
And your Fritzbox use PiHole as upstream DNS server? PiHole is running on that DietPi system? And you use DHCP instead of STATIC IP?

MurphyDoe0815, 2023-01-18:
Dietpi runs separately on proxmox with an additional lxc with pihole, both with their own IP

the IP in fritzbox as dns 1 is the IP of pihole and 8.8.8.8 as dns 2

Joulinar, 2023-01-18:
Ah ok now it make sense why disabling PiHole solved the issue. Your Fritzbox was failing back to Google public DNS. 

Ok did you checked PiHole if there was something blocked?

MurphyDoe0815, 2023-01-18:
just now I could not find it - now I have looked up the logs directly
2023-01-18 18:06:35 | A | dns9.quad9.net | fritz.box | Blocked (gravity) 
2023-01-18 18:06:35 | AAAA | dns9.quad9.net | fritz.box | Blocked (gravity) 

both I have now given free

Joulinar, 2023-01-18:
Usually there is a feature within PiHole where you could check from which block list this is coming from. Quite strange that someone added Quad9 global DNS server to a block list.

MurphyDoe0815, 2023-01-18:
https://raw.githubusercontent.com/hagezi/dns-blocklists/main/domains/doh-vpn-proxy-bypass.txt

quad9.net
dns.quad9.net
dns-nosec.quad9.net
dns10.quad9.net
dns11.quad9.net
dns12.quad9.net
dns13.quad9.net
dns9.quad9.net

Joulinar, 2023-01-18:
Wtf, this one is blocking Cloudflare host names as well. Did you add the block list recently? Any specific reason for this? I'm asking just out of interest.

MichaIng, 2023-01-18:
Based on the name of the block list, it seems to aim blocking DoH requests (e.g. by browsers) which could bypass the system DNS, VPN or proxy. A properly configured VPN forces also DoH requests through the tunnel, of course, so it doesn't make much sense. It can bypass Pi-hole itself as well, but then the blocklist isn't applied in the first place, and considering that Pi-hole itself might use one of these DNS upstream servers, it is nonsense.

However, it just is a good example why randomly applying any blocklist found on the Internet is not a good procedure. Check carefully what which blocklist is actually for 😉.

MurphyDoe0815, 2023-01-18:
i have them here
https://github.com/RPiList/specials/blob/master/Blocklisten.md
#(Methods to prevent the pi-hole from being bypassed: Blocks DoH/DoT/VPNs/Proxies)

it was just a trial because i run all my phones and computers with tailscale and i noticed how the dns was bypassed
that now also the quad dns was blocked surprised me now also

MichaIng, 2023-01-18:
I needed to think twice, but it actually makes sense. DoH and DoT can be used to bypass the system DNS and hence Pi-hole. It does not break Pi-hole's own upstream DNS because that is an IP and not a hostname. While if DoH or DoT is used, Pi-hole and hence the blocklist is bypassed, for resolving the upstream providers hostname as very first step, the system DNS (Pi-hole) is used, i.e. DoH and DoT cannot be bootstrapped anymore.

So whitelisting `dns9.quad9.net` or configuring another hostname for the DNS resolver check in `dietpi.txt` seems a good solution.

MurphyDoe0815, 2023-01-18:
i have another pihole with tailscale , on which within tailscale the pihole is entered as dns. here the mobile is almost ad-free - but also not always. 
wireguard on the other hand is with the fritzbox under wireguard with now 7.50 also on the way (also besides with the cell phone)
Here the success is less good and also not always.
Therefore, I found the experiment with the block list quite interesting.

Joulinar, 2023-01-18:
Personal I have PiHole/Wireguard running on my DietPi production system. Using Wireguard mobile client is giving me the very same AdBlock experience as I would stay at home. 😄

MurphyDoe0815, 2023-01-18:
well, whether this is all so right, what you are telling me now ;-)
 i want to believe it now like this :-)