MichaIng/DietPi #6076: dietpi-update fails

jquack101, 2023-01-18:
#### Details:
- Date           | Wed Jan 18 10:50:49 PST 2023
- DietPi version | v8.12.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | `Linux PiHole01 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux`
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | `apt-get -y update`
- Exit code      | 255
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Get:1 https://download.docker.com/linux/raspbian buster InRelease [33.6 kB]
Get:2 https://apt.grafana.com stable InRelease [5984 B]
Ign:3 https://download.webmin.com/download/repository sarge InRelease
Hit:4 https://download.webmin.com/download/repository sarge Release
Hit:5 http://raspbian.raspberrypi.org/raspbian buster InRelease
Err:2 https://apt.grafana.com stable InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 9E439B102CF3C0C6
Hit:6 https://archive.raspberrypi.org/debian buster InRelease
Fetched 39.6 kB in 1s (29.0 kB/s)
Reading package lists...
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://apt.grafana.com stable InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 9E439B102CF3C0C6
W: Failed to fetch https://apt.grafana.com/dists/stable/InRelease  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 9E439B102CF3C0C6
W: Some index files failed to download. They have been ignored, or old ones used instead.
```

Joulinar, 2023-01-18:
Please update to latest DietPi to solve the issue.

MichaIng, 2023-01-18:
Can you show the output of:
```sh
date -u '+%Y%m%d' -r '/etc/apt/trusted.gpg.d/dietpi-grafana.gpg'
```

jquack101, 2023-01-18:
Hiya - actually just now resolved the issue via this:
https://packages.grafana.com/oss/

Having said that, thank you for the quick reponse!

Joulinar, 2023-01-18:
Usually this should be done by our update script. Therefore it would have been interesting why this did not worked in your case. But you solved it now.

MichaIng, 2023-01-18:
Probably Grafana was not installed via `dietpi-software`. Otherwise you'd have now a redundant key installed 🙄. I.e. would still be great if you could run the above command so we can see whether you did install it via `dietpi-software` and if so, why the patch did not apply.

jquack101, 2023-01-18:
Ok, I am happy to help further if I can - logs, etc - let me know

jquack101, 2023-01-18:
root@PiHole01:~# date -u '+%Y%m%d' -r '/etc/apt/trusted.gpg.d/dietpi-grafana.gpg'
date: /etc/apt/trusted.gpg.d/dietpi-grafana.gpg: No such file or directory

Joulinar, 2023-01-18:
How did you installed Grafana? Yourself or via our tools?

jquack101, 2023-01-18:
I am almost 100% sure I did this myself.

Joulinar, 2023-01-18:
Ok that would explain why our update script did not detect the key file correctly.

MichaIng, 2023-01-18:
Okay, makes all sense then. We just update the key for a `dietpi-software` based Grafana install, where we know intention, file path etc.

jquack101, 2023-01-18:
Yep, understood and agreed. Here's the thing with me - I have been running ubuntu as my main desktop os for years and installing/updating via apt. Perhaps for my dietpi installs I should always use the dietpi-software method?

Joulinar, 2023-01-18:
You are free to choose the method on how to install software. There is no strict requirement to use our tooling. From our side we just can assume software being installed based on our standard approach. It's hard to estimate all possible ways how things could be done. Therefore in case like yours, manual user intervention is needed. 😁

Joulinar, 2023-01-18:
I guess initial issue has been solved. Going to close the topic for now. Feel free to reopen if needed.

closed 2023-01-18 by Joulinar