MichaIng/DietPi #6080: Dietpi-Software - WordPress / Webstack Optimized Install Auto Configure

holocronology, 2023-01-20:
#### Is your feature request related to a problem? Please describe:
Yes, to avoid having to delve into internals.

#### Describe the solution you'd like:
Wordpress install, with lighttpd webstack at least, does not default to answering on Port 80 or base url. One must go to http://DEVICE-IP/wordpress.

#### Describe alternatives you've considered:
Include an option in the optimized software install, after selecting webstack, to auto-configure the webserver to include any required configuration changes for WP install to answer on Port 80 (or other port) at base url. So one could go to http://DEVICE-IP or http://DEVICE-IP:PORT.

Joulinar, 2023-01-20:
Well, we explicitly install all web server apps to be setup on a sub path. Why? Because we like to ensure compatibility with all other web server apps, that user are able to run them in parallel.

holocronology, 2023-01-20:
Fair points. But in the interest of accessibility and ease of use, this could be built into the install process. At the least, the option to configure/install any redirect or rewrites as needed.

Joulinar, 2023-01-20:
if we do this for a single app, we will need to do it for all software apps.

holocronology, 2023-01-20:
ok, thanks.

MichaIng, 2023-01-20:
Feature request is here already: #1608
For Wordpress we need to add proper access limits which do not rely in the `.htaccess` files read by Apache only. I.e. currently I highly suggest to install it with Apache, not with Lighttpd and Nginx.

closed 2023-01-20 by holocronology