MichaIng/DietPi #6081: DietPi-Backup: Add option to generate tarball file (.tar)

StephanStS, 2023-01-20:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
See https://www.reddit.com/r/dietpi/comments/zzt22u/comment/j2du4n8/?context=3

#### Describe the solution you'd like:
The user wants to be able to generate .tar files out of DietPi-Backup and transfer them via ftp/sftp to a target location.

Optional, a .xz compression would be helpful (with/without password?).

MichaIng, 2023-01-20:
The idea came up again here, and was dropped: #816

Basically, compressing the freshly generated single `rsync` backup does not make sense in very most cases. If you want it on a remote server, mount it's NFS/FTP/SFTP directory and use this as target drive for the backup.

However, with the nowadays supported backup archive, the additional temporary space requirement is not such an argument anymore compared to the absolute space requirement. So I'd reconsider this feature.

For transferring to a remote system, aside of mounting it as network drive, `rsync` also natively supports syncing to a remote system directly (with optional compressed transfer) where either an SSH server or an `rsync` daemon is running: https://manpages.debian.org/rsync
Discussed here: #2364
Became more complicated to implement with the backup archive, where we'd need to move directories on the remote system.

StephanStS, 2023-01-21:
So, let's go back to define the use cases:
- Backup files shall be remote (different location)
- Remote files shall be encrypted
- Transmission to remote shall be encrypted

MichaIng, 2023-01-22:
Point 1 and 3 can be achieved with a network drive. Actually, not sure whether NFS and Samba are or can be encrypted? I guess not. Our FTP servers at least have the option. In doubt, SSHFS!

But I understand now that it is about encryption and not only about compression, while both can be achieved with the same tool set.

GvY85, 2023-01-23:
What I usually do is use Midnight Commander to generate a .tar.somecompression file of the DietPi backup directory to store somewhere else as backup.

paradix, 2023-02-27:
I do this by rsync'ing the backup made by dietpi-backup to a remote location over ssh, afterwards I execute a remote command (bash script) that handles the compression / encryption