MichaIng/DietPi #6082: Installing `multipath-tools` package results in error on RP4

nodje, 2023-01-20:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=13
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='not applicable'
```
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
`bullseye 0`
- Kernel version | `uname -a`
`Linux control01 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux`

apt install multipath-tools yields
` modprobe: WARNING: Module dm-multipath not found in directory /lib/modules/5.15.84-v8+`

This is used to mount iSCSI devices.

this apparently stems form a missing config CONFIG_DM_MULTIPATH=m in kernel builds.

This was recently fixed on Raspberry linux kernel repo: https://github.com/raspberrypi/linux/issues/4886

I guess this is going to be solved naturally when upgrading to the fixed RPI kernel, but I'm not sure really whether this is the kernel used by DietPi.

Joulinar, 2023-01-20:
DietPi don't have own kernel builds. We use the original one created by RPi foundation.

MichaIng, 2023-01-20:
> but I'm not sure really whether this is the kernel used by DietPi.

This kernel is used by DietPi, but changes in this repo do not appear immediately in the Raspberry Pi APT repository. You can follow the kernel versions merged for these builds here: https://github.com/raspberrypi/firmware/tree/stable
Everything merged into the `stable` branch, will be available via `apt upgrade` within a day or so.

closed 2023-01-20 by nodje