MichaIng/DietPi #6083: Pi Zero, First boot cannot update from 8.12.1 to 8.13.2 then cannot access anymore

rodriguez5789, 2023-01-20:
### Creating a bug report/issue
Hi,

First boot and first connection by ssh to Raspberry pi zero v2.

Cannot update from 8.12.1 to 8.13.2 and then cannot connect by ssh even after unplug replug.

#### Required Information
- DietPi version | `8.12.1`
- Distro version | `DietPi_RPi-ARMv6-Bullseye
- Kernel version | `uname -a`
- SBC model | `Raspberry Pi Zero v2
- Power supply used |  5V 1A power
- SD card used | Samsung

#### Additional Information (if applicable)
- Can this issue be replicated on a fresh installation of DietPi?
- Yes

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

First boot
First ssh connection

#### Expected behaviour

Updating and starting

#### Actual behaviour

unable to update and cannot connect to ssh anymore

[Terminal Saved Output.txt](https://github.com/MichaIng/DietPi/files/10468733/Terminal.Saved.Output.txt)

Joulinar, 2023-01-20:
In the middle of the kernel update, your file system went into read only mode. This indicates a file system corruption. I would recommend to use a different SD card.

rodriguez5789, 2023-01-20:
> In the middle of the kernel update, your file system went into read only mode. This indicates a file system corruption. I would recommend to use a different SD card.

Thank you for your quick reply.

Another SD card actually fixed the issue as you recommended.

I am closing the topic.

Thank you

closed 2023-01-20 by rodriguez5789