MichaIng/DietPi #6084: Gogs install error

holocronology, 2023-01-20:
#### Details:
- Date           | Fri Jan 20 16:17:10 EST 2023
- Bug report     | 0599a6c1-a096-47fc-9ae0-928d9f2a29c3
- DietPi version | v8.13.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi Zero W (armv6l) (ID=1)
- Kernel version | `Linux FuzzPiZero 5.15.84+ #1613 Thu Jan 5 11:58:09 GMT 2023 armv6l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `curl -sSfL https://dietpi.com/downloads/binaries/all/gogs_armv6.zip -o gogs_armv6.zip`
- Exit code      | 22
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
curl: (22) The requested URL returned error: 404
```

Joulinar, 2023-01-20:
hmm the `gogs` package for ARMv6 is missing completely on our download server. Something we need to check on our end.

MichaIng, 2023-01-20:
Whoops, my mistake. I made some cleanup of ancient downloads but missed that for ARMv6 we still downloaded it from there. However, probably better that way, since the version was ancient, not responsible to ship this anymore.

I'll have a look about compiling an own archive from source.

MichaIng, 2023-01-21:
The fresh Gogs v0.12.10 archive has been uploaded.

MichaIng, 2023-01-22:
Regarding #6087, I guess you tried to reinstall Gogs while it never started? Those dot files are created when using Gogs, but is not included in the archive, neither ours nor the official. So attempting to move dot files on reinstall fails as long as Gogs was not used/failed on initial install like in your case. In `dev` I already adjusted it, as we do not need to touch the whole install dir anyway, but simply update the binary/executable only, making it all much easier.

However, in your case, please try again to reinstall, and when this `mv /etc/gogs/.??*` error appears, choose "Change command" from the error handler and replace the whole line with true. I'll also do some more tests now. Was messing around yesterday with the build workflow itself, to provide per-distro version builds with newest possible toolchain each.

holocronology, 2023-01-22:
Yes, I installed and could not get to the first run setup. So I tried a clean install again with the same results. http://my-ip:3000/install did not work either (per GOGS help).

I did the dietpi-software reinstall 49, on the off chance that something didn't work or that I wasn't pulling the update correctly.

holocronology, 2023-01-22:
still fails to connect

MichaIng, 2023-01-22:
It works well on my Raspberry Pi Zero. Can you share the reinstall log? And can you show:
```sh
journalctl -u gogs
```

holocronology, 2023-01-23:
I did a clean reinstall again and it's working as expected.

Thanks for the assistance.

closed 2023-01-23 by holocronology