MichaIng/DietPi #6085: xserver-xorg-video-vmware missing

mews-se, 2023-01-21:
#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=13
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='not applicable'`
- Distro version | `bullseye`
- Kernel version | `Linux mbp 5.10.0-20-arm64 #1 SMP Debian 5.10.158-2 (2022-12-13) aarch64 GNU/Linux`
- SBC model | `Virtual Machine (aarch64)`

#### Additional Information (if applicable)
- Software title | X11 MATE

- Bug report ID | `ed20a119-46d3-41df-b832-6a1f5da0ffeb`

#### Actual behaviour
Package xserver-xorg-video-vmware is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

#### Extra details
Package most likely not necessary, since it runs fine without it

Joulinar, 2023-01-21:
ok the challenge is you are running ARM based virtual machine. But the package is available on AMD64. What physical hardware your VM is based on?

mews-se, 2023-01-21:
Yes this is the gist of it. It's a test VM running on a Macbook Pro M2 (in Parallels), based on an ARM netinstall of "true" Debian and then running the dietpi-installer to "convert" it. I tried to make it as messy as possible to see if something broke :)

Joulinar, 2023-01-21:
@MichaIng 
Maybe we would need to obtain the architecture before installation of `xserver-xorg-video-vmware` ??

MichaIng, 2023-01-21:
Makes sense.

MichaIng, 2023-01-21:
Fixed with: https://github.com/MichaIng/DietPi/commit/c69ff17

closed 2023-01-23 by MichaIng