MichaIng/DietPi #6086: mympd - not working after DietPi update 8.11 -> 8.13

sofad, 2023-01-21:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=13
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='not applicable'
```
- Distro version | buster 1
- Kernel version | 
Linux DietPi 5.10.103+ #1529 Tue Mar 8 12:19:18 GMT 2022 armv6l GNU/Linux- 
SBC model | RPi Zero W (armv6l)
- Power supply used | 5V 1A
- SD card used | (Samsung EVO

#### Additional Information (if applicable)
- Software title | mympd
- Was the software title installed freshly or updated/migrated? updated dietpi from 8.11 to 8.13
- Can this issue be replicated on a fresh installation of DietPi? 

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. Try to open http://local.ip:1333 -> not reachable
2. Tried to reinstall with dietpi-software reinstall 148
3.  Tried reinstall with uninstalling using these commands:
systemctl unmask mympd
systemctl disable --now mympd
apt purge mympd
rm /etc/apt/sources.list.d/dietpi-mympd.list
rm /etc/apt/trusted.gpg.d/dietpi-mympd.gpg
rm -Rf /var/lib/mympd
rm -Rf /var/lib/private/mympd
rm -Rf /var/cache/private/mympd
rm -Rf /var/cache/mympd
rm -f /usr/share/man/man1/mympd* /etc/mympd.conf*

#### Expected behaviour
mympd shoud open in browser
- ...

#### Actual behaviour

mympd is not reachable 
Tried also
https://local.ip:1333
http://local.ip:1333

https://local.ip:1332
http://local.ip:1332

#### Extra details

`root@DietPi:~# sudo mympd -c
**12:53:28 ERROR    mympd     User "mympd" does not exist**
Exiting erroneous, thank you for using myMPD`

`root@DietPi:~# sudo systemctl restart mympd`

`root@DietPi:~# journalctl -u mympd
-- Logs begin at Sat 2023-01-21 00:55:12 CET, end at Sat 2023-01-21 12:54:33 CET. --
Jan 21 00:55:49 DietPi systemd[1]: Starting LSB: myMPD - a standalone and mobile friendly web mpdclient...
Jan 21 00:55:49 DietPi mympd[522]: Starting the process: mympd.
Jan 21 00:55:49 DietPi systemd[1]: Started LSB: myMPD - a standalone and mobile friendly web mpdclient.
Jan 21 01:09:03 DietPi systemd[1]: Stopping LSB: myMPD - a standalone and mobile friendly web mpdclient...
Jan 21 01:09:03 DietPi mympd[994]: Stoppping the mympd process is not running ... failed!
Jan 21 01:09:03 DietPi systemd[1]: mympd.service: Succeeded.
Jan 21 01:09:03 DietPi systemd[1]: Stopped LSB: myMPD - a standalone and mobile friendly web mpdclient.
Jan 21 01:09:03 DietPi systemd[1]: Starting LSB: myMPD - a standalone and mobile friendly web mpdclient...
Jan 21 01:09:03 DietPi mympd[1000]: mympd process is not running ... failed!
Jan 21 01:09:03 DietPi mympd[1000]: Starting the process: mympd.
Jan 21 01:09:03 DietPi systemd[1]: Started LSB: myMPD - a standalone and mobile friendly web mpdclient.
Jan 21 12:38:25 DietPi systemd[1]: Stopping LSB: myMPD - a standalone and mobile friendly web mpdclient...
Jan 21 12:38:25 DietPi mympd[6131]: Stoppping the mympd process is not running ... failed!
Jan 21 12:38:25 DietPi systemd[1]: mympd.service: Succeeded.
Jan 21 12:38:25 DietPi systemd[1]: Stopped LSB: myMPD - a standalone and mobile friendly web mpdclient.
Jan 21 12:38:25 DietPi systemd[1]: Starting LSB: myMPD - a standalone and mobile friendly web mpdclient...
Jan 21 12:38:26 DietPi mympd[6137]: mympd process is not running ... failed!
Jan 21 12:38:26 DietPi mympd[6137]: Starting the process: mympd.
Jan 21 12:38:26 DietPi systemd[1]: Started LSB: myMPD - a standalone and mobile friendly web mpdclient.
Jan 21 12:54:32 DietPi systemd[1]: Stopping LSB: myMPD - a standalone and mobile friendly web mpdclient...
Jan 21 12:54:33 DietPi mympd[6283]: Stoppping the mympd process is not running ... failed!
Jan 21 12:54:33 DietPi systemd[1]: mympd.service: Succeeded.
Jan 21 12:54:33 DietPi systemd[1]: Stopped LSB: myMPD - a standalone and mobile friendly web mpdclient.
Jan 21 12:54:33 DietPi systemd[1]: Starting LSB: myMPD - a standalone and mobile friendly web mpdclient...
Jan 21 12:54:33 DietPi mympd[6289]: mympd process is not running ... failed!
Jan 21 12:54:33 DietPi mympd[6289]: Starting the process: mympd.
Jan 21 12:54:33 DietPi systemd[1]: Started LSB: myMPD - a standalone and mobile friendly web mpdclient.
root@DietPi:~# 
`

If any information is needed, i'm happy to provide it. Thank you.

MichaIng, 2023-01-21:
Many thanks for your report.

The last log entry says it was started. It's probably just listening on a different port?
```sh
ss -tlpn
```

sofad, 2023-01-21:
```
root@DietPi:~# ss -tlpn
State     Recv-Q    Send-Q                                   Local Address:Port          Peer Address:Port                                                                                                                  
LISTEN    0         128                                            0.0.0.0:111                0.0.0.0:*        users:(("rpcbind",pid=197,fd=4),("systemd",pid=1,fd=37))                                                     
LISTEN    0         1024                                           0.0.0.0:80                 0.0.0.0:*        users:(("lighttpd",pid=508,fd=4))                                                                            
LISTEN    0         128                                            0.0.0.0:58641              0.0.0.0:*        users:(("rpc.mountd",pid=480,fd=9))                                                                          
LISTEN    0         32                                             0.0.0.0:53                 0.0.0.0:*        users:(("pihole-FTL",pid=403,fd=5))                                                                          
LISTEN    0         128                                            0.0.0.0:22                 0.0.0.0:*        users:(("sshd",pid=377,fd=3))                                                                                
LISTEN    0         255                                            0.0.0.0:7000               0.0.0.0:*        users:(("shairport-sync",pid=376,fd=6))                                                                      
LISTEN    0         64                                             0.0.0.0:2049               0.0.0.0:*                                                                                                                     
LISTEN    0         128                                            0.0.0.0:49057              0.0.0.0:*        users:(("rpc.mountd",pid=480,fd=13))                                                                         
LISTEN    0         128                                            0.0.0.0:5252               0.0.0.0:*        users:(("dietpi-dashboar",pid=368,fd=6))                                                                     
LISTEN    0         5                                            127.0.0.1:4711               0.0.0.0:*        users:(("pihole-FTL",pid=403,fd=10))                                                                         
LISTEN    0         5                                            127.0.0.1:6600               0.0.0.0:*        users:(("mpd",pid=513,fd=11))                                                                                
LISTEN    0         64                                             0.0.0.0:33129              0.0.0.0:*                                                                                                                     
LISTEN    0         128                                            0.0.0.0:57645              0.0.0.0:*        users:(("rpc.mountd",pid=480,fd=17))                                                                         
LISTEN    0         128                                               [::]:111                   [::]:*        users:(("rpcbind",pid=197,fd=6),("systemd",pid=1,fd=39))                                                     
LISTEN    0         1024                                              [::]:80                    [::]:*        users:(("lighttpd",pid=508,fd=5))                                                                            
LISTEN    0         5            [fd9e:b49e:f446:4dea:ba27:ebff:fe3b:7df9]:56625                 [::]:*        users:(("shairport-sync",pid=376,fd=23))                                                                     
LISTEN    0         32                                                [::]:53                    [::]:*        users:(("pihole-FTL",pid=403,fd=7))                                                                          
LISTEN    0         128                                               [::]:22                    [::]:*        users:(("sshd",pid=377,fd=4))                                                                                
LISTEN    0         255                                               [::]:7000                  [::]:*        users:(("shairport-sync",pid=376,fd=7))                                                                      
LISTEN    0         128                                               [::]:50233                 [::]:*        users:(("rpc.mountd",pid=480,fd=15))                                                                         
LISTEN    0         64                                                [::]:44027                 [::]:*                                                                                                                     
LISTEN    0         128                                               [::]:43803                 [::]:*        users:(("rpc.mountd",pid=480,fd=11))                                                                         
LISTEN    0         64                                                [::]:2049                  [::]:*                                                                                                                     
[... cut ...]
LISTEN    0         5                                                [::1]:4711                  [::]:*        users:(("pihole-FTL",pid=403,fd=11))                                                                         
LISTEN    0         5                                                [::1]:6600                  [::]:*        users:(("mpd",pid=513,fd=10))                                                                                
LISTEN    0         5            [fd9e:b49e:f446:4dea:ba27:ebff:fe3b:7df9]:38605                 [::]:*        users:(("shairport-sync",pid=376,fd=20))  
```

MichaIng, 2023-01-21:
Okay, definitely not up. This combination of startup errors is strange as well, I didn't see this when testing last time:
```
Jan 21 12:54:33 DietPi systemd[1]: Starting LSB: myMPD - a standalone and mobile friendly web mpdclient...
Jan 21 12:54:33 DietPi mympd[6289]: mympd process is not running ... failed!
Jan 21 12:54:33 DietPi mympd[6289]: Starting the process: mympd.
Jan 21 12:54:33 DietPi systemd[1]: Started LSB: myMPD - a standalone and mobile friendly web mpdclient.
```

Okay on a fresh install, somehow the service unit is missing completely:
```console
# dpkg -L mympd
/.
/etc
/etc/init.d
/etc/init.d/mympd
/usr
/usr/bin
/usr/bin/mympd
/usr/bin/mympd-script
/usr/share
/usr/share/doc
/usr/share/doc/mympd
/usr/share/doc/mympd/changelog.Debian.gz
/usr/share/doc/mympd/changelog.gz
/usr/share/doc/mympd/copyright
/usr/share/man
/usr/share/man/man1
/usr/share/man/man1/mympd-script.1.gz
/usr/share/man/man1/mympd.1.gz
```
Something broke with the last release yesterday, but I cannot see something related in changelog, probably a packaging bug: https://github.com/jcorporation/myMPD/releases/tag/v10.2.0

Need to check back with developer.

MichaIng, 2023-01-21:
Reported: https://github.com/jcorporation/myMPD/issues/955

MichaIng, 2023-01-21:
Recognised and fixed a few other issues for fresh installs: https://github.com/MichaIng/DietPi/commit/4179be5

sofad, 2023-01-21:
I created the unit manually, as you have explained.
MyMpd 10.2.0 is running as expected again.

MichaIng, thank you very very much!

sofad, 2023-01-21:
Sorry to report, myMPD has issues: Music is playing, can be skipped. But the Icons like "Search" "Artist" "Album" produce a javascript error message:

[image]
 <
[image]

MichaIng, 2023-01-21:
I'll try to replicate, but am out of houre for ~3 hours.

jcorporation, 2023-01-21:
> Sorry to report, myMPD has issues: Music is playing, can be skipped. But the Icons like "Search" "Artist" "Album" produce a javascript error message:
> 
> [image]
> <
> 
> [image]

myMPD v10.2.0 has renamed some views. Please try:
- in the uri: remove all after the # character
- readd all home icons that point to database views

This is mentioned in the release notes: https://github.com/jcorporation/myMPD/releases/tag/v10.2.0

MichaIng, 2023-01-23:
I'll mark this as closed since the original issue is solved.

closed 2023-01-23 by MichaIng