MichaIng/DietPi #6087: Gogs - cannot connect to it after install

holocronology, 2023-01-22:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=13
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='not applicable'

- Distro version | bullseye 1
- Kernel version | Linux PiZero 5.15.84+ #1613 Thu Jan 5 11:58:09 GMT 2023 armv6l GNU/Linux
- SBC model | RPi Zero W (armv6l)

#### Additional Information (if applicable)
- Software title | Gogs
- Was the software title installed freshly or updated/migrated? Freshly
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Steps to reproduce

1. Install Gogs
2. browser to http://MY-IP:3000
3. browser reports "Unable to connect"

#### Expected behaviour

- Should being first time setup

#### Actual behaviour

- doesn't run

#### Extra details

- N/A

holocronology, 2023-01-22:
tried to reinstall and this:

#### Details:
- Date           | Sat Jan 21 19:33:21 EST 2023
- Bug report     | e2eb1e7a-0a4a-4ded-b365-a18bc5802b6d
- DietPi version | v8.13.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi Zero W (armv6l) (ID=1)
- Kernel version | `Linux PiZero 5.15.84+ #1613 Thu Jan 5 11:58:09 GMT 2023 armv6l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `mv /etc/gogs/.??* gogs`
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
mv: cannot stat '/etc/gogs/.??*': No such file or directory
```

MichaIng, 2023-01-22:
Let's stay at the other PR, this is all related to new packaging.

closed 2023-01-22 by MichaIng