MichaIng/DietPi #6088: Q: Best way to disable root login via SSH

Mausy5043, 2023-01-22:
### Question

CMIIW: I understand that DietPi is set-up by default to allow root-access via a remote SSH session.

IMHO allowing root login via SSH is a [not such a good idea from a security point of view](https://superuser.com/questions/1006267/why-should-i-really-disable-root-ssh-login). So, I always disable it on all my Linux machines. 

I've found that it is specifically *enabled* in `/etc/ssh/sshd_config.d/dietpi.conf`
```(bash)
$ more /etc/ssh/sshd_config.d/*.conf
# Added by DietPi:
PasswordAuthentication yes
PermitRootLogin yes
```

I can manually change this of course by commenting out this line.
But what would be the dietpi-way to disable SSH root access? Is there a command or setting available that I might have missed?

Joulinar, 2023-01-22:
> But what would be the dietpi-way to disable SSH root access? Is there a command or setting available that I might have missed?

There is an option inside `dietpi.txt` that can be used to disable password login via SSH.

https://github.com/MichaIng/DietPi/blob/dbfd976eb87e36550afcba5900e3dd93e64aedef/dietpi.txt#L223-L228

It can be set as follow.

```
G_CONFIG_INJECT 'SOFTWARE_DISABLE_SSH_PASSWORD_LOGINS=' 'SOFTWARE_DISABLE_SSH_PASSWORD_LOGINS=root' /boot/dietpi.txt
```

a reinstall of your SSH would be needed to activate the setting. Depending on your current SSH server, following will do.

```
dietpi-software reinstall 104 # reinstall Dropbear
dietpi-software reinstall 105 # reinstall OpenSSH 
```

MichaIng, 2023-01-22:
Another way:
```sh
/boot/dietpi/func/dietpi-set_software disable_ssh_password_logins root
```
This basically restores the OpenSSH default behaviour, permitting root login via SSH, but only via key authentication. To disable root login login completely: `PermitRootLogin no`, but at best with an own config override to assure it survives SSH server reinstalls:
```sh
echo 'PermitRootLogin no' > /etc/ssh/sshd_config.d/zz-disable-root-login.conf
systemctl restart ssh
```

closed 2023-01-29 by Mausy5043

centipedesquad, 2024-08-30:
> To disable root login login completely: `PermitRootLogin no`, but at best with an own config override to assure it survives SSH server reinstalls:
> 
> ```shell
> echo 'PermitRootLogin no' > /etc/ssh/sshd_config.d/zz-disable-root-login.conf
> systemctl restart ssh
> ```

By default, I don't have this location. I am getting this error:
`bash: /etc/ssh/sshd_config.d/zz-disable-root-login.conf: No such file or directory`
Do I just create this file?

Joulinar, 2024-08-30:
It is dedicated to OpenSSH. Therefore you don't have the file