MichaIng/DietPi #6089: NanoPi R5C | Serial console restart loop

MichaIng, 2023-01-22:
I notice one problem on the R5C :  
In syslog, every ten seconds, the following sequence appear : 

```
Jan 22 10:17:05 DietPi systemd[1]: Stopped Serial Getty on ttyFIQ0.
Jan 22 10:17:05 DietPi systemd[1]: Stopped Serial Getty on ttyFIQ0.
Jan 22 10:17:05 DietPi systemd[1]: Started Serial Getty on ttyFIQ0.
Jan 22 10:17:05 DietPi systemd[1]: Started Serial Getty on ttyFIQ0.
Jan 22 10:17:05 DietPi kernel: [23032.249773] ttyFIQ ttyFIQ0: tty_port_close_start: tty->count = 1 port count = 2
Jan 22 10:17:05 DietPi kernel: [23032.249773] ttyFIQ ttyFIQ0: tty_port_close_start: tty->count = 1 port count = 2
Jan 22 10:17:15 DietPi systemd[1]: serial-getty@ttyFIQ0.service: Succeeded.
Jan 22 10:17:15 DietPi systemd[1]: serial-getty@ttyFIQ0.service: Succeeded.
Jan 22 10:17:15 DietPi kernel: [23042.261266] ttyFIQ ttyFIQ0: tty_port_close_start: tty->count = 1 port count = 2
Jan 22 10:17:15 DietPi kernel: [23042.261266] ttyFIQ ttyFIQ0: tty_port_close_start: tty->count = 1 port count = 2
Jan 22 10:17:15 DietPi systemd[1]: serial-getty@ttyFIQ0.service: Scheduled restart job, restart counter is at 2245.
Jan 22 10:17:15 DietPi systemd[1]: serial-getty@ttyFIQ0.service: Scheduled restart job, restart counter is at 2245.
Jan 22 10:17:15 DietPi systemd[1]: Stopped Serial Getty on ttyFIQ0.
Jan 22 10:17:15 DietPi systemd[1]: Stopped Serial Getty on ttyFIQ0.
Jan 22 10:17:15 DietPi systemd[1]: Started Serial Getty on ttyFIQ0.
Jan 22 10:17:15 DietPi systemd[1]: Started Serial Getty on ttyFIQ0.
```
 This  makes syslog unreadable. I didn't find a proper way to stop this. I stopped services related to getty, but the thing continues.

_Originally posted by @Averell7 in https://github.com/MichaIng/DietPi/issues/6053#issuecomment-1399444317_

MichaIng, 2023-01-22:
Many thanks for reporting. I opened a new issue.

The serial console seems to be stuck in a restart loop. I'm not sure what is causing it to stop. I'm currently verifying that `ttyFIQ0` is the correct console device for the debug UART port, but that the kernel is reacting on it is basically an indicator that it is. We tweaked baud rate to be 1.5M, which is also correct according to FriendlyELEC's wiki.

However, if you do not use UART anyway, you can disable it:
```sh
systemctl disable --now serial-getty@ttyFIQ0
```

Averell7, 2023-01-22:
In case it would be useful for you, here is the result of the above command on my Nanopi R5C, after a reboot, and before I stopped serial-getty : 
[image]

MichaIng, 2023-01-22:
@Averell7 @Joulinar 
Let's keep NanoPi R5C discussion here. A release PR is not the right place to discuss device specific issues.

Still not sure why the serial console fails, it does not on my R5S, but I'll retest to be sure. However, since the debug UART header on R5S is not even extended by default (no pins soldered), I'm considering leaving it disabled. It'll still show boot logs when you attach a console (and in case it does really work) as of hardcoded `console=ttyFIQ0` kernel command-line argument, and it should even show a login console then, as systemd should start it automatically when recognising an attached console on a serial device.

The DebugFS related message you see on boot is known. We use the official FriendlyELEC kernel, which has it set like this. Don't ask me why...

Averell7, 2023-01-22:
I tested with the same SDcard on my R5S, the problem is the same. I wondered if it could come from the packages I installed, so I installed a new OS on a SDcard : the problem is the same, every 10 seconds. The version I use is DietPi_NanoPiR5S-ARMv8-Bullseye.7z, downloaded on the Dietpi site, January 18th (last friday). The only difference is that I do not have the lines doubled in rsyslog, but this problem is well known and comes surely from one of the packages I installed.

Averell7, 2023-01-22:
If the "NOTICE" is well known it is OK. 
A very small thing I noticed is that the second green led is on, even if no cable is connected to the second ethernet port. But may be it is intentional. The led works well, because, once I install my own handling system, it reacts correctly.

Averell7, 2023-01-22:
If I disable the service you mentioned, hte messages disappear, but just for this session. Once the device is restarted, the service is on again. The reason is probably explained in the UPDATE of the first post here : https://stackoverflow.com/questions/21596384/cannot-disable-systemd-serial-getty-service

MichaIng, 2023-01-22:
> rsyslog

Do not use `rsyslog` if you do not require permanent plaint text file logging for system logs. `journalctl` is the primary source for system logs with systemd.

> A very small thing I noticed is that the second green led is on, even if no cable is connected to the second ethernet port.

Yes, this is known: As long as there is not a cable connected once and then in case detached again, the state is not shown correctly. This is a limitation of this kernel feature and the same with FriendlyELEC's images (if I recall correctly).

> Once the device is restarted, the service is on again

Ah right, needs to be masked since systemd automatically starts the getty if boot outputs are passed to the console, even if nothing attached:
```sh
systemctl mask --now serial-getty@ttyFIQ0
```
Your Stack Overflow question indeed explains it well, the mechanism I referred in my last comment above. This however breaks the idea I had to disable but not mask it. `console=ttyFIQ0` is hardcoded into the boot config on R5S and R6S images, so it keeps starting if not masked.

That its the same on R5S is very strange indeed. We have three R5S in the team as noone of us is facing these errors, or am I wrong @Joulinar @StephanStS?

I cannot find something about possible hardware revisions released in the meantime which could make a difference.

StephanStS, 2023-01-23:
Regarding the logging feature also read our howto: https://dietpi.com/docs/usage/#how-to-use-the-logging-mechanism

MichaIng, 2023-01-23:
Would be great to find a solution. It is the same issue with NanoPi R2S, R4S, R5S, R6S. The `ledtrig-netdev` kernel module is used for this, but I failed to find a way to configure it so that it lids LEDs once a cable is connected, but without turning the LED on immediately at the same time, even without any cable connected.

Averell7, 2023-01-23:
Yes, I have seen the logging features of Dietpi and they are excellent. Presently, in development, I prefer a log on disk to be able to trace back several days before. 
About the leds : there is hope because I have a R5C which works perfectly, but I am still trying to understand the reason. Here is the full history : 

Install Dietpi 18.3.2  Device model NanoPi R5S/R5C
- The WAN and LAN leds are on, fixed green. 
- after installing Midnight Commander, it is strangely long to start : 10 seconds. After that it runs normally, with an exception : if I want to edit (F4) I have again a delay of 10 seconds. The reason is unknown. 
- I tried on a R5S, I have exactly the same phenomenons. 

I installed my software on the R5C, and after that the leds were working perfectly, and the delay with mc has disappeared. I made a slight change in udev to replace "lan2" (adapted to R5S) by "wlan0" (adapted to R5C), and now the three green leds work fine. fixed when no traffic, blinking when there is traffic, and black if I disconnect the cables. 

I tried the same thing on the R5S and installed all the packages, but not my software. Nothing changed, the two leds are on, and the delay for mc is still present. I will continue to install slowly to try to understand what is the thing which fixed the problem. I'll tell you what I find.

Averell7, 2023-01-23:
Oh, one thing : during install, when DietPI-Set is launched, there is a warning about a UART console running, and it is proposed to disable it. I remember that I said no, and this may be the cause of the problem mentioned earlier. (Same thing on R5S)
I had also an error during the same DietPi-Set : systemd.timesyncd.service could not be start. And the reason was clear, the link was broken. Prsently (in the R5C where the leds are OK) it is no longer broken, but the service is masked and another link is broken : dbus.org-freedeskto.timesync1.service. Probably this is unrelated.

Averell7, 2023-01-23:
I finished the installation on the R5S, and the leds WAN and LAN1 work fine now. New : LAN2 is always on, fixed. 
I guess this is the configuration of the network which made something. 
I deleted the files "eth0" and "eth1" in /etc/network/interfaces.d and rebooted, and now the three leds are fixed green. 
So I am sorry, what I did does not help. May be tweaking the udev file would do something ?

MichaIng, 2023-01-23:
> there is a warning about a UART console running, and it is proposed to disable it. I remember that I said no, and this may be the cause of the problem mentioned earlier.

The UART console is the serial console, which is doing these restart loop and spamming your logs. It shouldn't have such a system impact since it restarts every 10 seconds (?), but who knows.

> but the service is masked and another link is broken : dbus.org-freedesktop.timesync1.service

If the `systemd-timesyncd` service is masked, that link is surely broken because it points to this exact service. But it shouldn't be masked. Some 3rd party installers might mask it if they install other time sync daemons, like `ntp`, `chrony` and others. In this case, you should set time sync mode to custom in dietpi-config advanced options.

> I deleted the files "eth0" and "eth1" in /etc/network/interfaces.d and rebooted

These are not shipped with our images. If those files did exist, then they were created by you or a 3rd party installer. I can image that setting up the interfaces fixes the LEDs, but if those are not used, we do not want to set them up in the first place.

> So I am sorry, what I did does not help. May be tweaking the udev file would do something?

Currently it natively sets up the responsible kernel module. In theory, AFAIK it is possible to setup all network interfaces, then disable the once again which shall not be used. But this is a huge overhead I'd like to avoid. Looking for a native clean solution.

> change in udev to replace "lan2" (adapted to R5S) by "wlan0"

It is possible to let those LEDs blink on WiFi activity? That is awesome, I though I'd only work for Ethernet interfaces. Does it as well lid green when connected to an AP and go off when disconnecting from the AP?

Averell7, 2023-01-24:
About the WiFi : I do not know what would happen if connected to an AP, because my R5C is an AP. But in this mode, it is really interesting : if I stop hostapd, the led goes off. If I restart hostapd, it lights up again and it blinks on activity. That's really fine.

Averell7, 2023-01-24:
About the leds : there is a partial solution. I tried removing ATTR{link}="1" or setting it to 0 (the result seems to be the same) in the udev rule. In this case, the led blinks on traffic, but is out if no traffic. I don't like it too much, but it really prevents the leds to be on when the interface is not defined. Try it, and you will see if you prefer it like that or not. Or maybe we can find another attribute than "link" which would to a good job ? I do not know which attributes are available.

MichaIng, 2023-01-24:
> About the WiFi : I do not know what would happen if connected to an AP, because my R5C is an AP. But in this mode, it is really interesting : if I stop hostapd, the led goes off. If I restart hostapd, it lights up again and it blinks on activity. That's really fine.

Great. Since the LEDs are labelled, not something for a default setup, but great as additional documentation. I aim to write some blog post about the R*S/C devices where this info is a nice addition.

> I tried removing ATTR{link}="1"

This is the attribute which controls whether the LED is reacting to active links or not, i.e. cable connections, respectively a carrier signal, and in case of WiFi obviously either AP link or AP mode as well. If you set it to "0" it is and remains off. We want it to react to active links, but not being lid initially, if the link is not active. So currently the choices are only:
1. Leaving the link attribute off, so the LED only indicates activity but not an active link/cable/carrier state.
2. Setting the attribute on, so that the LED indicates those states, BUT it additionally is turned on directly, even if no link is active. It will turn off and show correct state indication once the interface has been set up once, but not if it is and remains off.

A device's attributes can be found in:
```sh
ls -l /sys/class/leds/wan_led/
```
respectively.

Averell7, 2023-01-24:
I made a very interesting test. I'll try to explain, if you cannot reproduce, tell me. 

- remove eth0 and eth1 configuration from /etc/network
- disconnect the device (reboot may not be sufficient) and restart
- The leds are green  (cable connected or not)
- send : echo > 0 /sys/class/leds/lan1_led/link     (and also for 2)
- The leds are black  (cables connected or not)
- configure ports eth0 and 1 in /etc/network
- ifup 
- The leds are now blinking on activity only
- send : echo > 1 /sys/class/leds/lan1_led/link     (and also for 2)
- The led are now green, blink on activity, and become black if I disconnect the cable

MichaIng, 2023-01-24:
Yes this is expected based on what I wrote above since you configure the network interfaces. The issues really only is if you do not (want to) configure the network interfaces, i.e. do not use those ports.

Averell7, 2023-02-12:
I always had a problem with a fresh installation of Dietpi for R5C/S : when the Dietpi-Set_Software is run in the beginning, it stops with a problem about systemd-timesyncd which is not installed. I went in the subshell and checked with apt, and the answer is that there is remaining configuration. So it has been uninstalled. I installed it and noticed that during the process it is indicated that ntp has been uninstalled. I do not like the tendency of systemd to do everything in the computer and I would prefer to keep ntp. Is it possible that the absence of systemd.timesyncd is accepted by DietPi ?

I continued and after selecting the software, new problem with :  getent hosts dns9.quad9.net  : exit code 2. 
Until now I have been unable to finish the first config. I sent a bug report under number : 

0636f6ef-e0ba-42f5-a076-63325fd8777c

MichaIng, 2023-02-12:
This is expected: `ntp` and `systemd-timesyncd` are both time sync daemons, doing he same thing. The `ntp` package (respectively `ntpsec` since Bookworm) has a conflict declared on all other time sync daemons, so that it is forcefully removed when `systemd-timesyncd` is installed and the other way round.

You have a lot of manual software installed, so I'm not sure whether you installed `ntp` intentionally? If so, disable DietPi's time sync attempts via `dietpi-config` > Advanced Options > Time sync mode > 0: Custom.

Otherwise, if you do not need high precision system time, leave `ntp` removed.

Joulinar, 2023-02-12:
Next it llooks like a DNS issue. Can you check DNS resolution is working fine.

`getent hosts dns9.quad9.net`

MichaIng, 2023-03-18:
Is here any issue left? Next DietPi version will ship a kernel upgrade, which might solve the one or the other quirk, btw.

Averell7, 2023-03-19:
I think in the last version I tested most issues where fixed. Just one strange thing remains. 
On a new install, I install just mc. It works, but when I open it there is a 10 seconds delay. And when I want to edit with F4, there is again a 10 seconds delay (I use the internal editor). Otherwise it works fine. I have no idea of the cause, and no idea what to investigate.

MichaIng, 2023-03-19:
I just tested it on my R5S and it works well there. Running `mc` opens it immediately and hitting `F4` on a text file, selecting `/usr/bin/mcedit` (that is what you mean with "internal editor"?) all has immediate result, no 10 seconds delay.

Did you try this via SSH or local console? Any error messages after closing `mc` or in `dmesg`?

Averell7, 2023-03-28:
I have been long to answer because I wanted to do a full new test. It worked fine and I didn't find the problems mentioned earlier. Mc  is running fine. 
So nothing to say about this version.

Averell7, 2023-03-28:
I have just a question : I need to build a kernel module we have written.  But with this strange 5.10.110 kernel, I cannot use the headers provided by apt-get. I got the source from cdn.linux.org, but I need the config file normally found in /boot (something like config-5.10.0-16-arm64). But on the Nanopi there is nothing significant in /boot, and I was unable to find this file. Do you know where I can find it ?

MichaIng, 2023-03-28:
`hostnamectl` is a systemd tool to control the `systemd-hostnamed` service, which is not used on DietPi. Use `hostname` to show the hostname (the command also has some options), or `cat /etc/hostname`, which works both on any UNIX system.

There are sadly no kernel headers for this FriendlyELEC vendor kernel. Their images had a headers package included, but this did not work/match the installed kernel. The kernel is on a dedicated partition, not in the filesystem. I can recheck when I find time, but for now there is no way to compile kernel modules for it.

MichaIng, 2023-03-28:
No you need to matching headers. Those from Debian and `cdn.linux.org` cannot work since Rockhip vendor kernel with FriendlyELEC patches is used. There are the sources: https://github.com/friendlyarm/kernel-rockchip/tree/nanopi5-v5.10.y_opt
But I am not sure which exact commit the kernel was built with.

Not sure what you would need the config file for, if you mean the kernel built time configuration? This can be always found in at `/proc/config.gz`. If you mean the cmdline, that is located on some other partition as well. This is the reason those NanoPi R5/R6 have so many partitions, as every part of bootloader, kernel and config has its own partition. They cannot be mounted since those are no filesystems, but the related files are flashed on the parts in raw format. Don't ask me why they go this awkward way. As fast as those SBCs have proper mainline kernel support, I'll create a proper single partition image with headers and everything.

Averell7, 2023-03-29:
Thanks for the clear answer. I will have to wait. It is strange that a development card "hides" the files necessary to build. In case I have news, I'll tell you.

MichaIng, 2023-04-03:
I'll mark this as closed. Feel free to reopen if required. I'll inform you once we have a mainline Linux image.

closed 2023-04-03 by MichaIng

MichaIng, 2023-04-22:
@Averell7 
I'm currently updating the Ethernet LED udev rules for the NanoPi R series: #6331
- Is it correct that on R5C the WAN adapter is the `eth0` interface and the LAN adapter is the `eth1` interface?
- The LAN LED can be configured via `/sys/class/leds/lan1_led`, i.e. this directory exists?

Averell7, 2023-04-23:
Yes, wan is eth0 on the R5C (as on R5S). About the leds the directory has the same content on both, with the additional "input3" on my R5S, but may be the version of the software (presently Friendly Elec Debian11) is slightly different. See both images (first is R5S, second R5C. In R5C, lan2_led is the WiFi : 
[image]
[image]

MichaIng, 2023-04-23:
> Yes, wan is eth0 on the R5C

Okay great, thanks for the info.

> About the leds the directory has the same content on both, with the additional "input3" on my R5S

That should show up as fast as a keyboard is attached. We use the same kernel as FriendlyELEC, so those should behave the same.

> In R5C, lan2_led is the WiFi

Interesting. On NanoPi R6C, the `lan2_led` does not exist, instead an additional `user_led`. A good idea to attach this to `wlan0` if one does not have another specific use. Was this set by default in your case (with the FriendlyELEC Debian image, if I understand correctly)? The `leds.service` found on FriendlyELECs image enforces `lan2_led` linked to `eth2`, if present, as far as I could see. This was on the RK3588 image, but the script looks pretty generic (and even attaches LEDs falsely on R6S).

I just recognised on my R6C, that the WAN and LAN (1) LEDs are now set to `netdev` trigger OOTB, without any related udev rule or service/script doing this. Must have been a change in the device tree since last kernel upgrade. Strangely even FriendlyELEC still ships their images with mentioned service. However, #6331 still remains relevant to fix the issue that the Ethernet LEDs remain on if the interface is disabled/unused.

Averell7, 2023-04-23:
I wrote : "In R5C, lan2_led is the WiFi". 
That's wrong in Friendly Elec Debian 11. 
It was true in Dietpi Debian 11 (but maybe I tweaked it, I do not remember), and I told you it was even blinking on activity. 
In Friendly Elec Debian, this led is never on. The two others are at startup if no cable is connected, and on when I connect a cable, but they do  not blink on activity. But I see my version is getting old : 
[image]

MichaIng, 2023-04-23:
> It was true in Dietpi Debian 11 (but maybe I tweaked it, I do not remember), and I told you it was even blinking on activity.

We ship the image with `lan2_led` linked to `eth2`, and indeed with `rx`/`tx` attributes enabled, i.e. blinking on activity. If there is not somehow a smart automatic adjustment of `eth2` to `wlan0`, then I guess you tweaked it (quite reasonably, did the same with my R6S `user_led` now 🙂).

> The two others are off at startup if no cable is connected, and on when I connect a cable

Interesting. Probably the interfaces are set "up" then on their Debian image. Will test it ones a new set of SD cards arrived. It did already according to shipment tracking, but my mailbox was empty 😠, hopefully A***** support solves this soon.

> But I see my version is getting old :

That is the latest Rockchip vendor kernel, all these manufacturers use: https://github.com/rockchip-linux/kernel/tree/develop-5.10
And as long as there is no proper mainline kernel support (done by mostly volunteers via backwards engineering and stuff), we have to use it.