MichaIng/DietPi #6091: Upgrade to Bullseye breaks 'motion'

nzbaxterman, 2023-01-23:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=13
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='not applicable'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
bullseye 1

- Kernel version | `uname -a`
Linux Tarrant 4.14.24+ #1097 Mon Mar 5 16:16:18 GMT 2018 armv6l GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
RPi B+ (armv6l)

- Power supply used | (EG: 5V 1A RAVpower)
5V 2A

- SD card used | (EG: SanDisk ultra)
Sandisk Ultra

#### Additional Information (if applicable)
- Software title | motion

- Was the software title installed freshly or updated/migrated?
Dietpi was upgraded from Jessie to Buster to Bullseye as per https://dietpi.com/docs/usage/#how-to-upgrade-to-buster

- Can this issue be replicated on a fresh installation of DietPi?
Not tried

- Bug report ID | `echo $G_HW_UUID`
e990ef07-5339-433c-a8bc-e32b6e8a98ff

#### Steps to reproduce

1. Upgrade to Bullseye
2. Attempt to start "motion"

#### Expected behaviour

- motion should start without error

#### Actual behaviour

- First I got errors with 
- error while loading shared libraries: libvchiq_arm.so: cannot open shared object file: No such file or directory
I found there were issues with not just libvchiq, but all these libraries being installed in /opt/vc/lib:
/opt/vc/lib/libvchiq_arm.so
/opt/vc/lib/libvcos.so
/opt/vc/lib/libmmal_core.so
/opt/vc/lib/libmmal_util.so
/opt/vc/lib/libmmal_vc_client.so
/opt/vc/lib/libbcm_host.so
/opt/vc/lib/libvcsm.so

rather than in /usr/lib/arm-linux-gnueabihf/ where motion expected them to be.

So (rather crudely) I tried sym linking them in /usr/lib/arm-linux-gnueabihf/:
cd /usr/lib/arm-linux-gnueabihf/
ln -s /opt/vc/lib/libvchiq_arm.so libvchiq_arm.so.0
ln -s /opt/vc/lib/libvcos.so libvcos.so.0
ln -s /opt/vc/lib/libmmal_core.so libmmal_core.so.0
ln -s /opt/vc/lib/libmmal_util.so libmmal_util.so.0
ln -s /opt/vc/lib/libmmal_vc_client.so libmmal_vc_client.so.0
ln -s /opt/vc/lib/libbcm_host.so libbcm_host.so.0
ln -s /opt/vc/lib/libvcsm.so libvcsm.so.0

which (progressively) eliminated the errors when launching motion.

But now I'm stuck with this error:
motion: symbol lookup error: /usr/lib/arm-linux-gnueabihf/libavcodec.so.58: undefined symbol: vcsm_init_ex

Any suggestions?

MichaIng, 2023-01-23:
Yes, it needs to be updated indeed: https://github.com/Motion-Project/motion/releases

I guess you installed it from there, right? The (older) package from the Raspbian Bullseye repo should work, of course.

Joulinar, 2023-01-23:
the system has been upgraded from Jessie. Maybe this is causing side effects as this is quite a heavy bump

MichaIng, 2023-01-23:
Ah, makes sense and explains the old kernel. The motion issue is however mismatching package. I just checked and there is no `pi_bullseye` variant. If I'm not mistaken, the `pi_buster` one works. If not, we need to ask the project maintainers to build it for Raspbian Bullseye.

nzbaxterman, 2023-01-23:
> Yes, it needs to be updated indeed: https://github.com/Motion-Project/motion/releases
> 
> I guess you installed it from there, right? The (older) package from the Raspbian Bullseye repo should work, of course.

No, I installed it through apt install, once on Bullseye.

MichaIng, 2023-01-23:
> No, I installed it through apt install, once on Bullseye.

Are you sure? This would be a bug in Raspbian (or Debian or both) repo then. Can you please show:
```sh
which motion
dpkg -l motion
apt policy motion
```

MichaIng, 2023-01-23:
> Tried the bullseye package:

This is not gonna work anyway. You must use the `pi_` prefixed package. The others are not compatible with your ARMv6 SoC.

Please test with the `pi_buster` package or otherwise the one from Raspbian repo:
```sh
apt install --reinstall --allow-downgrades motion=4.3.2-1
```

nzbaxterman, 2023-01-24:
Same problem I'm afraid:
dietpi@Tarrant:~$ sudo apt install --reinstall --allow-downgrades motion=4.3.2-1
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Suggested packages:
  default-mysql-client postgresql-client
The following NEW packages will be installed:
  motion
0 upgraded, 1 newly installed, 0 to remove and 5 not upgraded.
Need to get 481 kB of archives.
After this operation, 2,171 kB of additional disk space will be used.
Get:1 http://mirror.2degrees.nz/raspbian/raspbian bullseye/main armhf motion armhf 4.3.2-1 [481 kB]
Fetched 481 kB in 2s (251 kB/s)
Preconfiguring packages ...
Selecting previously unselected package motion.
(Reading database ... 51701 files and directories currently installed.)
Preparing to unpack .../motion_4.3.2-1_armhf.deb ...
Unpacking motion (4.3.2-1) ...
Setting up motion (4.3.2-1) ...
Installing new version of config file /etc/motion/camera1-dist.conf ...
Installing new version of config file /etc/motion/camera2-dist.conf ...
Installing new version of config file /etc/motion/camera3-dist.conf ...
Installing new version of config file /etc/motion/camera4-dist.conf ...

Configuration file '/etc/motion/motion.conf'
 ==> Modified (by you or by a script) since installation.
 ==> Package distributor has shipped an updated version.
 ==> Keeping old config file as default.
motion.service is a disabled or a static unit not running, not starting it.
dietpi@Tarrant:~$
dietpi@Tarrant:~$
dietpi@Tarrant:~$ motion
motion: symbol lookup error: /usr/lib/arm-linux-gnueabihf/libavcodec.so.58: undefined symbol: vcsm_init_ex

nzbaxterman, 2023-01-24:
And same if I install it from:
wget https://github.com/Motion-Project/motion/releases/download/release-4.5.1/pi_buster_motion_4.5.1-1_armhf.deb
apt remove motion
dpkg -i pi_buster_motion_4.5.1-1_armhf.deb

Maybe there's now a problem with libavcodec?
root@Tarrant:/usr/lib/arm-linux-gnueabihf# apt-cache policy libavcodec58
libavcodec58:
  Installed: 7:4.3.5-0+deb11u1+rpt3
  Candidate: 7:4.3.5-0+deb11u1+rpt3
  Version table:
 *** 7:4.3.5-0+deb11u1+rpt3 500
        500 https://archive.raspberrypi.org/debian bullseye/main armhf Packages
        100 /var/lib/dpkg/status
     7:4.3.5-0+deb11u1 500
        500 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf Packages

MichaIng, 2023-01-24:
So it is indeed a bug in the FFmpeg packages provided by the Raspberry Pi foundation. They were built against a wrong version of libraspberrypi0 with the legacy file paths and symbols which have been removed. We need to report this here: https://github.com/RPi-Distro/ffmpeg/issues

Can you test with the Raspbian build:
```sh
apt install --reinstall --allow-downgrades 'libavcodec58=7:4.3.5-0+deb11u1'
```
You might need to add further packages to the list with the same version enforced, to solve dependency conflicts.

nzbaxterman, 2023-01-25:
All fixed - after downgrading packages and resolving dependency conflicts.

sudo apt install -y --reinstall --allow-downgrades 'libavdevice58=7:4.3.5-0+deb11u1' 'libavformat58=7:4.3.5-0+deb11u1' 'libswscale5=7:4.3.5-0+deb11u1' 'libavfilter7=7:4.3.5-0+deb11u1' 'libchromaprint1' 'libpostproc55=7:4.3.5-0+deb11u1'

This also then automatically identified 'libchromaprint1 armhf 1.5.0-2' and motion 4.5.1-1'.

THANKS FOR YOUR HELP

Oscilllator, 2024-03-29:
fwiw I was getting this same error when running ffmpeg (just that, not even actually converting anything) and running this command:
```sudo apt install --reinstall libraspberrypi0 libraspberrypi-bin```
Fixed it. ChatGPT special, buyer beware.

MichaIng, 2024-03-29:
Why ever different `libraspberrypi` packages were installed on your system. Probably you did an `rpi-update`? However, the original issue has long been solved, I'll mark this hence as closed.

closed 2024-03-29 by MichaIng