MichaIng/DietPi #6093: Nanopi NEO | No soundcards found

eurya, 2023-01-25:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=13
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='not applicable'

- Distro version | bullseye
- Kernel version | Linux DietPi 5.15.89-sunxi #22.11.4 SMP Mon Jan 23 21:58:30 UTC 2023 armv7l GNU/Linux
- SBC model | NanoPi NEO (armv7l)
- Power supply used | 5V 1A 
- SD card used | SanDisk ultra

#### Steps to reproduce

1. Install latest Dietpi image for Nanopi NEO (version  DietPi v8.13.2)
2. Enable sound in dietpi-config
3. No soundcards listed or found: 
```
root@DietPi:~# aplay -l
aplay: device_list:274: no soundcards found...
```

#### Expected behaviour
- sound board listed to play sound

#### Actual behaviour
No soundcards listed or found: 
```
root@DietPi:~# aplay -l
aplay: device_list:274: no soundcards found...
```

MichaIng, 2023-01-25:
Please add `analog-codec` space separated to the end of the `overlays=` line in `/boot/dietpiEnv.txt`, reboot and try again.

eurya, 2023-01-25:
> Please add `analog-codec` space separated to the end of the `overlays=` line in `/boot/dietpiEnv.txt`, reboot and try again.

Thanks for the fast reply, after adding "analog-codec": 
```
root@DietPi:~# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Codec [H3 Audio Codec], device 0: CDC PCM Codec-0 [CDC PCM Codec-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

MichaIng, 2023-01-25:
Great. You should now be able to set it up as default sound card, e.g. via `dietpi-config`.

MichaIng, 2023-01-26:
Possibly a limitation of the driver with mainline kernel. Can you try to adjust the CPU governor, set it to `powersave` to stop voltage/power changes? Probably this affects the analogue signal negatively. Also how do you play sound? Can you try to adjust audio parameters, like sampling rate? Probably the sound card supports more than what the software is passing by default.

MichaIng, 2023-02-19:
@eurya 
Probably tweaking IRQ (hardware interrupt) affinities enhances sound quality. Could you show the output of:
```sh
cat /proc/interrupts
```

MichaIng, 2023-06-28:
Implemented with: #6447

closed 2023-06-28 by MichaIng