MichaIng/DietPi #6094: NanoPiNeo: Permission denied, cannot open /dev/ttyS1

MaikB85, 2023-01-25:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=13
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='not applicable'
- Distro version | bullseye
- Kernel version | Linux nanopineo 5.15.89-sunxi #22.11.4 SMP Mon Jan 23 21:58:30 UTC 2023 armv7l GNU/Linux
- SBC model | NanoPi NEO (armv7l)
- Power supply used | 5V 1,5A
- SD card used | SanDisk ultra

#### Steps to reproduce
1. Enabled Serial Port 1 with dietpi-config
2. Node-Red Error:
sudo journalctl -u node-red
... node-red[474]: 25 Jan 17:50:59 - [error] [serialconfig:27dba1b175b313e6] Serieller Port /dev/ttyS1 FEHLER: Error: Error: Permission denied, cannot open /dev/ttyS1 ......

#### Expected behaviour
NodeRed has no permission for Serial Port 1

#### Extra details
dietpi@nanopineo:~$ ls -la /dev/ttyS1
crw--w---- 1 root tty 4, 65 25. Jan 17:50 /dev/ttyS1

Tried to set permissions but after logout this resets to permissions above.

MichaIng, 2023-01-25:
Thanks for reporting.

`crw--w---- 1 root tty` are the default permissions it has, or is this how you changed it? On `x86_64` I actually see `crw-rw---- 1 root dialout`. This is coming from a udev rule:
```sh
/usr/lib/udev/rules.d/50-udev-default.rules:KERNEL=="tty[A-Z]*[0-9]|ttymxc[0-9]*|pppox[0-9]*|ircomm[0-9]*|noz[0-9]*|rfcomm[0-9]*", GROUP="dialout"
```
There are some rules which match your permissions, but those do not match the device name, do they?
```sh
/usr/lib/udev/rules.d/50-udev-default.rules:SUBSYSTEM=="tty", KERNEL=="tty[0-9]*", GROUP="tty", MODE="0620"
/usr/lib/udev/rules.d/50-udev-default.rules:SUBSYSTEM=="tty", KERNEL=="sclp_line[0-9]*", GROUP="tty", MODE="0620"
/usr/lib/udev/rules.d/50-udev-default.rules:SUBSYSTEM=="tty", KERNEL=="ttysclp[0-9]*", GROUP="tty", MODE="0620"
/usr/lib/udev/rules.d/50-udev-default.rules:SUBSYSTEM=="tty", KERNEL=="3270/tty[0-9]*", GROUP="tty", MODE="0620"
```
Can you show the output of:
```sh
ls -l /sys/class/tty
```

MaikB85, 2023-01-25:
> Thanks for reporting.
> 
> `crw--w---- 1 root tty` are the default permissions it has, or is this how you changed it? On `x86_64` I actually see `crw-rw---- 1 root dialout`. This is coming from a udev rule:
> 
> ```shell
> /usr/lib/udev/rules.d/50-udev-default.rules:KERNEL=="tty[A-Z]*[0-9]|ttymxc[0-9]*|pppox[0-9]*|ircomm[0-9]*|noz[0-9]*|rfcomm[0-9]*", GROUP="dialout"
> ```
> 
> There are some rules which match your permissions, but those do not match the device name, do they?
> 
> ```shell
> /usr/lib/udev/rules.d/50-udev-default.rules:SUBSYSTEM=="tty", KERNEL=="tty[0-9]*", GROUP="tty", MODE="0620"
> /usr/lib/udev/rules.d/50-udev-default.rules:SUBSYSTEM=="tty", KERNEL=="sclp_line[0-9]*", GROUP="tty", MODE="0620"
> /usr/lib/udev/rules.d/50-udev-default.rules:SUBSYSTEM=="tty", KERNEL=="ttysclp[0-9]*", GROUP="tty", MODE="0620"
> /usr/lib/udev/rules.d/50-udev-default.rules:SUBSYSTEM=="tty", KERNEL=="3270/tty[0-9]*", GROUP="tty", MODE="0620"
> ```
> 
> Can you show the output of:
> 
> ```shell
> ls -l /sys/class/tty
> ```

`dietpi@nanopineo:~$ ls -l /sys/class/tty
insgesamt 0
lrwxrwxrwx 1 root root 0 25. Jan 17:50 console -> ../../devices/virtual/tty/console
lrwxrwxrwx 1 root root 0 25. Jan 17:50 ptmx -> ../../devices/virtual/tty/ptmx
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty -> ../../devices/virtual/tty/tty
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty0 -> ../../devices/virtual/tty/tty0
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty1 -> ../../devices/virtual/tty/tty1
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty10 -> ../../devices/virtual/tty/tty10
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty11 -> ../../devices/virtual/tty/tty11
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty12 -> ../../devices/virtual/tty/tty12
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty13 -> ../../devices/virtual/tty/tty13
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty14 -> ../../devices/virtual/tty/tty14
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty15 -> ../../devices/virtual/tty/tty15
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty16 -> ../../devices/virtual/tty/tty16
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty17 -> ../../devices/virtual/tty/tty17
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty18 -> ../../devices/virtual/tty/tty18
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty19 -> ../../devices/virtual/tty/tty19
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty2 -> ../../devices/virtual/tty/tty2
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty20 -> ../../devices/virtual/tty/tty20
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty21 -> ../../devices/virtual/tty/tty21
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty22 -> ../../devices/virtual/tty/tty22
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty23 -> ../../devices/virtual/tty/tty23
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty24 -> ../../devices/virtual/tty/tty24
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty25 -> ../../devices/virtual/tty/tty25
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty26 -> ../../devices/virtual/tty/tty26
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty27 -> ../../devices/virtual/tty/tty27
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty28 -> ../../devices/virtual/tty/tty28
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty29 -> ../../devices/virtual/tty/tty29
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty3 -> ../../devices/virtual/tty/tty3
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty30 -> ../../devices/virtual/tty/tty30
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty31 -> ../../devices/virtual/tty/tty31
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty32 -> ../../devices/virtual/tty/tty32
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty33 -> ../../devices/virtual/tty/tty33
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty34 -> ../../devices/virtual/tty/tty34
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty35 -> ../../devices/virtual/tty/tty35
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty36 -> ../../devices/virtual/tty/tty36
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty37 -> ../../devices/virtual/tty/tty37
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty38 -> ../../devices/virtual/tty/tty38
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty39 -> ../../devices/virtual/tty/tty39
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty4 -> ../../devices/virtual/tty/tty4
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty40 -> ../../devices/virtual/tty/tty40
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty41 -> ../../devices/virtual/tty/tty41
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty42 -> ../../devices/virtual/tty/tty42
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty43 -> ../../devices/virtual/tty/tty43
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty44 -> ../../devices/virtual/tty/tty44
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty45 -> ../../devices/virtual/tty/tty45
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty46 -> ../../devices/virtual/tty/tty46
[... 13 lines cut ...]
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty59 -> ../../devices/virtual/tty/tty59
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty6 -> ../../devices/virtual/tty/tty6
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty60 -> ../../devices/virtual/tty/tty60
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty61 -> ../../devices/virtual/tty/tty61
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty62 -> ../../devices/virtual/tty/tty62
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty63 -> ../../devices/virtual/tty/tty63
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty7 -> ../../devices/virtual/tty/tty7
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty8 -> ../../devices/virtual/tty/tty8
lrwxrwxrwx 1 root root 0 25. Jan 17:50 tty9 -> ../../devices/virtual/tty/tty9
[... cut ...]
lrwxrwxrwx 1 root root 0 25. Jan 17:50 ttyS5 -> ../../devices/platform/serial8250/tty/ttyS5
lrwxrwxrwx 1 root root 0 25. Jan 17:50 ttyS6 -> ../../devices/platform/serial8250/tty/ttyS6
lrwxrwxrwx 1 root root 0 25. Jan 17:50 ttyS7 -> ../../devices/platform/serial8250/tty/ttyS7`

when I disable Serial1 and reboot:
`dietpi@nanopineowz:~$ ls -la /dev/ttyS1
crw-rw---- 1 root dialout 4, 65 25. Jan 18:46 /dev/ttyS1`

after enable serial 1 again and reboot:
`dietpi@nanopineo:~$ ls -la /dev/ttyS1
crw--w---- 1 root tty 4, 65 25. Jan 18:48 /dev/ttyS1`

MichaIng, 2023-01-25:
Ah, you mean when you enable/disable the login console on this serial device? Indeed, this triggers the change:
```console
root@dietpi:~# systemctl start serial-getty@ttyS0
root@dietpi:~# l /dev/ttyS0
crw-rw---- 1 root dialout 4, 64 Jan 25 18:55 /dev/ttyS0
root@dietpi:~# htop
root@dietpi:~# l /dev/ttyS*
crw-rw---- 1 root dialout 4, 64 Jan 25 18:55 /dev/ttyS0
crw-rw---- 1 root dialout 4, 65 Jan 22 18:33 /dev/ttyS1
crw-rw---- 1 root dialout 4, 66 Jan 22 18:33 /dev/ttyS2
crw-rw---- 1 root dialout 4, 67 Jan 22 18:33 /dev/ttyS3
root@dietpi:~# systemctl stop serial-getty@ttyS0.service
root@dietpi:~# l /dev/ttyS*
crw--w---- 1 root tty     4, 64 Jan 25 18:55 /dev/ttyS0
crw-rw---- 1 root dialout 4, 65 Jan 22 18:33 /dev/ttyS1
crw-rw---- 1 root dialout 4, 66 Jan 22 18:33 /dev/ttyS2
crw-rw---- 1 root dialout 4, 67 Jan 22 18:33 /dev/ttyS3
root@dietpi:~# systemctl stop serial-getty@ttyS1.service
root@dietpi:~# l /dev/ttyS*
crw--w---- 1 root tty     4, 64 Jan 25 18:55 /dev/ttyS0
crw-rw---- 1 root dialout 4, 65 Jan 22 18:33 /dev/ttyS1
crw-rw---- 1 root dialout 4, 66 Jan 22 18:33 /dev/ttyS2
crw-rw---- 1 root dialout 4, 67 Jan 22 18:33 /dev/ttyS3
root@dietpi:~# systemctl start serial-getty@ttyS1.service
root@dietpi:~# l /dev/ttyS*
crw--w---- 1 root tty     4, 64 Jan 25 18:55 /dev/ttyS0
crw--w---- 1 root tty     4, 65 Jan 22 18:33 /dev/ttyS1
crw-rw---- 1 root dialout 4, 66 Jan 22 18:33 /dev/ttyS2
crw-rw---- 1 root dialout 4, 67 Jan 22 18:33 /dev/ttyS3
root@dietpi:~# systemctl stop serial-getty@ttyS1.service
root@dietpi:~# systemctl start serial-getty@ttyS3.service
root@dietpi:~# l /dev/ttyS*
crw--w---- 1 root tty     4, 64 Jan 25 18:55 /dev/ttyS0
crw--w---- 1 root tty     4, 65 Jan 22 18:33 /dev/ttyS1
crw-rw---- 1 root dialout 4, 66 Jan 22 18:33 /dev/ttyS2
crw--w---- 1 root tty     4, 67 Jan 22 18:33 /dev/ttyS3
root@dietpi:~# systemctl stop serial-getty@ttyS3.service
root@dietpi:~# l /dev/ttyS*
crw--w---- 1 root tty     4, 64 Jan 25 18:55 /dev/ttyS0
crw--w---- 1 root tty     4, 65 Jan 22 18:33 /dev/ttyS1
crw-rw---- 1 root dialout 4, 66 Jan 22 18:33 /dev/ttyS2
crw--w---- 1 root tty     4, 67 Jan 22 18:33 /dev/ttyS3
```
Don't ask my why the change on `ttyS0` happened only after I stopped the service. But in all other cases it happens consistently. Makes sense, since this changes the function of that serial port to be an actual login console.

Is this what you actually need? Note that `dietpi-config` does not enable/disable the serial device/port itself, but only triggers boot output and the login prompt on those. And the latter enforces these permissions. It's actually not the service, but the `agetty` command itself which does this, as further tests reveal.

MaikB85, 2023-01-25:
Ah okay I think I misunderstood this function. Thought this would disable Serial port. 

Well after disable the console for serial1 the error message for permissions in nodered is gone, but now I get message:

`Jan 25 19:07:23 nanopineowz node-red[474]: 25 Jan 19:07:23 - [error] [serialconfig:27dba1b175b313e6] Serieller Port /dev/ttyS1 FEHLER: Error: Error: Input/output error setting custom baud rate of 115200`

MaikB85, 2023-01-25:
Okay this can be closed. I have to enable the uart1 devicetreeoverlay in /boot/dietpiEnv.txt

now it works.

closed 2023-01-25 by MaikB85

MichaIng, 2023-01-25:
Toggling the overlays is indeed something we need to implement. Interesting that the device is present even without the overlay enabled.