MichaIng/DietPi #6095: NanoPi R6S won't boot

fakemanhk, 2023-01-25:
I downloaded this [image](https://dietpi.com/downloads/images/DietPi_NanoPiR6S-ARMv8-Bullseye.7z) and flashed on 2 different SD cards (Kingston ValueSelect 32GB & SanDisk Ultra 64GB), my R6S doesn’t boot (no display from HDMI, waited for an hour), to confirm SD cards are good, I use them to flash the image for NanoPi R2S which works on it, flashing with FriendlyElec’s image (any of the R6S image) also bootable. To avoid the internal eMMC causing boot issue, I used FriendlyElec Ubuntu image to boot up and wiped boot record of eMMC to make sure it can only boot from SD card, but it still doesn’t work for me. Image flashing I used both Windows + Rufus / Linux Mint + balenEtcher, results are the same.

Tested with NanoPi's latest beta Debian Bullseye image, it's also booting, so definitely not hardware problem.

One thing I am not sure, looking at the SD card for R2S, there is only 1 partition, while the one for R6S has 8 (with only 1 mountable under linux), is that normal?

MichaIng, 2023-01-25:
> to make sure it can only boot from SD card,

If it does not boot, then obviously neither from eMMC 😉. It does boot from SD card as long as the bootloader on eMMC is the one from FriendlyELEC (which is the same we use for our image).

> One thing I am not sure, looking at the SD card for R2S, there is only 1 partition, while the one for R6S has 8 (with only 1 mountable under linux), is that normal?

Yes, R2S uses mainline kernel and mainline U-Boot which can boot from a single MBR partition. R6S uses FriendlyELEC's vendor kernel and bootloader, which requires this GPT partitioning.

We have quite a bunch of R6S successfully booting, never boot issues, so it's strange, also since we use FriendlyELEC's image, just replacing the root partition with a clean one, which doesn't affect early boot stages. I'll retest on my R6S, since the recent image was created just a week ago.

MichaIng, 2023-01-25:
1. Took https://dietpi.com/downloads/images/DietPi_NanoPiR6S-ARMv8-Bullseye.7z
2. SHA256: `b3fbc82491bd9533964666fe317a87b6ad647a06ff6154063e63c741d2384994`
3. Flashed from Windows via https://github.com/pbatard/rufus/releases/download/v3.21/rufus-3.21p.exe
4. to Samsung EVO Plus 128 GiB
5. Inserted, booted well: It took about 10 seconds on first boot before the HDMI screen showed more than a blinking underscore. FriendlyELEC's boot config obviously passes the `quiet` flag or `loglevel=1` to the kernel.

On first boot I powered it via USB hub powered via port from my host PC. This worked well until the step where I was choosing to disable the serial console. It then crashed due to insufficient power/voltage, expectedly. Doing the same via dedicated 3.8A USB PSU worked well and stable.

Based on that, since the screen doesn't show up something during first boot setup, and we do quite some stuff there above what FriendlyELEC's image does, it could be an insufficient PSU as well in your case. You can actually observe what happens via LEDs:
- When the red LED is lid, it is at best in bootloader stage.
- When the red LED starts blinking, it started the kernel.
- When it switches from lid to blinking to lid etc, it is in a reboot loop. This is what happened in my case with USB hub power, until I switched to a good PSU.
- When one of the green LEDs is lid, the kernel loaded already the Ethernet adapters, which means quite good boot stage. Those green LEDs turn off again sometimes, sometimes not, it's a bit unreliable, sadly. It shows cable connection state only if a cable was connected at least once in a boot session and the interface was configured (set "up").
- When green LEDs are blinking, it actually exchanges data with this port, i.e. the Ethernet access is up, which means kernel has fully loaded and first boot setup steps have finished.

One thing you need to take care of: If you use Ethernet and DHCP, attach the Internet facing cable to the WAN port. This is the `eth0` interface which is configured at first boot. If no Internet access can be established on first boot, the login prompt appears about 90 seconds later, which is the default timeout of systemd to wait for a missing dependency unit until proceeding.

Joulinar, 2023-01-26:
very same request has been opened on our forum https://dietpi.com/forum/t/dietpi-doesnt-boot-on-nanopi-r6s/15722

fakemanhk, 2023-01-26:
Yes I observed that only LAN2 port has signal during boot, the other 2 doesn't light up at all, but I monitored my DHCP server which doesn't show any DHCP request so pretty sure there is problem (also the monitor always showing "no signal").

BTW I just found 2 things that can make it working now but I have no idea why:

1.  I used same SD card, but used DD to wipe it again and used balenEtcher to flash, now during boot I can see DHCP request from R6S, but still no display from HDMI.
2.  Since I moved to another desk, I used another monitor and there's display now! I have no idea why no HDMI on my original HDMI monitor (using same R6S with FriendlyElec image was fine and with GUI), is there anything I can do on this? I think text console shouldn't be picky about monitor?

MichaIng, 2023-01-26:
> I think text console shouldn't be picky about monitor?

It doesn't matter what is shown on the screen, but resolution, cable length and signal strength play a role. Since we use the original kernel and bootloader of FriendlyELEC, and there have not been any significant commits since our latest pull (https://github.com/friendlyarm/kernel-rockchip/commits/nanopi5-v5.10.y_opt, https://github.com/friendlyarm/sd-fuse_rk3588/commits/master), there is actually nothing which could make difference on that topic, aside of probably general system load and hence voltage/power strain. So my best guess, again, is that your PSU is a little too weak.

One thing you could try is, if you can access the ext4 filesystem on partition 8 form another system, to change the CPU governor in `/boot/dietpi.txt` from `schedutil` to `powersave` to check whether a fixed low CPU clock stabilises it. If this works, you can experiment further on the booted system in `dietpi-config` with other governors, like `ondemand` (would be my 2nd choice) etc.

fakemanhk, 2023-01-27:
> It doesn't matter what is shown on the screen, but resolution, cable length and signal strength play a role. Since we use the original kernel and bootloader of FriendlyELEC, and there have not been [...]

I will try on CPU governor change later to see what will happen, but just as I mentioned above, using the same 60W PD power, I was able to run the FriendlyElec with proper output. And I just find that after booting DietPi to setup once and bring it back to my original monitor it works (just some display area shifted), however trying on a complete new SD card boot again there is no display, but anyway this is just a minor issue.

MichaIng, 2023-01-27:
What you describe sounds like a very random behaviour, all pointing into the same direction of system load/voltage causing instabilities. Changing the CPU governor to `powersave` would then be quite likely to help, but let's see.

> just some display area shifted

Same here, wrong resolution or overscan applied.

Sadly without access to the kernel command-line (hidden in the non-accessible partitions of the FriendlyELEC image) it is not trivial to change the console resolution outside of an X (desktop) session: https://wiki.archlinux.org/title/Kernel_mode_setting

With access to cmdline, there would be the `video` parameter. To change if after boot, it looks like `/sys/kernel/debug/dri/0/<devoce>/edid_override` can be used to pass an EDID override binary. But this all looks very hacky. Nasty that there does not seem to be a simply CLI tool or sysfs tunable to immediately enforce a resolution like `echo '1920x1080@60' > /sys/...` and good.

MichaIng, 2023-01-27:
In my case, the framebuffer/console size was 2560x1440 but the monitor itself was set to only 1920x1080. The resolut is that parts of the console content are outside of the screen, so you cannot see the command prompt when it is at the bottom of the console, leaving no other possibility than `clear`.

Not sure why, since the monitor as well as the NanoPi R6S both support 2560x1440. I tried to play with the EDID files. For whatever reason the one provided by the monitor does not explicitly contain a QHD line, not sure why, and I was too lazy now to figure out how to edit and re-compile edit firmware to be loaded. And whether doing so via modprobe at boot works at all, is another question, since the related kernel module is built into the kernel, i.e. not loaded as external module.

Pretty strange why the framebuffer/console is sized 2560x1440 then, since I couldn't see this in the EDID blob, or I do interpret it wrong. But it actually would make sense as this is the native monitor resolution...

Best solution I could find, as I didn't find a way to change the resolution of the monitor itself, was to instead reduce the size of the framebuffer/console:
```sh
apt install fbset
fbset -xres 1920 -yres 1080
# To automatically set it at boot:
echo -e '#!/bin/dash\nexec fbset -xres 1920 -yres 1080' > /var/lib/dietpi/postboot.d/force_1080p.sh
```

MichaIng, 2023-02-12:
Any news regarding CPU governor and system stability?

fakemanhk, 2023-02-14:
> Any news regarding CPU governor and system stability?

Now the system is stable, I didn't change the governor and with 8.14 it has no issue at all.

closed 2023-02-14 by fakemanhk

MichaIng, 2023-02-14:
Ah that is great, even that we didn't change something about the firmware, probably some other package upgrade somehow made a difference.

fakemanhk, 2023-02-14:
But the monitor still having weird initial display issue (only the first setup, after DietPi completed the setup it has no problem), well....after setup I just ignore that and use SSH.

jalsco, 2023-02-18:
@MichaIng  you wrote:  _"On first boot I powered it via USB hub powered via port from my host PC. This worked well until the step where I was choosing to disable the serial console. It then crashed due to insufficient power/voltage, expectedly. Doing the same via dedicated 3.8A USB PSU worked well and stable. "_

My nanopi boots fully, both network interfaces come up fully. No errors, nothing partially up". I am using a 12watt power supply. I can access the nanopi fully using ssh.  However, for some specific tests, I want to use the HDMI and direct attached keyboard. I did access it via hdmi and serial console a few weeks ago, without any problem, during the installation.  However the problem now is that - nothing shows up from the HDMI  (no blinking underline). HDMI screens report there is "no signal". 

What is the step you mentioned " the step where I was choosing to disable the serial console."  Maybe it did that , and now to use the HDMI I need to enable it again ?

MichaIng, 2023-02-18:
Then I'm out of ideas either. There is nothing which could affect HDMI output (in console mode), AFAIK, since we do not even have access to the kernel command-line. It works just reliable here. Typical tests would be using a different/shorter HDMI cable or a different monitor. And as always, never rule the PSU or power cable.

HDMI output and serial console are not related to each other. When I had the device insufficiently powered, it was just randomly that stage where the system crashed.

jalsco, 2023-02-19:
ah ok - I will keep testing. I had already tried different HDMI and different screens. 

Thanks for all your ideas, and your explanations 🙏🏻.  They are ALWAYS useful and appreciated, even if not resulting in immediate fix. 

I will report back if ever I find the reason why HDMI output seems off on my system.