MichaIng/DietPi #6101: "Working Schedule" not working on MotionEye

black00019, 2023-01-26:
Creating a bug report/issue
Required Information
DietPi version | G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=12
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applied'
Distro version | bullseye 0
Kernel version | Linux DietPi 5.15.76-v8+ SMP PREEMPT Fri Nov 4 12:16:41 GMT 2022 aarch64 GNU/Linux
SBC model | RPi 3 Model B+ (aarch64)
Power supply used | 5V 2A
SD card used | Kingstone 128GB

Additional Information (if applicable)
Software title | motionEye
Was the software title installed freshly or updated/migrated? Freshly
Can this issue be replicated on a fresh installation of DietPi? Yes

Steps to reproduce
I have setting "Wokring Schedule" and like this:
Monday - from 8:00 - to 12:00
Detect Motion: During Working Schedule

Expected behaviour
I expect to enable Motion detection from Working Schedule settings

Actual behaviour
The motion detection and notification not work

Extra details

Joulinar, 2023-01-26:
I guess it is more a challenge of the application itself and would need to be reported to the respective development.

black00019, 2023-01-26:
Ok , i search on motionEye Github and found this:
[https://github.com/motioneye-project/motioneye/issues/2663](url)

The project is not under development?