MichaIng/DietPi #6102: Pi-Hole | Api Token QRCode

black00019, 2023-01-26:
Creating a bug report/issue

Required Information
DietPi version | G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=12
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applied'
Distro version | bullseye 0
Kernel version | Linux DietPi 5.15.76-v8+ SMP PREEMPT Fri Nov 4 12:16:41 GMT 2022 aarch64 GNU/Linux
SBC model | RPi 3 Model B+ (aarch64)
Power supply used | 5V 2A
SD card used | Kingstone 128GB

Additional Information (if applicable)
Software title | Pi-Hole
Was the software title installed freshly or updated/migrated? Freshly
Can this issue be replicated on a fresh installation of DietPi? Yes

Steps to reproduce
From API/WEB INTERFACE, I click on "Show API Token" and "Yes show API Token"

Expected behaviour
I expect to show QRCODE's API Token

Actual behaviour
Pi-Hole show an error page without Qrcode

Extra details
[image]

Joulinar, 2023-01-26:
This is working fine on my PiHole installation. Try to use a different browser. Maybe as well something to report to PiHole guys.

MichaIng, 2023-01-27:
Is this installed?
```sh
apt install qrencode
```
And does it work on console?
```sh
qrencode -t ansiutf8 test
```

black00019, 2023-01-28:
> Is this installed?
> 
> ```shell
> apt install qrencode
> ```
> 
> And does it work on console?
> 
> ```shell
> qrencode -t ansiutf8 test
> ```

Yes qrencode is already install and work

black00019, 2023-01-28:
> This is working fine on my PiHole installation. Try to use a different browser. Maybe as well something to report to PiHole guys.

I try with:
Brave
Chrome
Firefox

Don't work anywhere

black00019, 2023-01-28:
I try with this direct link:
[http://192.168.10.5/html/admin/scripts/pi-hole/php/api_token.php](url) --- WORK

but if i try with this:
[http://pi.hole/html/admin/scripts/pi-hole/php/api_token.php](url)   --- DON'T WORK

The link is found on this thread:
[https://github.com/pi-hole/pi-hole/issues/1683](url)

Joulinar, 2023-01-28:
For me this look like an issue inside PiHole still, if the IP address is working but the DNS name not.

MichaIng, 2023-01-28:
Are there any errors in the browser console? Could be some sort of CSP/framing error.

When you access via hostname, this is done via IPv6, right?
```sh
curl -I http://pi.hole/ # Works only if the host uses Pi-hole itself, or from clients
```

One reason could be if the CSP rules deny embedding external resources and when Pi-hole then requests the QR code via IP address instead of the hostname it was accessed with.

black00019, 2023-02-04:
`curl -I http://pi.hole/           

HTTP/1.1 200 OK
Content-Type: text/html
Accept-Ranges: bytes
ETag: "1657326282"
Last-Modified: Sun, 10 Oct 2021 18:44:00 GMT
Content-Length: 3388
Date: Sat, 04 Feb 2023 08:24:52 GMT
Server: lighttpd/1.4.59
`

In console ha this:
[image]

black00019, 2023-02-04:
Maybe a misconfiguration of lighttpd ?

MichaIng, 2023-02-04:
Okay, it is the `X-Frame-Options` header then.

I'm wondering that directly accessing `http://192.168.10.5/html/admin/scripts/pi-hole/php/api_token.php` works, since in this case the header is expectedly set to "deny". It is set to "sameorigin" only, if the request is done from the settings page, respectively the referred is set so: https://github.com/MichaIng/DietPi/blob/master/.conf/dps_93/lighttpd.pihole.conf
This matches exactly how Pi-hole does it with their config: https://github.com/pi-hole/pi-hole/blob/master/advanced/pihole-admin.conf

... although does not match 100%. We use `setenv.add-response-header` here, they switched to `setenv.set-response-header` a month ago: https://github.com/pi-hole/pi-hole/commit/08c7691
This actually seems to make more sense to avoid 2 headers or an appended value. The docs are somehow unclear about the difference: https://redmine.lighttpd.net/projects/lighttpd/wiki/Mod_setenv

I'll do some tests. Can you check browser console for the response headers of the `http://pi.hole/html/admin/scripts/pi-hole/php/api_token.php` request? (Network tab)

MichaIng, 2023-02-04:
Okay, someone needs to explain this:
```console
root@VM-Bullseye:~# curl -I http://pi.hole/html/admin/scripts/pi-hole/php/api_token.php
HTTP/1.1 200 OK
Expires: Thu, 19 Nov 1981 08:52:00 GMT
Cache-Control: no-store, no-cache, must-revalidate
Pragma: no-cache
Set-Cookie: PHPSESSID=2book6tflv9e0cbcn5oanigjaq; path=/; HttpOnly; SameSite=Strict
Content-type: text/html; charset=UTF-8
X-Pi-hole: The Pi-hole Web interface is working!
X-Frame-Options: DENY
Date: Sat, 04 Feb 2023 15:52:01 GMT
Server: lighttpd/1.4.59
```
- As expected, `X-Frame-Options: DENY` by default.
```console
root@VM-Bullseye:~# curl -I -H 'Referer: /admin/settings.php' http://pi.hole/html/admin/scripts/pi-hole/php/api_token.php
HTTP/1.1 200 OK
Expires: Thu, 19 Nov 1981 08:52:00 GMT
Cache-Control: no-store, no-cache, must-revalidate
Pragma: no-cache
Set-Cookie: PHPSESSID=4ojk64iu4f7ai5b5rfa897ao8c; path=/; HttpOnly; SameSite=Strict
Content-type: text/html; charset=UTF-8
X-Frame-Options: SAMEORIGIN
Date: Sat, 04 Feb 2023 15:52:57 GMT
Server: lighttpd/1.4.59
```
- As expected, `X-Frame-Options: SAMEORIGIN` since the referer us the settings page now. No two headers and no appended value.
- Unexpected, even that `add-response-header` is used, the `X-Pi-hole` header is list. Not an issue, as it is a diagnose header only, used by Pi-hole's debugging script to check whether the web interface works, but not on this file, and not influence for client requests.

Now I switched to `set-response-header`, which sounds more like something is replaced instead of added, and the opposite is true:
```console
# curl -I -H 'Referer: /html/admin/settings.php' http://pi.hole/html/admin/scripts/pi-hole/php/api_token.php
HTTP/1.1 200 OK
Expires: Thu, 19 Nov 1981 08:52:00 GMT
Cache-Control: no-store, no-cache, must-revalidate
Pragma: no-cache
Set-Cookie: PHPSESSID=7ltcvfehkeajaiqrkg85qvache; path=/; HttpOnly; SameSite=Strict
Content-type: text/html; charset=UTF-8
X-Pi-hole: The Pi-hole Web interface is working!
X-Frame-Options: SAMEORIGIN
Date: Sat, 04 Feb 2023 15:57:47 GMT
Server: lighttpd/1.4.59
```
I think this is since `add-response-header` in conditionals overwrites the `add-response-header` directives in parent block completely, while `set-response-header` is a different directive, which obviously sets a single header only, and is executed additionally to parent block `add-response-header` directives.

However, so in my case it works as expected. I tested it also in browser with `pi,hole` hostname and IP address and in all cases, visible also in browser console, the header is set correctly.

You did install Pi-hole with `dietpi-software`, didn't you? Please also show:
```sh
cat /etc/lighttpd/conf-enabled/99-dietpi-pihole.conf
```

black00019, 2023-02-05:
Hi, i have installed pi-hole from dietpi-software

``root@DietPi:~# cat /etc/lighttpd/conf-enabled/99-dietpi-pihole.conf
# Based on: https://github.com/pi-hole/pi-hole/blob/master/advanced/lighttpd.conf.debian

# Admin panel
$HTTP["url"] =~ "^(/html)?/admin/" {
	# Create response header for Pi-hole debugger
	setenv.add-response-header = (
		"X-Pi-hole" => "The Pi-hole Web interface is working!",
		"X-Frame-Options" => "DENY"
	)

	# Allow admin panel access to local fonts
	$HTTP["url"] =~ "\.(eot|otf|tt[cf]|woff2?)$" {
		setenv.add-response-header = ( "Access-Control-Allow-Origin" => "*" )
	}
}

# Block . files from being served, such as .git, .github, .gitignore
$HTTP["url"] =~ "^(/html)?/admin/\." {
	url.access-deny = ("")
}

# Block public access to 404 page
$HTTP["url"] =~ "^(/html)?/pihole/" {
	$HTTP["remoteip"] !~ "^1(27|92\.168|0|72\.(1[6-9]|2[0-9]|3[0-1]))\." {
		url.access-deny = ("")
	}
}

MichaIng, 2023-02-05:
Ah, that is an old version of the file indeed. We didn't patch this on existing systems or forced a reinstall of Pi-hole. Since I just updated our config to add more security/privacy headers, if you are open to test:
```sh
cat << '_EOF_' > /etc/lighttpd/conf-enabled/99-dietpi-pihole.conf
# Based on: https://github.com/pi-hole/pi-hole/blob/development/advanced/pihole-admin.conf

# Admin panel
$HTTP["url"] =~ "^(/html)?/admin/" {
	# Block . files from being served, such as .git, .github, .gitignore
	$HTTP["url"] =~ "^(/html)?/admin/\." {
		url.access-deny = ("")
	}

	# Add response headers
	setenv.add-response-header = (
		"X-Pi-hole" => "The Pi-hole Web interface is working!",
		"X-Content-Type-Options" => "nosniff",
		"X-Frame-Options" => "deny",
		"X-XSS-Protection" => "1; mode=block",
		"X-Robots-Tag" => "none",
		"X-Permitted-Cross-Domain-Policies" => "none",
		"Referrer-Policy" => "same-origin",
		"Content-Security-Policy" => "default-src 'self' 'unsafe-inline'"
	)

	# Allow teleporter and API QR code iframes on settings page
	$HTTP["url"] =~ "/(teleporter|api_token)\.php$" {
		$HTTP["referer"] =~ "/admin/settings\.php" {
			setenv.set-response-header = ( "X-Frame-Options" => "sameorigin" )
		}
	}
}

# Block public access to 404 page
$HTTP["url"] =~ "^(/html)?/pihole/" {
	$HTTP["remoteip"] !~ "^(1(27|92\.168|0|72\.(1[6-9]|2[0-9]|3[0-1]))\.|::1|fe[89ab].:|f[cd]..:)" {
		url.access-deny = ("")
	}
}
_EOF_
```
As one copy&paste command block. Or copy the content between `cat` and `_EOF_` manually into your config.

black00019, 2023-02-08:
Great, the new config working
Thanks a lot

MichaIng, 2023-02-08:
Okay great, makes all sense then. Looks like with such required config changes related to changes in Pi-hole we should either patch the existing config on DietPi updates or interactively inform the admin about it and that a `dietpi-software reinstall` can be done to update the config as well as Pi-hole itself.

MichaIng, 2023-02-08:
However, marking this as closed since the issue is solved. Feel free to reopen if required.

closed 2023-02-08 by MichaIng