MichaIng/DietPi #6111: Upgrade Error on DietPi v8.13.2

theunknownstarman, 2023-01-28:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`

G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=12
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`

Buster 1

- Kernel version | `uname -a`

Linux Power 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)

RPi 3 Model B+ (armv7l)

- Power supply used | (EG: 5V 1A RAVpower)

5V 2A

- SD card used | (EG: SanDisk ultra)

SanDisk ultra

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?

Problem only for updating

- Can this issue be replicated on a fresh installation of DietPi?

not tested

- Bug report ID | `echo $G_HW_UUID`

e3befb0e-dcc0-42e4-ae57-f26949672eee

#### Steps to reproduce

1. starting dietpi-update
2. script running until error

#### Expected behaviour
it should updating ;)

#### Actual behaviour
Following Error:

Installing new DietPi scripts
│  - Command: cp -a DietPi-master/dietpi /boot/
│  - Exit code: 1
│  - DietPi version: v8.12.1 (MichaIng/master) | HW_MODEL: 3 | HW_ARCH: 2 | DISTRO: 5
│  - Image creator: DietPi Core Team
│  - Pre-image: Raspbian Lite
│  - Error log:
│ cp: cannot create regular file '/boot/dietpi/dietpi-autostart': Read-only file system
│ cp: cannot create regular file '/boot/dietpi/dietpi-backup': Read-only file system
│ cp: cannot create regular file '/boot/dietpi/dietpi-bugreport': Read-only file system
│ cp: cannot create regular file '/boot/dietpi/dietpi-cleaner': Read-only file system
│ cp: cannot create regular file '/boot/dietpi/dietpi-cloudshell': Read-only file system
│ cp: cannot create regular file '/boot/dietpi/dietpi-config': Read-only file system
│ cp: cannot create regular file '/boot/dietpi/dietpi-cpuinfo': Read-only file system
│ cp: cannot create regular file '/boot/dietpi/dietpi-cron': Read-only file system
│ cp: cannot create regular file '/boot/dietpi/dietpi-ddns': Read-only file system
│ cp: cannot create regular file '/boot/dietpi/dietpi-drive_manager': Read-only file system
│ cp: cannot create regular file '/boot/dietpi/dietpi-explorer': Read-only file system
│ cp: cannot create regular file '/boot/dietpi/dietpi-launcher': Read-only file system
│ cp: cannot create regular file '/boot/dietpi/dietpi-led_control': Read-only file system
│ cp: cannot create regular file '/boot/dietpi/dietpi-letsencrypt': Read-only file system
│ cp: cannot create regular file '/boot/dietpi/dietpi-login': Read-only file system
│ cp: cannot create regular file '/boot/dietpi/dietpi-morsecode': Read-only file system
│ cp: cannot create regular file '/boot/dietpi/dietpi-services': Read-only file system
│ cp: cannot create regular file '/boot/dietpi/dietpi-software': Read-only file system
│ cp: cannot create regular file '/boot/dietpi/dietpi-survey': Read-only file system
│ cp: cannot create regular file '/boot/dietpi/dietpi-sync': Read-only file system
│ cp: cannot create regular file '/boot/dietpi/dietpi-update': Read-only file system
│ cp: cannot create regular file '/boot/dietpi/dietpi-vpn': Read-only file system
│ cp: cannot create regular file '/boot/dietpi/func/change_hostname': Read-only file system
│ cp: cannot create regular file '/boot/dietpi/func/create_mysql_db': Read-only file system

#### Extra details
Option "retry" not helping

Joulinar, 2023-01-28:
> cp: cannot create regular file '/boot/dietpi/func/create_mysql_db': Read-only file system

Your SD card went into read only mode. Usually this indicates file system issues and probably a dying SD card. Can you check for kernel error messages.

```
dmesg -l err,crit,alert,emerg
```

theunknownstarman, 2023-01-28:
no errors on "dmesg -l err,crit,alert,emerg"

Joulinar, 2023-01-28:
try to reboot your system and check again if you can run the update.

theunknownstarman, 2023-01-28:
I rebootet much times. No solution. Sorry

At moment i prepare a new SD Card. Maybe in 1 hour i can test it

Joulinar, 2023-01-28:
Usually if this is an issue with SD card, you should see related errors within kernel log. The file system is not going into r/o mode just for fun. Basically, it's a protection mechanism of the operating system to protect data and avoid further data loss. Or as alternative, file system is defined to be mounted r/o. But this is not our default.

theunknownstarman, 2023-01-28:
ok. New SD Card but same error on updating.

How can i check wheter filesystem is R/o and if so how can i change it?

Joulinar, 2023-01-28:
you can try following

```
mount -o remount,rw /
dietpi-update
```

can you share following as well

```
cat /etc/fstab
```

theunknownstarman, 2023-01-28:
RW Mode not helped:

# You can use "dietpi-drive_manager" to setup mounts.
# NB: It overwrites and re-creates physical drive mount entries on use.
#----------------------------------------------------------------
# NETWORK
#----------------------------------------------------------------
192.168.2.17:/volume1/diet /mnt/diet nfs _netdev,nofail,noauto,x-systemd.automount
192.168.2.17:/volume2/NetBackup /mnt/nas_backup nfs nofail,noauto,x-systemd.automount

#----------------------------------------------------------------
# TMPFS
#----------------------------------------------------------------
tmpfs /tmp tmpfs size=1024M,noatime,lazytime,nodev,nosuid,mode=1777

#----------------------------------------------------------------
# MISC: ecryptfs, vboxsf, glusterfs, mergerfs, bind, Btrfs subvolume
#----------------------------------------------------------------

#----------------------------------------------------------------
# SWAP SPACE
#----------------------------------------------------------------
/var/swap none swap sw

#----------------------------------------------------------------
# PHYSICAL DRIVES
#----------------------------------------------------------------
PARTUUID=907af7d0-02 / ext4 noatime,lazytime,rw 0 1
PARTUUID=907af7d0-01 /boot vfat noatime,lazytime,ro 0 2
UUID=2352e967-b75b-4a63-af9c-52472b8efaa0 /mnt/sda1 ext3 noatime,lazytime,rw,nofail,noauto,x-systemd.automount

theunknownstarman, 2023-01-28:
Found a solution:

I switched the R/O Mode to R/W in the dietpi drive manager
After this an update was succesful.

I think the problem is away

Joulinar, 2023-01-28:
yes the `bootFS` was mounted read only. Did you set this yourself by purpose?

```
PARTUUID=907af7d0-01 /boot vfat noatime,lazytime,ro 0 2
```

theunknownstarman, 2023-01-28:
no. I never changed it.

Joulinar, 2023-01-28:
and you download a fresh image from our server? Btw, your original issue was opened for a system running Debian Buster.  As we don't offer Buster images for long, how did you flashed the new SD card?

theunknownstarman, 2023-01-28:
The dietpi server runs about 5 years without big changes.

Cause of the complex configurations i copied the old 16 GB SD Card to a new 64 GB card without new installation.

Joulinar, 2023-01-28:
> Cause of the complex configurations i copied the old 16 GB SD Card to a new 64 GB card without new installation.

OK, you have copied the problems over. Could it be a manuell mistake to set r/o mode by accident?

theunknownstarman, 2023-01-28:
may be a manual mistake cause i changed the mounts in order to add an NAS drive.

Joulinar, 2023-01-28:
ok, I guess nothing we can do in addition. Therefore I'm going to close the issue. Feel free to reopen if needed.

closed 2023-01-28 by Joulinar