MichaIng/DietPi #6112: InfluxDB install error

muibusan, 2023-01-28:
Hi!

I tried to install InfluxDB using dietpi-software, but it didn't succeed.

I sent a bug report, reference code: f7bdab83-8958-443f-94e1-7d5dab2dbb80.

The shell says:
```
The following signatures couldn't be verified because the public key is not available: NO_PUBKEY D8FF8E1F7DF8B07E

Reading package lists...
W: GPG error: https://repos.influxdata.com/debian bullseye InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY D8FF8E1F7DF8B07E

E: The repository 'https://repos.influxdata.com/debian bullseye InRelease' is not signed.
```

Joulinar, 2023-01-28:
We are already aware on this as the repository PGP key has been changed this week. We will provide a fix on next release. If you don’t like to wait until that time, you would need to go with manually fix and download the new key.

MichaIng, 2023-01-28:
Yes, they rotated the key and changed the URL for the valid key, for whatever reason, breaking everyones scripts and CI... https://github.com/influxdata/influxdb/pull/24061

Run this to fix it:
```sh
sed -i 's/influxdb.key/influxdata-archive_compat.key' /boot/dietpi/dietpi-software
```
Fixed already with next DietPi update.

MichaIng, 2023-01-28:
Reopened. It's better to have it here on GitHub, pinned and with the solution.

sibofax, 2023-02-01:
Whoops - apologies!

MichaIng, 2023-02-05:
Marking as closed now. But will stay pinned.

closed 2023-02-05 by MichaIng