MichaIng/DietPi #6113: add user for transmission

hanscees, 2023-01-28:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
I have added a user, because sending over the root passwd for transmission seems a bit unsafe, I have no https. 

I did this
`#first add user
useradd hanscees
usermod -aG debian-transmission hanscees # add it to group
groups hanscees -> shows hanscees : hanscees debian-transmission
passwd hanscees (made up a passwd)

systemctl stop transmission-daemon
vi /etc/transmission-daemon/settings.json
set rpc-username to hanscees
"rpc-password": "passwd", 
#note no passwd hash needed
systemctl start transmission-daemon

`

- ...

#### Describe the solution you'd like:
dietpi-config could have a funtion " add  transmission-user "
Or add user and then you can somehow make that user work for transmission.
Key is don't maker it root and dont let it use a sensitive passwd
- ...

#### Describe alternatives you've considered:
do it by hand
- ...

#### Additional context

- ...

MichaIng, 2023-01-28:
You are not really handing over the root users password to Transmission: It has an independant internal user management where the default user simply is named as well root. But the paasword is different (unless you chose to leave/set them the same). The global software password is used, can be changed and should be.

hanscees, 2023-01-28:
Oh, apologies. I couldn't get it to work with any password, so I resolved it in a way I used on OMV too.
But I have installed twice now because of an ssd disk making trouble, so perhaps the second time I forgot to change passwords of typed them wrong or something. 
Getting used to dietpi, I'm gonna keep it!

MichaIng, 2023-01-28:
Using `root` as default login name is probably not optimal. We decided to use `dietpi` as default some time ago, but only applied it when touching certain software code anyway and for new implementations. Something that is easy to remember, when used consistently as default, but not commonly used by brute-force bots.

Btw, since you mention the password needs to be hashed. If you add it unhashed, Transmission will hash it automatically for you. But we have an open ToDo comment in our code to hash it directly (so at no time the unhashed global software password is in this plain text file). What is the hashing algorithm, how did you hash it?

hanscees, 2023-01-28:
I couldnt get the hasing to work either. It wasnt just sha1. 
however a clue might be here:
https://github.com/transmission/transmission/issues/839 

and https://github.com/tomwijnroks/transmission-pwgen

MichaIng, 2023-01-28:
Ah perfect, thanks for the links. I'll add this to `dietpi-software`.

MichaIng, 2023-01-28:
Done: https://github.com/MichaIng/DietPi/commit/32e062d 😃
And a fix: https://github.com/MichaIng/DietPi/commit/38a5529

So an 8 character alphanumerical (+ dot and slash) salt needs to be appended to the password, then SHA1 summed (puhh, I hope they start supporting modern/secure hashes soon...), then the SHA1 with again salt appended and `{` prepended added to as `rpc-password` value.

closed 2023-02-10 by MichaIng