MichaIng/DietPi #6116: Attempting to install raspotify [after updating to bullseye where sonnar repo is set to buster]

tuosce, 2023-01-29:
Thanks for looking at this:

#### Details:
- Date           | Sun Jan 29 12:57:42 GMT 2023
- DietPi version | v8.13.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite (64-bit)
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPi 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `apt-get -y update`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:
1. Update from buster to bullseye
2. Attempt to install raspotify
#### Expected behaviour:
Raspotify installs successfully
#### Actual behaviour:
Even though I set sonnar to use buster release, I'm still getting the errors below and raspotify or other software is not installing.
- ...
#### Extra details:
n/a
#### Additional logs:
```
Hit:1 https://apt.sonarr.tv/debian buster InRelease
Hit:2 https://archive.raspberrypi.org/debian bullseye InRelease
Get:3 https://deb.debian.org/debian-security bullseye-security InRelease [48.4 kB]
Ign:4 https://download.mono-project.com/repo/debian bullseye InRelease
Hit:5 http://ftp.uk.debian.org/debian bullseye InRelease
Hit:6 http://ftp.uk.debian.org/debian bullseye-updates InRelease
Hit:7 http://ftp.uk.debian.org/debian bullseye-backports InRelease
Err:8 https://download.mono-project.com/repo/debian bullseye Release
  404  Not Found [IP: 152.199.19.161 443]
Get:9 https://deb.debian.org/debian-security bullseye-security/main arm64 Packages [212 kB]
Reading package lists...
E: The repository 'https://download.mono-project.com/repo/debian bullseye Release' does not have a Release file.
```

Joulinar, 2023-01-29:
> Err:8 https://download.mono-project.com/repo/debian bullseye Release

Did you upgrade from Debian Buster to Bullseye? Because Mono project did not offer any Bullseye package. You would need to switch package configuration for Mono back tu Buster?

tuosce, 2023-01-29:
> Did you upgrade from Debian Buster to Bullseye?

Yes. I didn't realise Mono affects Sonarr.

I used this code below to update both repos and **it now works, issue resolved**. Thank you!

`sed -i 's/bullseye/buster/' /etc/apt/sources.list.d/*mono*.list /etc/apt/sources.list.d/*sonarr*.list
apt update`

closed 2023-01-29 by tuosce

MichaIng, 2023-01-29:
Yes, both repos do not provide a "bullseye" suite. The "buster" suite however works fine on Debian Bullseye.