MichaIng/DietPi #6117: BIG problema after upgrade HA 3.10

maury77, 2023-01-29:
After reinstall HA via dietpi I have big issue

The HA don't start gui adn service the ripetitive error is

qlalchemy.exc.ArgumentError: Type annotation for "Statistics.metadata_id" can't be correctly interpreted for Annotated Declarative Table form.  ORM annotations should normally make use of the ``Mapped[]`` generic type, or other ORM-compatible generic type, as a container for the actual type, which indicates the intent that the attribute is mapped. Class variables that are not intended to be mapped by the ORM should use ClassVar[].  To allow Annotated Declarative to disregard legacy annotations which don't use Mapped[] to pass, set "__allow_unmapped__ = True" on the class or a superclass this class. (Background on this error at: https://sqlalche.me/e/20/zlpr)

Joulinar, 2023-01-29:
any reason not filling the bug report template?

#### Required 
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)
- Home Assistant system log | `journalctl -u home-assistant`

maury77, 2023-02-08:
any Idea  I rollback is ok ! now I try upgrade new and the same errore

HELP

Joulinar, 2023-02-08:
again, you did not answer my question. Do this first, afterwards we could assist.

maury77, 2023-02-08:
I update my post with data

DietPi version |

G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=13
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='not applicable'

Distro version | bullseye 1

Kernel version | uname -a
SBC model | ***@***.***:~# uname -a
Linux sdomotica 5.15.84-v7+ #1613
<https://github.com/MichaIng/DietPi/issues/1613> SMP Thu Jan 5 11:59:48 GMT
2023 armv7l GNU/Linux

SDcard used | (EG: SanDisk ultra 32GB)
Home Assistant system log | journalctl -u home-assistant

maury77, 2023-02-08:
I found the problem

The installation with dietpi-software reinstall 157 not install sqlalchemye

I have install the last version (2.02( that is not compatible with stack , I try install 1.46 and the problem is resolved

MichaIng, 2023-02-08:
Ah, then I wonder whether the dependency was missed to be declared in one of the integration you use. Usually HA installs them automatically on startup or when accessing features the first time. `dietpi-software` installs the modules needed by HA core declared via `requirements.txt`, for everything else integrations are expected to declare their dependencies internally, so that HA installs them on demand to `/mnt/dietpi_userdata/homeassistant/deps`.

But great that you found the solution 👍.

Joulinar, 2023-02-08:
I guess one of the installed python packages is not compatible with SQLAlchemy v2.  Probably some wheels would need to be updated by the individual maintainer https://docs.sqlalchemy.org/en/20/errors.html#error-zlpr

SQLAlchemy v2 has been released 2 weeks back. Na if not specified, it will be pulled automatically as latest version available, https://www.piwheels.org/project/sqlalchemy/

MichaIng, 2023-02-08:
Ah this can be the cause as well of course. I understood it so that SQLAlchemy was not installed OOTB but manually, in which case v2 is pulled. Since the integrations are no Python modules (are they?) and their dependencies are not installed into the system's or pyenv Python environment, but into this dedicated `/mnt/dietpi_userdata/homeassistant/deps` directory, `pip` cannot see or warn about conflicts, when you manually install modules.

maury77, 2023-02-10:
the integrations should all be python modules, here it was the native recorder module

After upgrade there was no SQLAlchemy and I had installed it manually

MichaIng, 2023-02-10:
> the integrations should all be python modules, here it was the native recorder module

That is right, but those are also not installed into the regular Python environment, so `pip` does not see them. Actually, probably one can make `pip` have a look into `/mnt/dietpi_userdata/homeassistant/deps` as well, if that is just some kind of venv with common Python directory structure 🤔.

maury77, 2023-02-16:
root@sdomotica:/mnt/dietpi_userdata/homeassistant/deps# ls -al
totale 52
drwxrwxr-x  8 homeassistant homeassistant 4096 27 dic 16.21 .
drwxrwxr-x 15 homeassistant homeassistant 4096 14 feb 16.15 ..
drwxr-xr-x  3 homeassistant homeassistant 4096  8 feb 13.04 bin
drwxr-xr-x  2 homeassistant homeassistant 4096  8 feb 11.53 cache
-rw-r--r--  1 homeassistant homeassistant   15  6 mar  2021 ciso-requirements.txt
drwxr-xr-x  2 homeassistant homeassistant 4096 11 ott 23.22 doc
-rw-r--r--  1 homeassistant homeassistant   21  6 mar  2021 extra-requirements.txt
drwxr-xr-x  3 homeassistant homeassistant 4096  9 mag  2021 include
drwxr-xr-x  5 homeassistant homeassistant 4096  8 feb 11.45 lib
-rw-r--r--  1 homeassistant homeassistant 1065 27 dic 16.21 LICENSE
-rw-r--r--  1 homeassistant homeassistant  121  6 mar  2021 requirements.txt
drwxr-xr-x  3 homeassistant homeassistant 4096 10 feb  2020 share
-rw-r--r--  1 homeassistant homeassistant  114  6 mar  2021 test-requirements.txt

root@sdomotica:/mnt/dietpi_userdata/homeassistant/deps# more requirements.txt 
rx >= 3.0.1
certifi >= 14.05.14
six >= 1.10
python_dateutil >= 2.5.3
setuptools >= 21.0.0
urllib3 >= 1.15.1
pytz>=2019.1

root@sdomotica:/mnt/dietpi_userdata/homeassistant/deps# more extra-requirements.txt 
pandas>=0.25.3
numpy

MichaIng, 2023-02-18:
Did you create these `requirements.txt` manually or is this some old leftover? Based on timestamps looks like the letter. So that is just a common Python environment and can be merged with pyenv's and then reset the same way on reinstalls. This assures that there are no version mismatches but means that all internal dependencies need to be reinstalled, in case recompiled as well.

I tested back and forth, and HA is actually quite robust working with merged pyenv and deps environments. Even if you cause conflicts by upgrading everything to versions not supported by the integrations, HA automatically downgrades affected modules, reinstalls missing ones etc.

Hence: https://github.com/MichaIng/DietPi/commit/a677610

Little bonus is that one can upgrade or cleanup the internal deps as well, within the borders or what the integrations support. One however cannot break it, as HA automatically assures that all dependencies are met, as mentioned 🙂.

maury77, 2023-02-21:
I have not created them manually, I think they are old things after several upgrades

closed 2023-02-26 by MichaIng

maury77, 2023-03-04:
Sorry , I try to upgrade tu minor version 2023.3.1 but I have new crash with sqlaalchemy and HA don't start

with many error as type  Setup failed for recorder: Unable to import component: cannot import name 'ColumnElement' from 'sqlalchemy'

MichaIng, 2023-03-04:
How did you try to upgrade HA, via `pip3` in the pyenv, or via reinstall?

Unexpected if the Python version itself has not changed. However, for now, before you reinstall it:
```sh
rm -R /mnt/dietpi_userdata/homeassistant/deps
```
This will be done automatically on reinstalls after next DietPi release.

maury77, 2023-03-04:
I use this script 

**exec sudo -u homeassistant dash -c '. /home/homeassistant/pyenv-activate.sh; exec pip3 install --no-cache-dir -U homeassistant'**

MichaIng, 2023-03-04:
A little simpler 🙂:
```sh
/home/homeassistant/homeassistant-update.sh
```
It does the sudo stuff automatically.

But indeed strange how it could have caused that issue then. If any other sqlalchemy version was needed, if would have pulled it as dependency, and as well on service start, this should be part of the automatic deps installs.

maury77, 2023-03-06:
the problem is the same of the initial issue.

Also upgrade with "/home/homeassistant/homeassistant-update.sh" doesn't install right version of SQLAlchemy

In the 2023.3.1 SQLAlchemy 2.0.2 is necessary and I had the 1.46 

I install manualy SQLAlchemy 2.0.2 ando now it is ok

MichaIng, 2023-03-06:
Then there is a dependency conflict within Home Assistent, respectively between Home Assistant and one of your integrations. An older SQLAlchemy wouldn't be installed of not either HA or an integration declares to be not compatible with the recent one.

Do you find SQLAlchemy within `/mnt/dietpi_userdata/homeassistant/deps` or only with
```bash
sudo -u homeassistant dash -c '. /home/homeassistant/pyenv-activate.sh; exec pip3 freeze'
```
?

cazzoo, 2024-02-28:
I notice some issues with the latest update as well about webrtc-noise-gain missing. I've been trying to add the package directly but that did complain for missing gcc (expected AFAIK).
I'm facing that issue on fresh install as well, meaning that the current startup scripts are not in position to find all the necessary deps.
I'm currently installing g++ again to add webrtc-noise-gain back to my py env and check if the issue vanishes or not

Joulinar, 2024-02-28:
See #6931