MichaIng/DietPi #6119: DietPi-Drive_Manager | Rework to not rewrite /etc/fstab anymore

MichaIng, 2023-01-31:
This is quite intrusive and unnecessary. The drive detection and menu listing is completely independent of `/etc/fstab`. We can simply add/remove/rewrite the single entry when a drive is changed.

This makes adding non-implemented drive types to the preserved "misc" exclusion obsolete and solves many other issues when users add custom mount options etc. This also makes solves #5843 the proper way.

Joulinar, 2023-04-18:
Don't use drive manager as long as behaviour has not been changed.

pkoevesdi, 2026-01-03:
Just want to leave a very "plus" for this feature. Just lost several precious entries in my fstab, only because I used drive-manager afterwards. :-(