MichaIng/DietPi #6128: DietPi-Drive_Manager attempts to format a device that does not exist

bennettp123, 2023-02-04:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
    ```
    G_DIETPI_VERSION_CORE=8
    G_DIETPI_VERSION_SUB=13
    G_DIETPI_VERSION_RC=2
    G_GITBRANCH='master'
    G_GITOWNER='MichaIng'
    G_LIVE_PATCH_STATUS[0]='applied'
    G_LIVE_PATCH_STATUS[1]='not applicable'
    ```
- Distro version | `bullseye`
- Kernel version | `Linux pihole 5.10.0-21-amd64 #1 SMP Debian 5.10.162-1 (2023-01-21) x86_64 GNU/Linux`
- SBC model | `Native PC (x86_64)`
- Power supply used | whatever the vhost provides 😛  
- SD card used | nope 😛 (it's a qcow2 volume on xfs on a crucial mx500 connected using sata3)

#### Additional Information (if applicable)
- Software title | `DietPi-Drive_Manager` (I think)
- Was the software title installed freshly or updated/migrated? fresh install
- Can this issue be replicated on a fresh installation of DietPi? yes

- Bug report ID | 3e32d712-b811-4ff8-b41a-64a7d7df6718

#### Steps to reproduce
1. Launch DietPi inside a virtual machine
2. Connect a virtual disk (eg /dev/vda, or in this case, /dev/vdc)
3. Launch `dietpi-drive_manager`, and format the disk using GPT, ext4

#### Expected behaviour

- DietPi-Drive_Manager creates the GPT partition table
- DietPi-Drive_Manager creates a new partition in the partition table
- DietPi-Drive_Manager formats the new partition as ext4 

#### Actual behaviour

- DietPi-Drive_Manager creates the GPT partition table (as expected)
- DietPi-Drive_Manager creates a new partition in the partition table (as expected)
- DietPi-Drive_Manager is unable to format the partition

#### Extra details

- Failing command: `mkfs.ext4 -F -m 0 /dev/vdcp1`
- Exit code: 1
- Error log:
    ```
    mke2fs 1.46.2 (28-Feb-2021)
    The file /dev/vdcp1 does not exist and no size was specified.
    ```              

- `lsblk` shows that the device name is `vdc1`, not `vdcp1`

#### Workaround
1. Adjust the command to `mkfs.ext4 -F -m 0 /dev/vdc1`
2. Make the same adjustment to any other commands that fail (eg `mount -o noatime,lazytime,rw /dev/vdcp1 /mnt/backups` &rarr; `mount -o noatime,lazytime,rw /dev/vdc1 /mnt/backups`, pictured below)
3. No need to update `/etc/fstab`—it uses `UUID=`, not `/dev/vdc1`, so it'll work regardless

[image]

[image]

MichaIng, 2023-02-04:
Many thanks for your report.

`/dev/vdcN`, what is this for a type if drive? Never seen it, and as unknown type, the drive manager guesses partitions to be added with `pN` instead of just `N`. I'll add it.

MichaIng, 2023-02-04:
Fixed with: https://github.com/MichaIng/DietPi/commit/ee86e9c
Would be great if you tested it.

If even possible, when used as root filesystem, it will now also be expended automatically on first boot.

closed 2023-02-05 by MichaIng