MichaIng/DietPi #6130: Open Beta v8.14 | Please help testing and hardening our upcoming release

MichaIng, 2023-02-04:
RC version | v8.14.0
---------- | -------
v8.14.0 PR | #6078
Raw changelog | https://github.com/MichaIng/DietPi/blob/beta/CHANGELOG.txt
Code changes | https://github.com/MichaIng/DietPi/compare/master...beta
Release date | **2023-02-11**

### How to apply: https://github.com/MichaIng/DietPi/blob/master/BRANCH_SYSTEM.md

#### Related/solved issues: https://github.com/MichaIng/DietPi/issues?q=is%3Aissue+milestone%3Av8.14
____
### Beta v8.14.0
_(2022-02-04)_

#### New images
- Orange Pi 5 | Support for Orange Pi 5 SBC has been added to DietPi with hardware ID 80.
- Odroid N2/HC4 | New images have been generated with support for the default Hardkernel SPI bootloader petitboot. Thanks to this, boot from USB devices is now possible, and on Odroid HC4 it is not necessary anymore to erase the SPI flash or flash mainline U-Boot to it. Many thanks to @hstarcm and @Power-onoff for bringing up this topic and help testing adjustments to the boot script: https://github.com/MichaIng/DietPi/issues/5634
- ROCK 5B | New images with ext4 boot filesystem have been generated, to solve possible issues and limitations of the previously used FAT filesystem. Many thanks to @isarrider for reporting this issue: https://github.com/MichaIng/DietPi/issues/6122

#### Enhancements
- NanoPi R2S/NEO3 | Video related kernel modules are now blacklisted on these headless SBCs, saving some bits of RAM and load at boot.
- DietPi-Globals | G_CHECK_NET: When hitting "Retry" from the error handler, the IPv6 connection check is now skipped as well when IPv6 is disabled via dietpi-config network settings or manually. The same is true for changes to the CONFIG_CHECK_CONNECTION/DOMAIN dietpi.txt settings, which can now also be changed right from the error handler. Many thanks to @derSebastian for reporting a related issue: https://github.com/MichaIng/DietPi/issues/5667
- DietPi-Config | It is now possible to change the dietpi.txt network connection test settings CONFIG_CHECK_CONNECTION_IP, CONFIG_CHECK_CONNECTION_IPV6 and CONFIG_CHECK_DNS_DOMAIN via dietpi-config Network Options: Misc menu. Those are used when checking Internet access via IPv4 and IPv6 and DNS resolver functionality.
- DietPi-Software | Gogs: Our ARMv6 build for Raspberry Pi 1 and Zero (1) models has been updated to latest version 0.12.10. The update can be applied via reinstall: dietpi-software reinstall 49

#### Bug fixes
- ROCK Pi S | Resolved an issue where the common serial console on UART0 did not show any output and login prompt. Many thanks to @Marsu31 for reporting this issue: https://github.com/MichaIng/DietPi/issues/5972
- ROCK Pi 4 | Resolved an issue where onboard WiFi on ROCK Pi 4C Plus did not work, and potentially other variants than ROCK Pi 4B. Many thanks to `@Marco` for reporting this issue: https://dietpi.com/forum/t/rock4c-no-wifi-hardware-found/15654
- DietPi-Drive_Manager | Resolved an issue where KVM VirtIO virtual disk volumes were not handled correctly. Many thanks to @bennettp123 for reporting this issue: https://github.com/MichaIng/DietPi/issues/6128
- DietPi-Software | Docker: Resolved a v8.13 regression where Docker dependants (Portainer, Roon Extension Manager) failed to install as the Docker socket was not activated. Many thanks to @omavoss for reporting this issue: https://github.com/MichaIng/DietPi/issues/6065
- DietPi-Software | ownCloud/Nextcloud: Resolved an issue where a global software password with dashes lead to a failing install. Many thanks to @maxmoonshark for reporting this issue: https://github.com/MichaIng/DietPi/issues/4199
- DietPi-Software | myMPD: Resolved an issue where intended initial configs were not applied correctly and web UI changes did not survive a service restart.
- DietPi-Software | X.Org X Server: Resolved an issue where the install failed on ARM VMs with KMS support, where the xserver-xorg-video-vmware package was tried to be installed. Many thanks to @mews-se for reporting this issue: https://github.com/MichaIng/DietPi/issues/6085
- DietPi-Software | Amiberry: Resolved an issue where using keyboard LEDs as power/activity indicators did not work, and stopping the service could hang. You can apply the fix by reinstalling Amiberry via "dietpi-software reinstall 108". Many thanks to @DJP75 for reporting this issue: https://github.com/MichaIng/DietPi/issues/5536

MichaIng, 2023-02-04:
**Odroid N2** and **HC4** users, if you need/want to test USB boot support, this can be easily added to existing images as well.

Also **ROCK 5B** system one can reformat the root filesystem to `ext4` to permanently solve potential issues on kernel upgrades.

Let me know and I'll share the required steps.

Roalkege, 2023-02-07:
> **Odroid N2** and **HC4** users, if you need/want to test USB boot support, this can be easily added to existing images as well.
> 
> Also **ROCK 5B** on systems one can reformat the root filesystem to `ext4` to permanently solve potential issues on kernel upgrades.
> 
> **Orange Pi 5** images can be found on our download server: https://dietpi.com/downloads/images/ They are shipped with DietPi beta branch applied until v8.14 has been released.
> 
> Let me know and I'll share the required steps.

Is it possible to update the beta v8.14 to stable release afterwards?
Otherwise I would need the instructions on how to enable this for Odroid N2/N2+.

MichaIng, 2023-02-07:
> Is it possible to update the beta v8.14 to stable release afterwards?

The new images ship with stable DietPi, the change was only in our installer script we use to generate the image.

To enable it on your running image (so you can clone/dd it over to a USB drive): https://github.com/MichaIng/DietPi/issues/5634#issuecomment-1419842361

sajimenon, 2023-02-09:
I downloaded the latest beta image "[DietPi_RPi-ARMv8-Bullseye_F2FS.7z](https://dietpi.com/downloads/images/testing/DietPi_RPi-ARMv8-Bullseye_F2FS.7z)" and installed on my RPi 3B which had Samsung Ultra SD card of size 64GB. I was running stable version of dietpi prior which was formatted for this beta install.

While running the first boot setup itself I could see that the drive manager is only showing less than 1GB file system size for my / folder and is not allowing to increase though I had a 64GB SD card. My install of additional software was all failing since there is not enough free space.

Again formatted the SD and reinstalled the stable version of Dietpi and I could see that the drive manager shows full 64 GB.

MichaIng, 2023-02-10:
#### Beta v8.14.1 has been merged: #6138

MichaIng, 2023-02-11:
#### Beta v8.14.2 has been merged: #6146

MichaIng, 2023-02-11:
### DietPi v8.14 has been released: #6139
Many thanks to all testers ❤️!

closed 2023-02-11 by MichaIng