MichaIng/DietPi #6132: Automated install : first system update does not run, network issues

notsocafe, 2023-02-05:
#### Required Information
- DietPi version | `cat /boot/dietpi/.version`

G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=13
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`

bullseye 0

- Kernel version | `uname -a`

Linux DietPi 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)

RPi 4 Model B (aarch64)

- Power supply used | (EG: 5V 1A RAVpower)

Raspberry Pi Official USB-C Power Supply

- SD card used | (EG: SanDisk ultra)

Panasonic SDHC Class 10 U 1 32GB (from the HutoPi kit)

#### Steps to reproduce
1. Flash SD card with Rufus
2. Fill in dietpi.txt with either Wi-Fi or Ethernet enabled, any NTP configuration, and automatic install enabled
3. Plug the RPi in
4. Install crashes at first apt update or DietPi update check
5. Upon gaining control back after the error message, connection is OK

#### Expected behaviour
- Installation should work

#### Actual behaviour
- Installation crashes after the NTP setup (either cURL or apt fails to resolve domain names)

#### Extra details
- ethtool hangs upon trying to limit Ethernet speed
- Cable used is a CAT6 over a 1Gbps (max 10Gbps) network
- Different networks (both Ethernet and Wi-Fi) have the same issue
- DNS servers attempted are either 9.9.9.9, 1.1.1.1, 8.8.8.8 or 8.8.4.4

Joulinar, 2023-02-05:
can you share logs from inside `/var/tmp/dietpi/logs`

notsocafe, 2023-02-05:
[Here is a dump of all files in that folder.](https://gist.github.com/notsocafe/bee2252a4429e2f4a890f2c7c76a2bc7)
dietpi.txt was only altered to enable automatic installation.

Note : On SSH'ing into the paused install (after setting a root password), DietPi complained about `ping -4nc 1 -W 10 9.9.9.9` failing, while being obviously online.

MichaIng, 2023-02-06:
I think `dietpi.txt` line endings have been converted to Linux format, causing the various errors when applying settings.. Which Windows version and editor do you use to edit it?

MichaIng, 2023-02-06:
Do you see the same `[FAILED]` and `Unknown input value (...). Nothing has been applied.` entries in `dietpi-firstboot.log`?

While this caused some issues and e.g. is the reason why Ethernet speed was not changed, the network was still up:
> bound to 192.168.0.49 -- renewal in 18248 seconds.

So if pinging 9.9.9.9 fails, then the line endings are probably unrelated and it is an issue with LAN/firewall/ISP. Can you check in console:
```sh
ping -4 9.9.9.9
ping -6 2620:fe::fe
getent hosts dns9.quad9.net
```

notsocafe, 2023-02-07:
I am unavailable to pursue testing at the moment, I will investigate the newline issue whenever possible.

Network is OK (no particular configuration), commands work and I can SSH to the "networkless" Pi.

MichaIng, 2023-02-07:
SSH does not mean Internet access. But aside of this it is perfectly possible that `9.9.9.9` is blocked either by some firewall, some implementation to prevent local DNS bypasses (e.g. to ensure Pi-hole or such is used), or by the ISP. You can change this IP in `/boot/dietpi.txt` to e.g. `1.1.1.1` and see whether this works better, or increase the timeout right from the error handler if it just takes too long.

> commands work

Wait, you mean `ping -4 9.9.9.9` works (no lost packets, response within a second)? Because this is exactly what the test does.

notsocafe, 2023-02-08:
I know, but I can access the internet from either other computers on the network on the Pi.

> Wait, you mean ping -4 9.9.9.9 works (no lost packets, response within a second)? Because this is exactly what the test does.

Yes, pings are successful. I can run various operations (such as package installations) successfully.

MichaIng, 2023-02-08:
> Yes, pings are successful. I can run various operations (such as package installations) successfully.

Then simply retry the setup/installation as it should succeed then as well. Probably the network just needed a little longer to set up, e.g. DHCP timed out one time or similar. If you have the time to flash the image again, take care the `dietpi.txt` line endings this time, enable automated setup again, and when you run into the same connection test/ping error, check the following logs:
```sh
journalctl -u networking -u ifup@eth0.service -u ifup@wlan0.service
last | head -1
```
So we see what happened when the network was meant to be configured and at what time the login happened, starting the automated setup attempt.

MichaIng, 2023-03-18:
I'll mark this as closed. Feel free to reopen if the issue persists.

closed 2023-03-18 by MichaIng