MichaIng/DietPi #6134: Docker CE 23.0.0 incorrectly tries to apply an AppArmor profile when not actually enabled on the system

irousom, 2023-02-06:
### Creating a bug report/issue

#### Required Information
- DietPi version 8.13.2
- Distro version Bullseye
- Kernel version Linux dc1-qa 5.10.0-21-amd64 #1 SMP Debian 5.10.162-1 (2023-01-21) x86_64 GNU/Linux
- Intel x64 virtual machine

#### Additional Information (if applicable)
- Software title Docker CE
- Issue: Docker CE 23.0.0 seems to think that AppArmor is enabled (and therefore tries to apply a suitable profile) when in fact it is not, as DietPi doesn't include AppArmor in its base installation.
- This issue does NOT appear to occur on ARM64 hardware

#### Steps to reproduce
1. This assumes you have a previously working Docker CE installation and one or more containers running.
2. Upgrade to Docker CE using the normal apt update process; confirm that version 23.0.0 is now installed.

#### Expected behaviour
Docker should not attempt to apply an AppArmor profile on systems that don't have it enabled, thereby allowing containers to start/restart as they did before.

#### Actual behaviour
Containers that were previously running, configured to automatically restart, etc. will fail; attempting to start/restart any container fails with the following error message:

docker: Error response from daemon: AppArmor enabled on system but the docker-default profile could not be loaded: running `apparmor_parser apparmor_parser --version` failed with output:
error: exec: "apparmor_parser": executable file not found in $PATH.

#### Extra details
This isn't a DietPi issue per se but more likely a Docker CE bug; I am sharing it for awareness as other x64 distributions that in fact have AppArmor enabled by default do not likely experience any issues.  It appears to only be an issue for systems running the x64 architecture; ARM64 does not appear to have this same issue.  **Explicitly** disabling AppArmor in the kernel configuration seems to be a suitable workaround; for example:

sudo echo 'GRUB_CMDLINE_LINUX_DEFAULT="$GRUB_CMDLINE_LINUX_DEFAULT apparmor=0"' | sudo tee /etc/default/grub.d/apparmor.cfg
sudo update-grub
sudo reboot

It may be desirable to include this explicit disabling of AppArmor in future DietPi releases and to only explicitly enable it if the user installs all the necessary AppArmor support elements in their particular deployment.

Joulinar, 2023-02-06:
basically you could have connected to the pinned issue https://github.com/MichaIng/DietPi/issues/6126

irousom, 2023-02-06:
Yeah - didn't see that one until now.  The only useful part is potentially that workaround that doesn't require installing any extra packages.  Closing this duplicate...

closed 2023-02-06 by irousom