MichaIng/DietPi #6135: Docker - buildx

isarrider, 2023-02-07:
Hi,

I just noticed that buildx is (not anymore?) installed,
when docker is installed...
Is this intentional?

(easy fix to install the plugins:
sudo apt-get install docker-buildx-plugin docker-compose-plugin)

BR;
Alex

Joulinar, 2023-02-07:
from our side we don't change / touch anything. Our install script is going to install `docker-ce` package (only). This is the case for long. Probably thinks changed on lates Docker release v23 https://docs.docker.com/engine/release-notes/23.0/#new

MichaIng, 2023-02-07:
> docker-compose-plugin

This is deprecated: https://docs.docker.com/compose/install/linux/
Use the Docker Compose install option from DietPi-Software, or install the new binary manually.

isarrider, 2023-02-07:
actually it make sense that it doesnt come with docker by defautl... (keepin it lean)
Only on build system it is needed... Maybe another install option?

MichaIng, 2023-02-07:
As long as it is not more than `apt install`, we usually do not implement install options, at least not if those are not a dependency of other implemented install options. But we could add a hint to our docs about other Docker packages available from the Docke repo.

MichaIng, 2023-02-10:
Done with: https://github.com/MichaIng/DietPi/commit/f95bd3a

isarrider, 2023-02-11:
I feel my thread was hijacked :wink:
For me it was about buildx...
But happy for the folks who have the Go version of compose now 😉

Joulinar, 2023-02-11:
We ship the Go version vor long time already.

MichaIng, 2023-02-11:
> I feel my thread was hijacked 😉

Yeah, you are right, especially since with #6142 an explicit request to use the Docker Compose APT package was opened, though including the same misunderstanding that this would be Python/v1 to Go/v2 switch, which it is not, but which was done with DietPi v8.2 already. However, I didn't know about the APT package until you posted it here 😄.

isarrider, 2023-02-11:
well - then I am glad...
So means leaving docker buildx as it is now
(fine w me,
maybe a short line in the docu would be good?)

MichaIng, 2023-02-11:
I think it was installed together with Docker previously either, same as Docker Compose. And it is good like that, as it is not needed by most users. Good idea to add some hint to the docs about plugins/packages available from the Docker APT repo: https://github.com/MichaIng/DietPi-Docs/issues/850

closed 2023-02-11 by MichaIng