MichaIng/DietPi #6136: Beta branch — R6S kernel appears configured insecurely

thinkpanzer, 2023-02-08:
### Creating a bug report/issue

#### Required Information
- DietPi version |
```
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=14
G_DIETPI_VERSION_RC=0
G_GITBRANCH='beta'
G_GITOWNER='MichaIng'
```
- Distro version | `bullseye`
- Kernel version | `Linux r6s 5.10.110 #1 SMP Sat Dec 3 01:25:15 CST 2022 aarch64 GNU/Linux`
- SBC model | `NanoPi R6S (aarch64)`
- Power supply used | `Anker 20W PD`
- SD card used | `PNY 64GB Elite-X`

#### Additional Information (if applicable)
- Can this issue be replicated on a fresh installation of DietPi? `YES`

#### Steps to reproduce
1. `The R6S image appears to be using the FriendlyElec kernel`
2. `The kernel config is not secure`

#### Expected behaviour
- `dmesg should not be reporting insecure kernel config`

#### Actual behaviour
- `dmesg includes`
```
********************************************************************
**     NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE           **
**                                                                **
**  WRITEABLE clk DebugFS SUPPORT HAS BEEN ENABLED IN THIS KERNEL **
**                                                                **
** This means that this kernel is built to expose clk operations  **
** such as parent or rate setting, enabling, disabling, etc.      **
** to userspace, which may compromise security on your system.    **
**                                                                **
** If you see this message and you are not debugging the          **
** kernel, report this immediately to your vendor!                **
**                                                                **
**     NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE           **
********************************************************************
```

#### Extra details

- `This configuration makes me unwilling to expose the system to a publicly routable network, especially since it suggests that there might also be other suboptimal configuration choices in effect.`

thinkpanzer, 2023-02-08:
See also https://lkml.org/lkml/2022/1/24/3382

Joulinar, 2023-02-08:
Just to avoid any misunderstanding, we don't do any kernel development. It should be mainline kernel. I guess @MichaIng could explain in detail which kernel is used exactly.

MichaIng, 2023-02-08:
It is indeed FriendlyELECs kernel. There is currently no alternative since mainline kernel support is not there and also Armbian has no proper build/kernel support for them (both R5S and R6S) yet.

Don't ask me why FriendlyELEC has this enabled in their official kernel builds. I'll check with their sources and open a PR to disable it.

MichaIng, 2023-02-08:
Interesting, here the commit: https://github.com/friendlyarm/kernel-rockchip/commit/861a024589b3f42ca93877bf74f8a67f59c66ca8

> This doesn't introduce security or safety issues for Android
since debugfs itself is already considered unsafe and mounting it isn't
allowed on user devices.

So this is an Android-only change while this is not feasible for Linux. Since FriendlyELEC user the same branch for Android and Linux, there may be other such Android commits merged into the Linux kernel which shouldn't be there. I'll open an issue, since not sure how to resolve best. Actually splitting branches and reverting ALL commits merged from AOSP/Android repos would be best.

MichaIng, 2023-02-08:
Even more interesting, this commit along with all those merges from Android is coming from Rockchip's Linux branch, so FriendlyELEC is not responsible, but Rockchip: https://github.com/rockchip-linux/kernel/blob/develop-5.10/drivers/clk/clk.c#L3110

MichaIng, 2023-02-08:
Okay, I informed now both, FriendlyELEC and the Rockchip kernel developer. I'm confused that this made it's way into official end user images since a long time already and no-one recognised or complained until our users see the hard to overlook warning emitted by the kernel 🤔.

MichaIng, 2023-02-08:
What you can do is this:
```sh
find /sys/kernel/debug/clk -perm 644 -exec chmod a-w {} +
# or
chmod a-w /sys/kernel/debug/clk/*/clk_{rate,prepare_enable}
```
It is two files in all clk sub directory which are writable by the root user `clk_prepare_enable` and `clk_rate`. This removes the write bit. But the root user can re-add it so it is not really a solution but at best prevents against accidents. This needs to be changed in kernel sources, to that the UNIX permissions have no effect but you get
```
-bash: echo: write error: Permission denied
```
whatever permissions are set when trying to change the value.

thinkpanzer, 2023-02-09:
Thanks for all the updates. I find it somewhat surprising that Rockchip asserts that this kernel config is required for Android; I'm moderately confident it isn't.

My broader concern is what other poor security choices did they make in their kernel config that don't trigger stern warnings.

@MichaIng do you have any idea when we'll see mainline RK3588(S) support?

MichaIng, 2023-02-09:
The commit may be even just pulled in from the Android branch, where, as the commit text states, it has no negative impact for end user systems since the debugfs cannot be mounted anyway, at least not without rooting the phone first, I guess.

So the central problem really is that Linux and Android branches have been merged into one.

> My broader concern is what other poor security choices did they make in their kernel config that don't trigger stern warnings.

Dito, although I do not worry so much about security (I think it is rare that Android and mainline Linux diverge in this regards), but stability and performance. I can imagine that the Android kernel has optimisations and switches set to best serve the Android OS and the common setup of it on Android mobile phones. Even that SBCs use the same CPU architecture, I can imagine that on a regular Linux distro, those changes have a negative impact on kernel features, performance and stability.

However, all speculation so far. I am not even sure how the device clock rate can have a security impact. Probably has something to do with the way hardware random generators work to generate entropy used for cryptographic tasks 🤔. Or I'm on the wrong track 😄.

I hope Rockchip and FriendlyELEC answer soon to get some clarification and a fix for the writable `clk_rate` at a minimum.

MichaIng, 2023-02-11:
Rockchip answered that they are looking into splitting Linux and Android kernel trees with next release. But that means probably next Linux LTS which is not applied by vendors for their current SBCs. FriendlyELEC will most likely stay at Linux 5.10 for NanoPi 5 and 6 series, so they'd need to patch it their end.

thinkpanzer, 2023-02-12:
Well, that doesn't sound great at all.

Did Rockchip give you any indication about whether they were putting the effort into mainline 3588(S) support? Seems to me that mainline would then be the sustainable path fwd vs relying on vendor-patched kernels.

MichaIng, 2023-02-12:
> Did Rockchip give you any indication about whether they were putting the effort into mainline 3588(S) support?

While I totally agree with you, I gave up hoping for this. To me it looks like Rockchip does not aim to support mainline kernel. They create an own branch, vendors use it, and mainline kernel support is done by volunteers or probably also paid kernel developers, but unrelated to Rockchip or any vendor.

For the good of all, the only correct thing to do would be if they would not have an own kernel repository at all, respectively only as feature branch, contributing and syncing everything with mainline kernel directly. Then we wouldn't need to wait for 2-3 years until we finally have an acceptable kernel and images for SBCs with new Rockchip SoCs. Same is true for most other vendors and chip manufacturers as well. I think Allwinner is the only one which seems to care a bit about mainline kernel support?

MichaIng, 2023-02-12:
FYI: https://github.com/friendlyarm/kernel-rockchip/pull/6

thinkpanzer, 2023-02-12:
Yeah, ideally no custom kernel repo or u-boot branches; I'd like to use this hardware for multiple purposes and with different OSes and u-boot is in a number of respects even more important.

Thanks for being so responsive and filling some gaps in my understanding.

MichaIng, 2023-02-12:
> Yeah, ideally no custom kernel repo or u-boot branches

No change to achieve this with FriendlyELEC, but as soon as others implemented support into mainline kernel, we'll switch of course 🙂.

thinkpanzer, 2023-02-18:
Thanks for the update.

They definitely changed the download links. The one that's working for me at the moment is https://1drv.ms/u/s!AkNn1aO7NlsfvljhexlJNkGU2VtO

Does the PR give you any indication whether they made any other kernel config changes that would be security-relevant?

MichaIng, 2023-02-18:
Thanks, that works now. Wasn't it actually Google Drive before?

MichaIng, 2023-02-18:
Kernel build is from January 13th, so we need to wait.

What is this?

[image]

Raspberry Pi bootloader files, even the RPi 4 and pre-4 variants, further split into regular, cut-down, extended and debug variants. This cannot work anywhere else than on Raspberry Pi, and of course those images use U-Boot, so why the hack are these files there?

Next to it:

[image]

Some GRUB files with a config pointing to Linux images which do not exist. Not sure what was experimented with, but that all this stuff is just left (alone with a large number of other useless files spread across the image) on the end user image is... well... good that we purge and re-create the whole rootfs from scratch 😄.

thinkpanzer, 2023-02-18:
Yes, it was Google Drive before. No idea why they changed it to OneDrive.

I am increasingly concerned about Rockchip/FriendlyElec release engineering, and whether even taking their kernel directly is such a good idea 🫤

MichaIng, 2023-02-18:
> I am increasingly concerned about Rockchip/FriendlyElec release engineering

Note that this is common, I saw similar and worse with most other SBC manufacturers. Only Raspberry Pi as own real quality images, Hardkernel the second best, I'd say.

For R5S there is one alternative: https://github.com/pyavitz/binary/releases/tag/images
For R6S sadly none I'm aware of.

thinkpanzer, 2023-02-20:
Looks like RK3588/3588S support is coming in mainline 6.3: https://lore.kernel.org/lkml/4f42519a-d087-416a-b7d6-aa9f63d2c395@app.fastmail.com/

MichaIng, 2023-02-20:
Hopefully NanoPi R6S support in particular as well (device tree at least). Same with U-Boot.

Jensen567, 2023-02-25:
Anyone have a current link or mirror for the R5S downloads that works?  OneDrive links are still down, and I'm not about to install the Baidu software for the Mainland China links.

thinkpanzer, 2023-02-26:
The OneDrive files for R6S disappeared in the last 24 hours as well.

MichaIng, 2023-02-26:
Works here:
- https://dl.friendlyelec.com/nanopir5s
- https://dl.friendlyelec.com/nanopir6s

Jensen567, 2023-02-26:
You are correct, seems they have fixed it for now.  I did manage to find an HTTP server as well but it is very slow.

skyuplam, 2023-03-05:
Not only affect NanoPi R5S/R6S, I have also found the same issue with my Rock Pi 5B.

MichaIng, 2023-03-05:
Interesting. It is based on the same Rockchip kernel sources, but I'd have though that Armbian patched this out. Looks like we should open the same PR on Radxa's kernel branch, so Armbian and everyone else, using it, benefits as well.

MichaIng, 2023-03-05:
- https://github.com/radxa/kernel/pull/69
- https://github.com/radxa/kernel/pull/70

skyuplam, 2023-03-06:
According to GinKage, the flag is disabled via patch in BSP https://github.com/radxa-repo/bsp/blob/main/linux/rockchip/0100-vendor/0008-Revert-ANDROID-clk-Enable-writable-debugfs-files.patch

skyuplam, 2023-03-07:
Yes, I know and what is the way to move forward for DietPi to support Rock Pi 5b?

MichaIng, 2023-03-07:
DietPi does support ROCK 5B? If you mean the issue with insufficient boot partition size, very different topic, so let's keep it over there.

MichaIng, 2023-03-18:
Fixed with: https://github.com/MichaIng/DietPi/commit/738678c
The kernel upgrade is enforced during next DietPi update, as this seems to be a sufficient serious reason. But it is a little risky, as the kernel itself on partition 5 needs to be upgraded as well, otherwise the new kernel modules cannot be loaded with the old kernel. All partitions from 1 to 7 are now re-flashed from a minimal FriendlyELEC base image, which includes SPL, U-Boot, kernel, device tree etc, so everything which is or may be relevant. If someone manually adjusted the U-Boot environment (I'm not even sure where it is located), that will be overwritten, but I think its better this way than skipping it and risking failing boot if the old env does not work with the new bootloader for some reason.

I tried to implement it as failsafe as possible, assuring that base image and modules did successfully download, extract and that the base image as well as the system have the expected 8th partition, before starting with modules package install and flashing the partitions over.

MichaIng, 2023-03-18:
@skyuplam 
The issue is still present on ROCK 5 with latest Armbian kernel?

Otherwise we can add the patch from
https://github.com/radxa-repo/bsp/blob/main/linux/rockchip/0100-vendor/0008-Revert-ANDROID-clk-Enable-writable-debugfs-files.patch
to
https://github.com/armbian/build/tree/main/patch/kernel/rockchip-rk3588-legacy

MichaIng, 2023-03-18:
Nice side-effect of the R5S/R6S kernel upgrade: The console resolution does not fit the screen. On R6S I do not get 1080p console with 1080p screen resolution, on R5S even 1440p both. Not sure where the difference comes from, but it doesn't really matter for the text console.

skyuplam, 2023-03-19:
@MichaIng, yes, the DebugFS is turned on with Armbian_23.02.2_Rock-5b_bullseye_legacy_5.10.110_minimal.img

```
[    6.230289] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
[    6.230315] Linux version 5.10.110-rockchip-rk3588 (root@5280a832a398) (aarch64-linux-gnu-gcc (GNU Toolchain for the A-profile Architecture 8.3-2019.03 (arm-rel-8.36)) 8.3.0, GNU ld (GNU Toolchain for the A-profile Architecture 8.3-2019.03 (arm-rel-8.36)) 2.32.0.20190321) #23.02.2 SMP Fri Feb 17 23:59:20 UTC 2023
[    6.241502] Machine model: Radxa ROCK 5B
[    6.241662] efi: UEFI not found.
[    6.245488] OF: fdt: Reserved memory: failed to reserve memory for node 'drm-logo@00000000': base 0x0000000000000000, size 0 MiB
[    6.245505] OF: fdt: Reserved memory: failed to reserve memory for node 'drm-cubic-lut@00000000': base 0x0000000000000000, size 0 MiB
[    6.245593] Reserved memory: created CMA memory pool at 0x0000000010000000, size 256 MiB
[    6.245597] OF: reserved mem: initialized node cma, compatible id shared-dma-pool
[    6.503630] Zone ranges:
[    6.503637]   DMA      [mem 0x0000000000200000-0x00000000ffffffff]
[    6.503647]   DMA32    empty
[    6.503653]   Normal   [mem 0x0000000100000000-0x00000003ffefffff]
[    6.503660] Movable zone start for each node
[    6.503663] Early memory node ranges
[    6.503669]   node   0: [mem 0x0000000000200000-0x00000000efffffff]
[    6.503677]   node   0: [mem 0x0000000100000000-0x00000003fbffffff]
[    6.503686]   node   0: [mem 0x00000003fc500000-0x00000003ffefffff]
[    6.503692] Initmem setup node 0 [mem 0x0000000000200000-0x00000003ffefffff]
[    6.503698] On node 0 totalpages: 4126720
[    6.503704]   DMA zone: 15352 pages used for memmap
[    6.503707]   DMA zone: 0 pages reserved
[    6.503712]   DMA zone: 982528 pages, LIFO batch:63
[    6.503717]   Normal zone: 49148 pages used for memmap
[    6.503721]   Normal zone: 3144192 pages, LIFO batch:63
[    6.586389] On node 0, zone Normal: 256 pages in unavailable ranges
[    6.586556] psci: probing for conduit method from DT.
[    6.586567] psci: PSCIv1.1 detected in firmware.
[    6.586571] psci: Using standard PSCI v0.2 function IDs
[    6.586577] psci: MIGRATE_INFO_TYPE not supported.
[    6.586582] psci: SMC Calling Convention v1.2
[    6.586990] percpu: Embedded 32 pages/cpu s93416 r8192 d29464 u131072
[    6.587043] pcpu-alloc: s93416 r8192 d29464 u131072 alloc=32*4096
[    6.587050] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 [0] 4 [0] 5 [0] 6 [0] 7 
[    6.587212] Detected VIPT I-cache on CPU0
[    6.587258] CPU features: detected: GIC system register CPU interface
[    6.587263] CPU features: detected: Virtualization Host Extensions
[    6.587273] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    6.587285] alternatives: patching kernel code
[    6.587637] Built 1 zonelists, mobility grouping on.  Total pages: 4062220
[    6.587647] Kernel command line: root=UUID=f8aa49f6-90b1-4f96-bb4b-a490410f4e86 rootwait rootfstype=ext4 splash=verbose console=ttyS2,1500000 console=tty1 consoleblank=0 loglevel=1 ubootpart=91ec68a1-cfa7-954f-a09b-8fe50ae18f21 usb-storage.quirks= net.ifnames=0  cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory swapaccount=1
[    6.589553] Dentry cache hash table entries: 2097152 (order: 12, 16777216 bytes, linear)
[    6.590282] Inode-cache hash table entries: 1048576 (order: 11, 8388608 bytes, linear)
[    6.590289] mem auto-init: stack:off, heap alloc:off, heap free:off
[    6.596218] software IO TLB: mapped [mem 0x00000000ec000000-0x00000000f0000000] (64MB)
[    6.783832] Memory: 15814304K/16506880K available (16576K kernel code, 3432K rwdata, 6320K rodata, 6912K init, 625K bss, 430432K reserved, 262144K cma-reserved)
[    6.783971] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=8, Nodes=1
[    6.784006] ftrace: allocating 63645 entries in 249 pages
[    6.881148] ftrace: allocated 249 pages with 6 groups
[    6.881438] rcu: Hierarchical RCU implementation.
[    6.881443] rcu: 	RCU event tracing is enabled.
[    6.881448] 	Rude variant of Tasks RCU enabled.
[    6.881451] 	Tracing variant of Tasks RCU enabled.
[    6.881458] rcu: RCU calculated value of scheduler-enlistment delay is 30 jiffies.
[    6.886673] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    6.889563] GICv3: GIC: Using split EOI/Deactivate mode
[    6.889569] GICv3: 480 SPIs implemented
[    6.889573] GICv3: 0 Extended SPIs implemented
[    6.889607] GICv3: Distributor has no Range Selector support
[    6.889617] GICv3: 16 PPIs implemented
[    6.889662] GICv3: CPU0: found redistributor 0 region 0:0x00000000fe680000
[    6.889787] ITS [mem 0xfe640000-0xfe65ffff]
[    6.889838] ITS@0x00000000fe640000: allocated 8192 Devices @1001e0000 (indirect, esz 8, psz 64K, shr 0)
[    6.889863] ITS@0x00000000fe640000: allocated 32768 Interrupt Collections @1001f0000 (flat, esz 2, psz 64K, shr 0)
[    6.889871] ITS: using cache flushing for cmd queue
[    6.889910] ITS [mem 0xfe660000-0xfe67ffff]
[    6.889952] ITS@0x00000000fe660000: allocated 8192 Devices @100210000 (indirect, esz 8, psz 64K, shr 0)
[    6.889976] ITS@0x00000000fe660000: allocated 32768 Interrupt Collections @100220000 (flat, esz 2, psz 64K, shr 0)
[... 1239 lines cut ...]
[   26.550636] EXT4-fs (mmcblk1p2): resized to 7340032 blocks
[   37.727279] EXT4-fs (mmcblk1p2): resized to 12582912 blocks
[   48.340962] EXT4-fs (mmcblk1p2): resized to 16777216 blocks
[   58.490774] EXT4-fs (mmcblk1p2): resized to 20447232 blocks
[... cut ...]
[  162.086116] eth0: 0xffffffc0159b0000, 00:e0:4c:68:00:55, IRQ 155
[  164.934776] dwhdmi-rockchip fde80000.hdmi: use tmds mode
[  167.284447] r8125: eth0: link up
[  167.284578] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[  171.825290] ttyFIQ ttyFIQ0: tty_port_close_start: tty->count = 1 port count = 2
[  172.850497] dwhdmi-rockchip fde80000.hdmi: use tmds mode
[  256.651591] ttyFIQ ttyFIQ0: tty_port_close_start: tty->count = 1 port count = 3
[  256.841369] ttyFIQ ttyFIQ0: tty_port_close_start: tty->count = 1 port count = 2
```

MichaIng, 2023-03-19:
@skyuplam 
Could you check some other things on ROCK 5B:
```sh
apt install rng-tools5
sleep 5
systemctl status rngd
```
This reports a running service? If so, this is to be preferred over haveged:
```sh
G_AGP haveged
```
And:
```sh
ls -l /dev/ttyS2
ls -l /dev/ttyFIQ0
```
The first does not exist, the second does. And in case you use a serial console, it is actually on /dev/ttyFIQ0.

Both was true on Orange Pi 5 with same SoC and same kernel, so I'm pretty sure both is true on ROCK 5B as well and regarding the serial console there is another bug in the Armbian boot configs, which try to use ttyS2.

skyuplam, 2023-03-20:
Yes, the rngd is running

<details>
  <summary>Logfile attached. Click to expand!</summary>

```
rngd.service - Start entropy gathering daemon (rngd)
     Loaded: loaded (/lib/systemd/system/rngd.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2023-03-20 04:44:00 UTC; 9min ago
       Docs: man:rngd(8)
   Main PID: 2075 (rngd)
      Tasks: 1 (limit: 18532)
     Memory: 172.0K
        CPU: 5ms
     CGroup: /system.slice/rngd.service
             └─2075 /usr/sbin/rngd -f

Mar 20 04:44:00 rock-5b systemd[1]: Started Start entropy gathering daemon (rngd).
```

</details>

---

I cannot find `/dev/ttyS2` but I can find `/dev/ttyFIQ0`

<details>
  <summary>Logfile attached. Click to expand!</summary>

```
crw-rw-rw- 1 root tty       5,  0 Mar 20 04:43 /dev/tty
crw--w---- 1 root tty       4,  0 Mar 20 04:41 /dev/tty0
crw--w---- 1 root tty       4,  1 Mar 20 04:41 /dev/tty1
crw--w---- 1 root tty       4, 10 Mar 20 04:41 /dev/tty10
crw--w---- 1 root tty       4, 11 Mar 20 04:41 /dev/tty11
crw--w---- 1 root tty       4, 12 Mar 20 04:41 /dev/tty12
crw--w---- 1 root tty       4, 13 Mar 20 04:41 /dev/tty13
crw--w---- 1 root tty       4, 14 Mar 20 04:41 /dev/tty14
crw--w---- 1 root tty       4, 15 Mar 20 04:41 /dev/tty15
crw--w---- 1 root tty       4, 16 Mar 20 04:41 /dev/tty16
crw--w---- 1 root tty       4, 17 Mar 20 04:41 /dev/tty17
crw--w---- 1 root tty       4, 18 Mar 20 04:41 /dev/tty18
crw--w---- 1 root tty       4, 19 Mar 20 04:41 /dev/tty19
crw--w---- 1 root tty       4,  2 Mar 20 04:41 /dev/tty2
crw--w---- 1 root tty       4, 20 Mar 20 04:41 /dev/tty20
crw--w---- 1 root tty       4, 21 Mar 20 04:41 /dev/tty21
crw--w---- 1 root tty       4, 22 Mar 20 04:41 /dev/tty22
crw--w---- 1 root tty       4, 23 Mar 20 04:41 /dev/tty23
crw--w---- 1 root tty       4, 24 Mar 20 04:41 /dev/tty24
crw--w---- 1 root tty       4, 25 Mar 20 04:41 /dev/tty25
crw--w---- 1 root tty       4, 26 Mar 20 04:41 /dev/tty26
crw--w---- 1 root tty       4, 27 Mar 20 04:41 /dev/tty27
crw--w---- 1 root tty       4, 28 Mar 20 04:41 /dev/tty28
crw--w---- 1 root tty       4, 29 Mar 20 04:41 /dev/tty29
crw--w---- 1 root tty       4,  3 Mar 20 04:41 /dev/tty3
crw--w---- 1 root tty       4, 30 Mar 20 04:41 /dev/tty30
crw--w---- 1 root tty       4, 31 Mar 20 04:41 /dev/tty31
crw--w---- 1 root tty       4, 32 Mar 20 04:41 /dev/tty32
crw--w---- 1 root tty       4, 33 Mar 20 04:41 /dev/tty33
crw--w---- 1 root tty       4, 34 Mar 20 04:41 /dev/tty34
crw--w---- 1 root tty       4, 35 Mar 20 04:41 /dev/tty35
crw--w---- 1 root tty       4, 36 Mar 20 04:41 /dev/tty36
crw--w---- 1 root tty       4, 37 Mar 20 04:41 /dev/tty37
crw--w---- 1 root tty       4, 38 Mar 20 04:41 /dev/tty38
crw--w---- 1 root tty       4, 39 Mar 20 04:41 /dev/tty39
crw--w---- 1 root tty       4,  4 Mar 20 04:41 /dev/tty4
crw--w---- 1 root tty       4, 40 Mar 20 04:41 /dev/tty40
crw--w---- 1 root tty       4, 41 Mar 20 04:41 /dev/tty41
crw--w---- 1 root tty       4, 42 Mar 20 04:41 /dev/tty42
crw--w---- 1 root tty       4, 43 Mar 20 04:41 /dev/tty43
crw--w---- 1 root tty       4, 44 Mar 20 04:41 /dev/tty44
crw--w---- 1 root tty       4, 45 Mar 20 04:41 /dev/tty45
crw--w---- 1 root tty       4, 46 Mar 20 04:41 /dev/tty46
crw--w---- 1 root tty       4, 47 Mar 20 04:41 /dev/tty47
crw--w---- 1 root tty       4, 48 Mar 20 04:41 /dev/tty48
crw--w---- 1 root tty       4, 49 Mar 20 04:41 /dev/tty49
crw--w---- 1 root tty       4,  5 Mar 20 04:41 /dev/tty5
crw--w---- 1 root tty       4, 50 Mar 20 04:41 /dev/tty50
crw--w---- 1 root tty       4, 51 Mar 20 04:41 /dev/tty51
crw--w---- 1 root tty       4, 52 Mar 20 04:41 /dev/tty52
crw--w---- 1 root tty       4, 53 Mar 20 04:41 /dev/tty53
crw--w---- 1 root tty       4, 54 Mar 20 04:41 /dev/tty54
crw--w---- 1 root tty       4, 55 Mar 20 04:41 /dev/tty55
crw--w---- 1 root tty       4, 56 Mar 20 04:41 /dev/tty56
crw--w---- 1 root tty       4, 57 Mar 20 04:41 /dev/tty57
crw--w---- 1 root tty       4, 58 Mar 20 04:41 /dev/tty58
crw--w---- 1 root tty       4, 59 Mar 20 04:41 /dev/tty59
crw--w---- 1 root tty       4,  6 Mar 20 04:41 /dev/tty6
crw--w---- 1 root tty       4, 60 Mar 20 04:41 /dev/tty60
crw--w---- 1 root tty       4, 61 Mar 20 04:41 /dev/tty61
crw--w---- 1 root tty       4, 62 Mar 20 04:41 /dev/tty62
crw--w---- 1 root tty       4, 63 Mar 20 04:41 /dev/tty63
crw--w---- 1 root tty       4,  7 Mar 20 04:41 /dev/tty7
crw--w---- 1 root tty       4,  8 Mar 20 04:41 /dev/tty8
crw--w---- 1 root tty       4,  9 Mar 20 04:41 /dev/tty9
crw--w---- 1 root tty     253,  0 Mar 20 04:41 /dev/ttyFIQ0
crw-rw---- 1 root dialout   4, 70 Mar 20 04:41 /dev/ttyS6
```
</details>

MichaIng, 2023-03-21:
Good news, I just wanted to add the patch, but next kernel upgrade should have it fixed for Orange Pi 5 and ROCK 5B as well: https://github.com/armbian/linux-rockchip/pull/9

closed 2023-03-21 by MichaIng