MichaIng/DietPi #6141: first run setup is failed multiple time...

bharatmohanty, 2023-02-09:
first run setup is failed multiple time.cant dietpi-update failed multiple time.issue started two days ago while fresh installation setup. 
[dietpi-firstrun-setup.log](https://github.com/MichaIng/DietPi/files/10693932/dietpi-firstrun-setup.log)

Joulinar, 2023-02-09:
can you share logs from inside `/var/tmp/dietpi/logs`

MichaIng, 2023-02-09:
`dietpi-firstrun-setup.log` looks all good. WiFi was successfully set up, connected, and the system received an IP address via DHCP. If there is no `dietpi-update.log`, then it must have failed either on connection checks, or retrieving the current DietPi version info from GitHub. We'd need the error message/output/prompt then to help.

bharatmohanty, 2023-02-09:
Yes it failed retrieving the current DietPi version info from GitHub...

Joulinar, 2023-02-09:
can you share an exact error message?

bharatmohanty, 2023-02-09:
as i said in terminal it shows  'retrieving the current DietPi version info from GitHub failed'.(timeout). then a window / dialog box prompts with the msg "first run setup  failed. would you like to try it again.. and if this issue continues multiple time report it on 
https://github.com/MichaIng/DietPi/issues/ ". i tried  "dietpi-update" but it shows "no such command exists"...

MichaIng, 2023-02-10:
Does it work when you try it from console manually:
```sh
curl 'https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/version'
```

bharatmohanty, 2023-02-10:
No I haven't tried it manually...I will let you know.

bharatmohanty, 2023-02-11:
this is the response i get on terminal 
 curl: (28) Failed to connect to raw.githubusercontent.com port 443: Connection timed out

Joulinar, 2023-02-11:
Do you use any local DNS server that could block the access?

MichaIng, 2023-02-11:
or a remote one with customisable blocking feature like NextDNS? And does accessing that URL from a browser work?
We saw `raw.githubusercontent.com` on block lists before, so I'd would need to be whitelisted, and at best the author of the list informed about this false entry.

bharatmohanty, 2023-02-12:
No I am not using any local DNS server...but it seems the real culprit was my ISP.... I changed it and now everything is fine. Thanks for your help and support.👍🏼

Joulinar, 2023-02-12:
Ok I'm going to close the issue for now, feel free to reopen if needed.

closed 2023-02-12 by Joulinar