MichaIng/DietPi #6142: DietPi-Software | Docker Compose: near EoL version in use

guahki, 2023-02-09:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- I just installed docker compose from the ``dietpi-software`` helper and realized, I could not use it with ``docker compose`` but have to use ``docker-compose``. This is, as you install the so called "V1" (see note on https://docs.docker.com/compose/) version of docker compose (https://github.com/MichaIng/DietPi/blob/dbfd976eb87e36550afcba5900e3dd93e64aedef/dietpi/dietpi-software#L10402) which will be discontinued this summer. You should switch to installing the "V2" version (which is a plugin and no standalone script, thus ``docker compose`` instead of ``docker-compose``), which is quite simple, as you just ``apt install`` it (assuming, the docker-repo is configured, but this must be the case, as you do it for installing docker).

#### Describe the solution you'd like:

- Switch to the new docker compose V2

Joulinar, 2023-02-09:
Not sure if I understood you correctly but we download v2 version already.

```
 DietPi-Software
─────────────────────────────────────────────────────
 Step: Installing Docker Compose: Manage multi-container Docker applications

[  OK  ] DietPi-Software | cd /tmp/DietPi-Software
[  OK  ] DietPi-Software | curl -sSfL https://github.com/docker/compose/releases/download/v2.16.0/docker-compose-linux-aarch64 -o docker-compose-linux-aarch64
[  OK  ] DietPi-Software | mv docker-compose-linux-aarch64 /usr/local/bin/docker-compose
[  OK  ] DietPi-Software | chmod +x /usr/local/bin/docker-compose
```

```
root@DietPiR6S:~# docker-compose version
Docker Compose version v2.16.0
root@DietPiR6S:~#
```

MichaIng, 2023-02-09:
Indeed Docker v1 was a Python module: https://pypi.org/project/docker-compose/#history

The Docker Compose plugin is a wrapper for the binary we download, AFAIK there are no benefits. But installing it as package has the benefit that upgrades can be done via APT and notifications about it shown in banner. It is also easier to install compared to first checking the latest version via GitHub API etc. I just learned about the package here: https://github.com/MichaIng/DietPi/issues/6135

guahki, 2023-02-09:
Sorry, did not read it thoroghly enough myself. Of course you download V2, but the whole installation as standalone binary (which I confused with V1, sorry for this) is deprecated:

[image]

(see https://docs.docker.com/compose/install/)

Joulinar, 2023-02-09:
at the end it is the very same file download from `https://github.com/docker/compose/releases/download/`. The difference is just the location 

`/usr/local/bin/docker-compose` vs `$DOCKER_CONFIG/cli-plugins/docker-compose`

guahki, 2023-02-09:
Yea, especially this will not be EoL in summer, sorry for this confusion. It is unclear what "This install scenario is no longer supported." means, other than they do not want to bother with issues you might have using this way.

On the other hand I was/am used to the new style ``docker compose`` syntax and assume more people will adopt that, so having to use ``docker-compose`` seems "not the future way to do it".

TL;DR: I just would very kindly ask to review, if the install process as package simplifies your life and allows the use to use the "new style" ``docker compose`` commands

MichaIng, 2023-02-09:
They want to move end users who install Docker Compose manually over to the APT package, which makes sense. But otherwise, as said, the plugin is a wrapper for the very same binary. You can even call `/usr/libexec/docker/cli-plugins/docker-compose` the same way and bypass the `docker compose` command by this. It is even a documented way.

However, as said, the package has the benefit of being easier to update and get notified about available updates, hence the milestone I already applied to the linked issue to switch. Will do that later today.

guahki, 2023-02-09:
Thanks so much for your awesome work with DietPi and your very fast answers.

I will take this as a reminder to not open issues after glancing in some documentation, but to really have a close look. And I look forward to using ``docker compose`` in near future ;)

closed 2023-02-09 by guahki

MichaIng, 2023-02-10:
Migration done with: https://github.com/MichaIng/DietPi/commit/f95bd3a