MichaIng/DietPi #6143: I downloaded the latest beta image "[DietPi_RPi-ARMv8-Bullseye_F2FS.7z](https://dietpi.com/downloads/images/testing/DietPi_RPi-ARMv8-Bullseye_F2FS.7z)" and installed on my RPi 3B which had Samsung Ultra SD card of size 64GB. I was running stable version of dietpi prior which was formatted for this beta install.

MichaIng, 2023-02-09:
I downloaded the latest beta image "[DietPi_RPi-ARMv8-Bullseye_F2FS.7z](https://dietpi.com/downloads/images/testing/DietPi_RPi-ARMv8-Bullseye_F2FS.7z)" and installed on my RPi 3B which had Samsung Ultra SD card of size 64GB. I was running stable version of dietpi prior which was formatted for this beta install.

While running the first boot setup itself I could see that the drive manager is only showing less than 1GB file system size for my / folder and is not allowing to increase though I had a 64GB SD card. My install of additional software was all failing since there is not enough free space.

Again formatted the SD and reinstalled the stable version of Dietpi and I could see that the drive manager shows full 64 GB.

_Originally posted by @sajimenon in https://github.com/MichaIng/DietPi/issues/6130#issuecomment-1424307892_

MichaIng, 2023-02-09:
@sajimenon
I made a new image from this since it is not related to the beta. The F2FS and Btrfs images are experimental for testing purpose only. However, since we aim to offer them as well when proven stable. Can you show the output of:
```sh
cat /var/tmp/dietpi/logs/fs_partition_resize.log
```

MichaIng, 2023-02-09:
> mount: /: mount point is busy.

Ah that is the problem, remounting R/O fails, and online resizing is not supported on F2FS.

Can you try it manually:
```sh
mount -o remount,ro /
resize.f2fs "$G_ROOTFS_DEV"
mount -o remount,rw /
```

Ah, I think the problem is that we are writing this exact log file. Uff this is complicated to solve since we want to have the log file, of course.

Probably a temporary little tmpfs to write the log file to and move to rootfs via exit trap. Or stop writing the log file temporarily, but this makes the script very complicated. I need to think about this...

sajimenon, 2023-02-09:
Nope it does not work. Fails with the first command itself...

***@***.***:~# mount -o remount,ro /
mount: /: mount point is busy.

Warm regards,

Saji Menon
9740210230

MichaIng, 2023-02-22:
> I found that the partition was resized correctly but the F2FS filesystem on the partition was not

Yes, partition resizing of course works (otherwise the whole thing would break with any rootfs type), it is F2FS only which has this limitation of not allowing online resizing. Together with the fact that shrinking F2FS is not possible at all, it makes it quite hard to ship images with F2FS rootfs, but we'll be able to deal with those issues. There are a bunch of ways, I just did not decide yet which one is most robust while having least impact on rootfs of other types.

rekcodocker, 2023-11-23:
I check every now and then.
I still like F2FS support.
I have run an f2fs system running Docker, PiHole, Zigbee2Mqtt, NodeRed and MQTT for almost a year now - without FS problems.

MichaIng, 2025-09-15:
I have a PR which adds F2FS expansion support to our script. But for unknown reasons the RPi kernel fails to mount the root filesystem. Images for testing are here, will investigate the next days what is missing: https://dietpi.com/downloads/images/testing/

Btrfs does not work since the RPi kernel does not have it embedded, hence it would require an initramfs. I'll add the initramfs install on RPi depending on rootfs type, but probably one DietPi release later.

MichaIng, 2025-09-17:
Okay more problems:
- F2FS resizing does not even work on R/O-mounted filesystems, generally not on the rootfs.
- I am currently experimenting with creating a loop device from the root drive and expanding the F2FS there, which generally works (goes through without error). So far, however, this seems to corrupt the filesystem, unless I add the `-s` option. But that one mans that the added space is practically not usable, causing
  ```
  F2FS-fs (mmcblk0p2): invalid blkaddr: 4096, type: 10, run fsck to fix.
  ```
  For every block from that point on. I.e. the maximum block count was not adjusted with `-s`. Running another `resize.f2fs` without `-s` or `fsck.f2fs` with any option afterwards does not seem to adjust the metadata anymore. I.e. once expanded with `-s` means the filesystem becomes unusable, forever. So one wonders why this option even exist. Doing some more tests, but will probably need to give up automatic F2FS expansion. I.e. it might simply strictly need to be done on another Linux system instead.

Checking source code for further info reveals something:
- https://git.kernel.org/pub/scm/linux/kernel/git/jaegeuk/f2fs-tools.git/commit/?id=1ccd831
- https://git.kernel.org/pub/scm/linux/kernel/git/jaegeuk/f2fs-tools.git/commit/?id=f614bcc
- So `-s` option was meant for shinking only in the first place, but only the latest version prevents using it for expansion. And hey, there is hence shrinking support with that flag 🥳. The manual page and CLI help output are totally worthless 😄. But does not help much if there is no way to expand it.

At least Btrfs works perfectly fine now, after adding support to U-Boot. Just on RPi it requires an initramfs at the moment.

MichaIng, 2025-09-17:
Closing this, not a bug, but technically impossible with our resize script. Moving general discussion around F2FS back to #606.

rekcodocker, 2025-09-18:
@MichaIng Thanks a lot for trying it!