MichaIng/DietPi #6145: Update transmission-daemon to 4.0.0

abubaca4, 2023-02-09:
Is there any way to update transmission to newer version?
It contains a lot of [improvements](https://github.com/transmission/transmission/releases/tag/4.0.0)(big performance optimizations and hybrid torrent support)
Waiting for debian repo can take a lot of time
Now I use docker image but it has problem with disk io for me

MichaIng, 2023-02-10:
It can be compiled from source: https://github.com/transmission/transmission#building
The readme is not up-to-date, of course you'd download the v4 archive then. And remember to install cmake and compiler:
```sh
apt install cmake g++
```

abubaca4, 2023-02-13:
So I tried to do it
1. Install a lot of package (cmake make g++ libcurl4-openssl-dev libssl-dev libb64-dev libfmt-dev libdeflate-dev libevent-dev libnatpmp-dev libpsl-dev libminiupnpc-dev libutfcpp-dev). Without *-dev transmission embeds all of libs and result binary is ~100mb
2. Compilation near 2 hours in 4 a53 core on 1.3Ghz (make -j4)
3. Impossible to build web ui, debian version of node lower than required(yes we can take web ui from net and put to right place)
4. Look like make install not add systemd daemon(I just copy binary to /usr/local/bin/ and make new daemon to not broke repo transmission). In unknown reason my daemon not work with Type=notify, only with Type=simple(transmission started, work, but systemd stop it with timeout).

So it take a lot of time(compiling big c++ project on single board) and need additional work to make it work like repo version. In slower device compilation can make bigger problem.

And at all I don't want to do make install, it make files without package manager control.
I think it's good idea make deb package for supporting arch, it save a lot of time for those who want to install the new version. In my tests it's really big improvement to performance, earlier I has cpu load near 70% in 1.3Ghz, now 50-60% in 0.6-0.8Ghz with the same speed.

MichaIng, 2023-02-13:
> Impossible to build web ui, debian version of node lower than required

Try to install Node.js via `dietpi-software`, which pulls the very latest version.

For the systemd and Type=notify support, likely `libsystemd-dev` needs to be installed and in case also a build flag set. But it is common that optional features are silently skipped if related headers are missing, and added when they are there.

> And at all I don't want to do make install, it make files without package manager control.

debhelper can be used to do the compiling, common post-comping steps like `strip`, and DEB packaging all in once. That is the tool Debian uses for its own repo. Can be also done manually but then its some more work, e.g. check out the build script we use for GMediaRender: https://github.com/MichaIng/DietPi/blob/master/.build/software/gmediarender/build.bash

> I think it's good idea make deb package for supporting arch

Now you know that it is quite some work, trail&error and takes time. Debian by design never updates software within stable Debian releases, but only backports security and bug fix patches. We are only hosting own DEB packages in rare cases when Debian is far behind, including testing (currently Bookworm), major enhancements included in the new version, and a lot of requests for this are given.

We can move this to discussions and have others vote for DEB packages hosted on our server.

abubaca4, 2023-02-16:
So earlier I had near x2 worse performance with docker in armbian. Now I tested in dietpi with net=host and apparmor=unconfined and not see big difference
But it still has worse performance than in usual pc with the same drive(MG04ACA400N) and I don't see bottleneck.
Drive connected by usb-sata adapter with uasp support(hdparm test give normal result). CPU core(transmission is single core app except verify data) not loaded at 100% and frequency is not in max bust. iotop not show 100% load of the drive. But in common pc I can load my internet channel 150 mbit download and 30 mbit upload. On singleboard(nanopineo3) I got 150 download if I stop all upload. If upload not stopped I got 60-80 mbit download. So cpu not loaded fully, iotop not show problem with drive, where is bottleneck?

Summarizing I still think that native app better than container especially in low perfomance device. But now in this case I can't find enough performance difference to justify supporting own version of package. But maybe dietpi-software should install docker container that has settings like current daemon from repo.

This just about why I use transmission, not qbittorrent. You can feel free skip it.
Difference between transmission(with own bittorrent realization) and clients based on libtorrent-rasterbar(qBittorrent Deluge and etc). rasterbar can use many cores, transmission not. But transmission much more optimized and give more perfomance on one core than rasterbar on 2-3 equal core. So transmission 3.0 on very slow cpu MT7621AT load 100mbit channel. 4.0.0 can be double faster(but still one core). rasterbar has problem with big count of torrents, if you add 2600 torrents it can tage 7gb of ram and gui will be vary laggy. Transmission can run ~15k seeding torrents with 700mb of ram and without control issues. rasterbar support all new features like bittorrent v2 and other. Transmission not, but they spended a lot of time to replace codebase from C90 to C++17 and now fixing bags, new features will be later(but repository has higth commit activity and I hope improvmets will be soon).
So I think for comon user now better to use qbittorrent. But if it has a lot of torrents use transmission is good idea. I think that on slow cpu like routers and single board better to use transmission, but not in case when cpu has 4 or more cores and torrent count not big.

closed 2023-02-16 by abubaca4