MichaIng/DietPi #6147: dietpi-imager fails on external drive

DrCWO, 2023-02-11:
Just a short question:
Is there a chance to write a created image with dietpi-imager on a separate drive?

I mounted the external drive (/dev/sdb1) to /mnt1 and entered /mnt1/backup as destination.
Backup source is /sda. But dietpi-imager says that there is not enough space???

Support would be appreciated. 
Best DrCWO

```
root@DietPi:~# df
Filesystem      1K-blocks      Used  Available Use% Mounted on
/dev/root        15160408    755260   13766232   6% /
devtmpfs           937676         0     937676   0% /dev
tmpfs              972364         0     972364   0% /dev/shm
tmpfs              388948      5524     383424   2% /run
tmpfs                5120         0       5120   0% /run/lock
tmpfs             1048576         4    1048572   1% /tmp
tmpfs               51200         8      51192   1% /var/log
/dev/mmcblk0p1     129039     32550      96489  26% /boot
/dev/sdb1      1953512032 762353608 1191158424  40% /mnt1

root@DietPi:~# export SKIP_ARCHIVE=1
root@DietPi:~# bash -c "$(curl -sSfL 'https://raw.githubusercontent.com/MichaIng/DietPi/dev/.build/images/dietpi-imager')"
[  OK  ] DietPi-Imager | cd /root
[ INFO ] DietPi-Imager | Checking for required APT packages: parted fdisk zerofree p7zip
[ INFO ] DietPi-Imager | Unmounting all filesystems below selected /dev/sda ...
[ INFO ] DietPi-Imager | MBR partition table detected
[ INFO ] DietPi-Imager | e2fsck -fpD /dev/sda2, please wait...
/dev/sda2: Inode 140078 extent tree (at level 1) could be shorter.  IGNORED.
/dev/sda2: 117655/758880 files (0.7% non-contiguous), 2982036/3047424 blocks
[  OK  ] DietPi-Imager | e2fsck -fpD /dev/sda2
[  OK  ] DietPi-Imager | mkdir /tmp/DietPi-Imager_mnt
[  OK  ] DietPi-Imager | mount /dev/sda2 /tmp/DietPi-Imager_mnt
[  OK  ] DietPi-Imager | rm -f /tmp/DietPi-Imager_mnt/root/.bash_history /tmp/DietPi-Imager_mnt/home/*/.bash_history /tmp/DietPi-Imager_mnt/var/lib/dhcp/*.leases
[  OK  ] DietPi-Imager | sync
[  OK  ] DietPi-Imager | sleep 1
[  OK  ] DietPi-Imager | umount -R /tmp/DietPi-Imager_mnt
[  OK  ] DietPi-Imager | partprobe /dev/sda
[  OK  ] DietPi-Imager | partx -u /dev/sda
[ INFO ] DietPi-Imager | e2fsck -fpD /dev/sda2, please wait...
/dev/sda2: Inode 140078 extent tree (at level 1) could be shorter.  IGNORED.
/dev/sda2: 117655/758880 files (0.7% non-contiguous), 2982036/3047424 blocks
[  OK  ] DietPi-Imager | e2fsck -fpD /dev/sda2
[  OK  ] DietPi-Imager | tune2fs -O ^has_journal /dev/sda2
[  OK  ] DietPi-Imager | sync
[  OK  ] DietPi-Imager | sleep 1
[  OK  ] DietPi-Imager | tune2fs -O has_journal /dev/sda2
[  OK  ] DietPi-Imager | sync
[  OK  ] DietPi-Imager | sleep 1
[ INFO ] DietPi-Imager | Shrinking last filesystem to minimum size...
resize2fs 1.46.2 (28-Feb-2021)
Resizing the filesystem on /dev/sda2 to 3046549 (4k) blocks.
The filesystem on /dev/sda2 is now 3046549 (4k) blocks long.

resize2fs 1.46.2 (28-Feb-2021)
Resizing the filesystem on /dev/sda2 to 3046548 (4k) blocks.
The filesystem on /dev/sda2 is now 3046548 (4k) blocks long.

resize2fs 1.46.2 (28-Feb-2021)
The filesystem is already 3046548 (4k) blocks long.  Nothing to do!

[  OK  ] DietPi-Imager | Leaving last filesystem size at 11905 MiB
[  OK  ] DietPi-Imager | resize2fs /dev/sda2 3047424
[ INFO ] DietPi-Imager | e2fsck -fpD /dev/sda2, please wait...
/dev/sda2: Inode 140078 extent tree (at level 1) could be shorter.  IGNORED.
/dev/sda2: 117655/758880 files (0.7% non-contiguous), 2982036/3047424 blocks
[  OK  ] DietPi-Imager | e2fsck -fpD /dev/sda2
[ INFO ] DietPi-Imager | Overriding filesystems free space with zeros to purge removed data and allow better compression...
[  OK  ] DietPi-Imager | mount /dev/sda1 /tmp/DietPi-Imager_mnt
[ INFO ] DietPi-Imager | fstrim -v --quiet-unsupported /tmp/DietPi-Imager_mnt, please wait...
[FAILED] DietPi-Imager | fstrim -v --quiet-unsupported /tmp/DietPi-Imager_mnt
[  OK  ] DietPi-Imager | sync
[  OK  ] DietPi-Imager | sleep 1
[  OK  ] DietPi-Imager | umount -R /tmp/DietPi-Imager_mnt
[ INFO ] DietPi-Imager | fsck /dev/sda1, please wait...
fsck from util-linux 2.36.1
fsck.fat 4.2 (2021-01-31)
/dev/sda1: 364 files, 65078/258077 clusters
[  OK  ] DietPi-Imager | fsck /dev/sda1
[ INFO ] DietPi-Imager | zerofree -v /dev/sda2, please wait...
0/65388/3047424
[  OK  ] DietPi-Imager | zerofree -v /dev/sda2
[  OK  ] DietPi-Imager | rmdir /tmp/DietPi-Imager_mnt
[FAILED] DietPi-Imager | Free space check: path=. | available=13444 MiB | required=13941 MiB
root@DietPi:~#
```

MichaIng, 2023-02-11:
> entered /mnt1/backup as destination

The image name is not meant to be a path but a filename only, at least it is handled like that. The image will always be created within the current directory. So navigate there first:
```sh
/mnt1
```
However, I checked and indeed when you enter an actual path, aside of the free space check, it should work mostly as intended. If the archive is created, it will probably have the directories added as well, though. I'll verify it and adjust the description in the menu. Easiest to assure that all happens as intended is probably to strip the path from the input and `cd` to it.

DrCWO, 2023-02-11:
Just some feedback for your information. 
My external drive I mounted on /mnt1 was a NTFS USB stick.

I did cd to /mnt1 and started from there --> Did not work, a lot of continuous error messages probably caused by NTFS

MichaIng, 2023-02-11:
Can you paste the exact errors?

DrCWO, 2023-02-11:
I would if I could but I am in the middle of building a new release and don't want to get back to this step.
Will do it later...

DrCWO, 2023-02-11:
This is what happens until I press CTRL-C and I pressed quick ;-)

```
root@DietPi:/mnt1# bash -c "$(curl -sSfL 'https://raw.githubusercontent.com/MichaIng/DietPi/dev/.build/images/dietpi-imager')"
[  OK  ] DietPi-Imager | cd /mnt1
[ INFO ] DietPi-Imager | Checking for required APT packages: parted fdisk zerofree p7zip
[ INFO ] DietPi-Imager | Unmounting all filesystems below selected /dev/sda ...
[ INFO ] DietPi-Imager | MBR partition table detected
[ INFO ] DietPi-Imager | e2fsck -fpD /dev/sda2, please wait...
/dev/sda2: Inode 140078 extent tree (at level 1) could be shorter.  IGNORED.
/dev/sda2: 109953/962880 files (0.5% non-contiguous), 948642/3855744 blocks
[  OK  ] DietPi-Imager | e2fsck -fpD /dev/sda2
[  OK  ] DietPi-Imager | mkdir /tmp/DietPi-Imager_mnt
[  OK  ] DietPi-Imager | mount /dev/sda2 /tmp/DietPi-Imager_mnt
[  OK  ] DietPi-Imager | rm -f /tmp/DietPi-Imager_mnt/root/.bash_history /tmp/DietPi-Imager_mnt/home/*/.bash_history /tmp/DietPi-Imager_mnt/var/lib/dhcp/*.leases
[  OK  ] DietPi-Imager | sync
[  OK  ] DietPi-Imager | sleep 1
[  OK  ] DietPi-Imager | umount -R /tmp/DietPi-Imager_mnt
[  OK  ] DietPi-Imager | partprobe /dev/sda
[  OK  ] DietPi-Imager | partx -u /dev/sda
[ INFO ] DietPi-Imager | e2fsck -fpD /dev/sda2, please wait...
/dev/sda2: Inode 140078 extent tree (at level 1) could be shorter.  IGNORED.
/dev/sda2: 109953/962880 files (0.5% non-contiguous), 948642/3855744 blocks
[  OK  ] DietPi-Imager | e2fsck -fpD /dev/sda2
[  OK  ] DietPi-Imager | tune2fs -O ^has_journal /dev/sda2
[  OK  ] DietPi-Imager | sync
[  OK  ] DietPi-Imager | sleep 1
[  OK  ] DietPi-Imager | tune2fs -O has_journal /dev/sda2
[  OK  ] DietPi-Imager | sync
[  OK  ] DietPi-Imager | sleep 1
[ INFO ] DietPi-Imager | Shrinking last filesystem to minimum size...
tee: resize2fs_out: Permission denied
resize2fs 1.46.2 (28-Feb-2021)
Resizing the filesystem on /dev/sda2 to 974669 (4k) blocks.
Begin pass 2 (max = 187632)
Relocating blocks             XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 3 (max = 118)
Scanning inode table          XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 4 (max = 17011)
Updating inode references     XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
The filesystem on /dev/sda2 is now 974669 (4k) blocks long.

grep: resize2fs_out: No such file or directory
grep: resize2fs_out: No such file or directory
tee: resize2fs_out: Permission denied
resize2fs 1.46.2 (28-Feb-2021)
Resizing the filesystem on /dev/sda2 to 967937 (4k) blocks.
Begin pass 2 (max = 6726)
Relocating blocks             XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 3 (max = 30)
Scanning inode table          XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
The filesystem on /dev/sda2 is now 967937 (4k) blocks long.

grep: resize2fs_out: No such file or directory
grep: resize2fs_out: No such file or directory
tee: resize2fs_out: Permission denied
resize2fs 1.46.2 (28-Feb-2021)
Resizing the filesystem on /dev/sda2 to 967927 (4k) blocks.
Begin pass 2 (max = 10)
Relocating blocks             XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 3 (max = 30)
Scanning inode table          XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
The filesystem on /dev/sda2 is now 967927 (4k) blocks long.

grep: resize2fs_out: No such file or directory
grep: resize2fs_out: No such file or directory
tee: resize2fs_out: Permission denied
resize2fs 1.46.2 (28-Feb-2021)
The filesystem is already 967927 (4k) blocks long.  Nothing to do!

[... 207 lines cut ...]
resize2fs 1.46.2 (28-Feb-2021)
The filesystem is already 967927 (4k) blocks long.  Nothing to do!

grep: resize2fs_out: No such file or directory
grep: resize2fs_out: No such file or directory
tee: resize2fs_out: Permission denied
resize2fs 1.46.2 (28-Feb-2021)
The filesystem is already 967927 (4k) blocks long.  Nothing to do!

grep: resize2fs_out: No such file or directory
grep: resize2fs_out: No such file or directory
tee: resize2fs_out: Permission denied
resize2fs 1.46.2 (28-Feb-2021)
The filesystem is already 967927 (4k) blocks long.  Nothing to do!

grep: resize2fs_out: No such file or directory
grep: resize2fs_out: No such file or directory
tee: resize2fs_out: Permission denied
resize2fs 1.46.2 (28-Feb-2021)
The filesystem is already 967927 (4k) blocks long.  Nothing to do!

[  OK  ] DietPi-Imager | rmdir /tmp/DietPi-Imager_mnt

root@DietPi:/mnt1#
```

MichaIng, 2023-02-11:
> tee: resize2fs_out: Permission denied

Looks like the NTFS drive is mounted read-only, and likely also the ntfs-3g driver for UNIX permissions emulation is not installed, respectively the drive is not mounted with emulated permissions. Did you mount it via `dietpi-drive_manager`? Else
```sh
apt install ntfs-3g
```
Then add `permissions` as another mount option.

It doesn't solve your issue, since you need to write to this drive anyway, but the resize2fs output is now stored to /tmp instead of the current directory, so you do not end up in a loop here: https://github.com/MichaIng/DietPi/commit/fc81aa1