MichaIng/DietPi #6150: Donations via cryptocurrencies

tcables, 2023-02-11:
Hi, I would prefer to donate via cryptocurrencies such as bitcoin or Ethereum. Is this a possibility?

MichaIng, 2023-02-11:
Coincidentally I was just offered an ENS domain, but we do not have an Ethereum or Bitcoin account. BAT only 😄. But I'll have a look into it. Any recommendations where to create a crypto wallet, as a German resident?

tcables, 2023-02-12:
I am a total n00b when it comes to the crypto universe. I really hate fiat/visa/mc/paypal with a passion, and want the world to move towards people fully controlling their money. I personally use the offline version of My Ether Wallet (https://github.com/MyEtherWallet/MyEtherWallet/releases/), but hope to switch to a hardware one (Trezor or Ledger) which support a ton of coins and 2FA/password vault.

Do you have a BAT address? I looked a little bit but it seems its tied to Brave browser pretty much, but i do know its possible to 'send' BAT

MichaIng, 2023-02-12:
Great, I'll look into that when I find time.

Yes BAT is tied to the Brave browser advertisement system. I think it isn't even possible to send us BAT without having them collected with and donating through Brave, since I'm unable to connect the Brave/BAT wallet to a custodial. Only two are supported for this (Gemini and Uphold) and both are not supported in my resident country... Brave is negotiating with further partners, but no ETA. The idea behind it is good, but practically for content creators it cannot be taken serious for such major practical issues, sadly.

Phil1988, 2023-03-26:
I want to donate via brave, but it sais that dietpi is an unverified author:

[image]

How to proceed?

MichaIng, 2023-03-26:
I recognised this as well. Brave changed it in a way that one must connect the Brave Rewards account with a custodial crypto wallet, while they only support Gemini and Uphold. Both are not supported in Germany and the country must match to be verified. So there is, since we have this account, no way to get the BAT collected in the account, and now there isn't even any way to donate BAT to any content creator located in Germany 😞.

I'm checking for updates by times, and they say they are doing negotiations with custodials and that it is do difficult due to German regulations. Not sure, recently Japan was re-added with a Japan-only custodial, but otherwise process is very slow.

Phil1988, 2023-03-29:
That is especially sad because I started using brave because of dietpi - to contribute every BAT I collect to you guys.
It wouldnt be too much, but the users sum up doing this it might get to a point where its enough :)

Unfortunate that there is currently no way to contribute.
Would be nice if you can update here if it changes, so I can fire all my collected BATs to you :)

MichaIng, 2023-03-29:
I'll keep you updated once German creators are able to receive BAT again.

HyperCriSiS, 2023-04-04:
Hi, I recommend https://play.google.com/store/apps/details?id=com.wallet.crypto.trustapp

Or

https://play.google.com/store/apps/details?id=io.metamask

MichaIng, 2023-04-04:
Are you sure that I use an Android phone and if so that I want to manage a crypto wallet with this Android phone? 😛

But thanks for the recommendations 🙂:
- https://www.binance.com/ (the custodian, right?) > https://trustwallet.com/ (non-custodial wallet of Binance, right?)
- https://metamask.io/