MichaIng/DietPi #6153: Failure to Uninstall Docker-Compose

bccarbone, 2023-02-12:
**Issue:**

1) I tried to uninstall docker-compose and was given the below errors.
2) I tried to reinstall in hopes of then being able to do an uninstall.  Didn't work.
3) I tried to restore a backup in hopes I could go back to yesterday's state and that failed.

**Questions Are As Follows:**

1) How do I fix this issue?  
2) Why did the restore of my backup fail?
3) Appreciate the help, I'm pretty new to this, but willing to learn.  I hope I didn't mess up my install of Diet-Pi. 

**Diet-Pi Restore Backup Failure**

File: [dietpi-backup failure.txt](https://github.com/MichaIng/DietPi/files/10714973/dietpi-backup.failure.txt)

**Docker-Compose Uninstall Errors**

- DietPi version | v8.13.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPi 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `pip3 uninstall -y docker-compose`
- Exit code      | 1
- Software title | DietPi-Software

│ Traceback (most recent call last):
│   File "/usr/local/bin/pip3", line 5, in <module>
│     from pip._internal.cli.main import main
│   File "/usr/local/lib/python3.9/dist-packages/pip/_internal/cli/main.py", line 9, in <module>
│     from pip._internal.cli.autocompletion import autocomplete
│   File "/usr/local/lib/python3.9/dist-packages/pip/_internal/cli/autocompletion.py", line 10, in <module>
│     from pip._internal.cli.main_parser import create_main_parser
│   File "/usr/local/lib/python3.9/dist-packages/pip/_internal/cli/main_parser.py", line 9, in <module>
│     from pip._internal.build_env import get_runnable_pip
│   File "/usr/local/lib/python3.9/dist-packages/pip/_internal/build_env.py", line 20, in <module>
│     from pip._internal.cli.spinners import open_spinner
│   File "/usr/local/lib/python3.9/dist-packages/pip/_internal/cli/spinners.py", line 9, in <module>
│     from pip._internal.utils.logging import get_indentation
│   File "/usr/local/lib/python3.9/dist-packages/pip/_internal/utils/logging.py", line 29, in <module>
│     from pip._internal.utils.misc import ensure_dir
│   File "/usr/local/lib/python3.9/dist-packages/pip/_internal/utils/misc.py", line 42, in <module>
│     from pip._internal.locations import get_major_minor_version
│   File "/usr/local/lib/python3.9/dist-packages/pip/_internal/locations/__init__.py", line 67, in <module>
│     from . import _distutils
│   File "/usr/local/lib/python3.9/
dist-packages/pip/_internal/locations/_distutils.py", line 20, in <module>
from distutils.cmd import Command as DistutilsCommand
│ ModuleNotFoundError: No module named 'distutils.cmd'

MichaIng, 2023-02-12:
It looks like parts of Python are uninstalled already, at least the `python3-dev` package with its dependencies. Try this:
```sh
apt install python3-distutils
```
Afterwards you can purge this package again, or Python all together if you do not need it for something else.

bccarbone, 2023-02-12:
Thanks, that did the trick.

MichaIng, 2023-02-12:
Was that `docker-compose` Python module actually still installed? We install the Go v2 binary since DietPi v8.2 but try to uninstall the Python module whenever `pip` is installed. It think we can skip that step by checking first whether e.g. the `docker-compose` command is actually available (part of the module).

bccarbone, 2023-02-12:
I'm not sure exactly.  Just curious, why did the restore of my backup fail?  I used dietpi-backup and performed a restore from the menu.  Additionally is there a guide on how to restore a backup if my dietpi installation becomes unbootable?  Thanks, just trying to make sure I'm doing all of this properly.

bccarbone, 2023-02-12:
Ok, so I tried reinstalling docker and docker compose.  It looks like I can no longer install docker compose.  Any way to fix this?

DietPi-Software
─────────────────────────────────────────────────────
 Step: Installing Docker Compose: Manage multi-container Docker applications

[ INFO ] DietPi-Software | pip3 uninstall -y docker-compose, please wait...
WARNING: Skipping docker-compose as it is not installed.
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
[  OK  ] DietPi-Software | pip3 uninstall -y docker-compose
[ INFO ] DietPi-Software | APT install docker-compose-plugin, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
The following NEW packages will be installed:
  docker-compose-plugin
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 8678 kB of archives.
After this operation, 45.8 MB of additional disk space will be used.
Get:1 https://download.docker.com/linux/debian bullseye/stable arm64 docker-compose-plugin arm64 2.16.0-1~debian.11~bullseye [8678 kB]
debconf: delaying package configuration, since apt-utils is not installed
Fetched 8678 kB in 1s (11.9 MB/s)
                                 Selecting previously unselected package docker-compose-plugin.
(Reading database ... 53386 files and directories currently installed.)
Preparing to unpack .../docker-compose-plugin_2.16.0-1~debian.11~bullseye_arm64.deb ...
Unpacking docker-compose-plugin (2.16.0-1~debian.11~bullseye) ...
Setting up docker-compose-plugin (2.16.0-1~debian.11~bullseye) ...
[  OK  ] DietPi-Software | APT install docker-compose-plugin

Joulinar, 2023-02-12:
docker compose has been installed from apt package now

bccarbone, 2023-02-12:
When I type: docker-compose, I get this message:

bash: docker-compose: command not found

Why can I not access this command?

MichaIng, 2023-02-12:
Jep, all perfectly fine. It needs to be called now without the dash, as it is a Docker plugin:
```sh
docker compose
```

bccarbone, 2023-02-12:
Gotcha, that appears to be working.  I could have sworn docker-compose is what I've used in the past.  Odd.

MichaIng, 2023-02-12:
Yes, it was installed as standalone binary before, now as APT package and Docker plugin. See changelog: https://github.com/MichaIng/DietPi/blob/3a98e7b/CHANGELOG.txt#L25

Joulinar, 2023-02-12:
Yeah Docker introduced some changed and Compose is becoming a plugin now https://docs.docker.com/compose/compose-v2/
we just follow official change at Docker side.

bccarbone, 2023-02-12:
Ahh...that's what I get for upgrading and not reading the changelog first.  Thank you :-)

bccarbone, 2023-02-12:
In regards to my restore failure from a backup, should I log as a separate ticket?  Additionally, if I want to understand how to restore my backup if for some reason my dietpi server no longer boots, should that be a forum question?  Thanks.

Joulinar, 2023-02-12:
Usually `dietpi-backup` will perform a full system restore. Best is to have it stored on an external usb drive/stick. This way it can be automatically restored on a fresh install.

MichaIng, 2023-02-12:
Regarding restore errors, probably a dedicated issue is easier. Add some info, error logs/output you get.

Restoring a dietpi-backup when the system does not boot can be done from another system. But one needs to call `rsync` manually with correct options. A better approach is probably to check the reason for the boot failure and restore the individual related file(s). It's not such much which can be responsible, broken kernel, device tree oder initramfs, all located in `/boot`.

> This way it can be automatically restored on a fresh install.

That is also a good hint. You can flash a new image, attach it together with the backup, either copied to the same filesystem or onto a dedicated drive, set a related flag in `/boot/dietpi.txt` and can then select the backup to restore on boot, or even have this all automated.

MichaIng, 2023-02-12:
Fixed with: https://github.com/MichaIng/DietPi/commit/90127ec

closed 2023-02-15 by MichaIng