MichaIng/DietPi #6154: Logitech Media Server:Can't player m3u8

yu13740000, 2023-02-12:
Issue:
Logitech Media Server:Can't player m3u8 internet radio,But MPD can player。
https://cnlive.cnr.cn/hls/hjyyzs.m3u8
http://live.xmcdn.com/live/1205/64.m3u8

 Required Information
- DietPi version | v8.13.2 (MichaIng/master)
- Distro version | bullseye
- Kernel version | Linux DietPi 5.3.0-aml-g12
- SBC model | Generic Amlogic S905 (aarch64)
- Power supply used | (EG: 5V 1A RAVpower)
- SD card used | (EG: SanDisk ultra)

Additional Information (if applicable)
- Software title | Logitech Media Server - aka LMS, Squeezebox Server
- Was the software title installed freshly or updated/migrated? | updated
- Can this issue be replicated on a fresh installation of DietPi?  | yes

MichaIng, 2023-02-12:
Many thanks for your report.

Can you give us some more details, like the steps to reproduce the issue (bug report template has its reason), in case any error messages you see, also service logs:
```sh
journalctl -u logitechmediaserver
```
and matching ones within `/var/log/squeezeboxserver`.

Since it looks more an issue of the software, than of our implementation, also check their logs whether this playlist format is supposed to work. Some resources:
- Official forum: https://forums.slimdevices.com/
- Source code repo: https://github.com/Logitech/slimserver

yu13740000, 2023-02-12:
root@DietPi:~# journalctl -u logitechmediaserver
-- Journal begins at Sun 2023-02-12 14:50:16 GMT, ends at Sun 2023-02-12 15:03:17 GMT. --
Feb 12 14:50:29 DietPi systemd[1]: Started Logitech Media Server.
Feb 12 14:50:39 DietPi squeezeboxserver[976]: [23-02-12 14:50:35.9062] main::init (377) Starting Logitech Media Server (v8.3.0, 1667251155, Fri 04 Nov 2022 09:16:39 AM CET) perl 5.032001 - aarch64-linux-gnu-thread-multi
Feb 12 14:50:39 DietPi squeezeboxserver[976]: [23-02-12 14:50:39.4539] Slim::Plugin::FullTextSearch::Plugin::_initPopularTerms (587) Fulltext index missing or outdated - re-building
Feb 12 14:50:39 DietPi squeezeboxserver[976]: [23-02-12 14:50:39.4549] Slim::Plugin::FullTextSearch::Plugin::_rebuildIndex (475) Starting fulltext index build
Feb 12 14:50:39 DietPi squeezeboxserver[976]: [23-02-12 14:50:39.4558] Slim::Plugin::FullTextSearch::Plugin::_rebuildIndex (483) Initialize fulltext table
Feb 12 14:50:39 DietPi squeezeboxserver[976]: [23-02-12 14:50:39.4617] Slim::Plugin::FullTextSearch::Plugin::_rebuildIndex (496) Create fulltext index for tracks
Feb 12 14:50:39 DietPi squeezeboxserver[976]: [23-02-12 14:50:39.4639] Slim::Plugin::FullTextSearch::Plugin::_rebuildIndex (506) Create fulltext index for albums
Feb 12 14:50:39 DietPi squeezeboxserver[976]: [23-02-12 14:50:39.4658] Slim::Plugin::FullTextSearch::Plugin::_rebuildIndex (515) Create fulltext index for contributors
Feb 12 14:50:39 DietPi squeezeboxserver[976]: [23-02-12 14:50:39.4813] Slim::Plugin::FullTextSearch::Plugin::_rebuildIndex (525) Create fulltext index for playlists
Feb 12 14:50:39 DietPi squeezeboxserver[976]: [23-02-12 14:50:39.5394] Slim::Plugin::FullTextSearch::Plugin::_rebuildIndex (540) Optimize fulltext index
Feb 12 14:50:39 DietPi squeezeboxserver[976]: [23-02-12 14:50:39.5452] Slim::Plugin::FullTextSearch::Plugin::_rebuildIndex (555) Fulltext index build done!
Feb 12 14:50:39 DietPi squeezeboxserver[976]: [23-02-12 14:50:39.9002] main::checkDataSource (1107) Warning: Schema updated or no media found in the database, initiating scan.
Feb 12 14:50:42 DietPi squeezeboxserver[1025]: [23-02-12 14:50:42.0398] main::main (202) Starting Logitech Media Server scanner (v8.3.0, 1667251155, Fri 04 Nov 2022 09:16:39 AM CET) perl 5.032001
Feb 12 14:50:44 DietPi squeezeboxserver[1025]: [23-02-12 14:50:44.8168] Slim::Music::Import::runImporter (579) Starting Slim::Media::MediaFolderScan scan
Feb 12 14:50:44 DietPi squeezeboxserver[1025]: [23-02-12 14:50:44.8182] Slim::Utils::Scanner::Local::rescan (179) Discovering audio files in /mnt/dietpi_userdata/Music
Feb 12 14:50:44 DietPi squeezeboxserver[1025]: [23-02-12 14:50:44.8796] Slim::Utils::Scanner::Local::__ANON__ (191) Start processing found tracks
Feb 12 14:50:44 DietPi squeezeboxserver[1025]: [23-02-12 14:50:44.8807] Slim::Utils::Scanner::Local::__ANON__ (199) Connect do DB
Feb 12 14:50:44 DietPi squeezeboxserver[1025]: [23-02-12 14:50:44.8820] Slim::Utils::Scanner::Local::__ANON__ (202) Get latest ID
Feb 12 14:50:44 DietPi squeezeboxserver[1025]: [23-02-12 14:50:44.8838] Slim::Utils::Scanner::Local::__ANON__ (225) Build temporary table for new tracks
Feb 12 14:50:44 DietPi squeezeboxserver[1025]: [23-02-12 14:50:44.8862] Slim::Utils::Scanner::Local::__ANON__ (245) Build temporary table for changed tracks
Feb 12 14:50:44 DietPi squeezeboxserver[1025]: [23-02-12 14:50:44.8885] Slim::Utils::Scanner::Local::__ANON__ (282) Get deleted tracks count
Feb 12 14:50:44 DietPi squeezeboxserver[1025]: [23-02-12 14:50:44.8896] Slim::Utils::Scanner::Local::__ANON__ (289) Get new tracks count
Feb 12 14:50:44 DietPi squeezeboxserver[1025]: [23-02-12 14:50:44.8913] Slim::Utils::Scanner::Local::__ANON__ (294) Get changed tracks count
Feb 12 14:50:44 DietPi squeezeboxserver[1025]: [23-02-12 14:50:44.8925] Slim::Utils::Scanner::Local::deleteTracks (370) Removing deleted audio files (0)
Feb 12 14:50:44 DietPi squeezeboxserver[1025]: [23-02-12 14:50:44.8938] Slim::Utils::Scanner::Local::updateTracks (537) Rescanning changed audio files (0)
Feb 12 14:50:44 DietPi squeezeboxserver[1025]: [23-02-12 14:50:44.8950] Slim::Utils::Scanner::Local::addTracks (453) Scanning new audio files (1)
Feb 12 14:50:45 DietPi squeezeboxserver[1025]: [23-02-12 14:50:45.0338] Slim::Music::Import::endImporter (712) Completed Slim::Media::MediaFolderScan Scan in 0.217 seconds.
Feb 12 14:50:45 DietPi squeezeboxserver[1025]: [23-02-12 14:50:45.0355] Slim::Music::Import::runImporter (579) Starting Slim::Music::PlaylistFolderScan scan
Feb 12 14:50:45 DietPi squeezeboxserver[1025]: [23-02-12 14:50:45.0369] Slim::Utils::Scanner::Local::rescan (179) Discovering audio files in /mnt/dietpi_userdata/Music
Feb 12 14:50:45 DietPi squeezeboxserver[1025]: [23-02-12 14:50:45.0697] Slim::Utils::Scanner::Local::__ANON__ (191) Start processing found tracks
Feb 12 14:50:45 DietPi squeezeboxserver[1025]: [23-02-12 14:50:45.0709] Slim::Utils::Scanner::Local::__ANON__ (199) Connect do DB
Feb 12 14:50:45 DietPi squeezeboxserver[1025]: [23-02-12 14:50:45.0720] Slim::Utils::Scanner::Local::__ANON__ (202) Get latest ID
Feb 12 14:50:45 DietPi squeezeboxserver[1025]: [23-02-12 14:50:45.0736] Slim::Utils::Scanner::Local::__ANON__ (225) Build temporary table for new tracks
Feb 12 14:50:45 DietPi squeezeboxserver[1025]: [23-02-12 14:50:45.0756] Slim::Utils::Scanner::Local::__ANON__ (245) Build temporary table for changed tracks
[... cut ...]
Feb 12 14:59:38 DietPi systemd[1]: logitechmediaserver.service: Succeeded.
Feb 12 14:59:38 DietPi systemd[1]: Stopped Logitech Media Server.
Feb 12 14:59:38 DietPi systemd[1]: logitechmediaserver.service: Consumed 13.103s CPU time.
Feb 12 15:01:36 DietPi systemd[1]: Started Logitech Media Server.
Feb 12 15:01:43 DietPi squeezeboxserver[3372]: [23-02-12 15:01:40.1515] main::init (377) Starting Logitech Media Server (v8.3.0, 1667251155, Fri 04 Nov 2022 09:16:39 AM CET) perl 5.032001 - aarch64-linux-gnu-thread-multi

yu13740000, 2023-02-12:
Here's Dietpi just installed, along with the Logitech Media Server

MichaIng, 2023-02-12:
> Can you give us some more details, like the steps to reproduce the issue (bug report template has its reason), in case any error messages you see
> ...
> and matching ones within `/var/log/squeezeboxserver`.

And when you try to play an m3u8, no additional log entry appears in `journalctl -u logitechmediaserver`? Because there is not a single log entry from after the install.

yu13740000, 2023-02-12:
root@DietPi:~# /var/log/squeezeboxserver
-bash: /var/log/squeezeboxserver: Is a directory
root@DietPi:~# journalctl -u logitechmediaserver
-- Journal begins at Sun 2023-02-12 23:17:01 GMT, ends at Sun 2023-02-12 23:54:11 GMT. --
Feb 12 23:50:03 DietPi systemd[1]: Started Logitech Media Server.
root@DietPi:~# sudo date 021307552023.20
Mon Feb 13 07:55:20 GMT 2023
root@DietPi:~#  journalctl -u logitechmediaserver
-- Journal begins at Sun 2023-02-12 23:17:01 GMT, ends at Mon 2023-02-13 07:56:24 GMT. --
Feb 12 23:50:03 DietPi systemd[1]: Started Logitech Media Server.
Feb 13 07:55:50 DietPi squeezeboxserver[1040]: [23-02-12 23:50:08.8945] main::init (377) Starting Logitech Media Server (v8
.3.0, 1667251155, Fri 04 Nov 2022 09:16:39 AM CET) perl 5.032001 - aarch64-linux-gnu-thread-multi
root@DietPi:~#  /var/log/squeezeboxserver
-bash: /var/log/squeezeboxserver: Is a directory
root@DietPi:~# journalctl -u logitechmediaserver
-- Journal begins at Sun 2023-02-12 23:17:01 GMT, ends at Mon 2023-02-13 07:57:05 GMT. --
Feb 12 23:50:03 DietPi systemd[1]: Started Logitech Media Server.
Feb 13 07:55:50 DietPi squeezeboxserver[1040]: [23-02-12 23:50:08.8945] main::init (377) Starting Logitech Media Server (v8
.3.0, 1667251155, Fri 04 Nov 2022 09:16:39 AM CET) perl 5.032001 - aarch64-linux-gnu-thread-multi
root@DietPi:~#  /var/log/squeezeboxserver
-bash: /var/log/squeezeboxserver: Is a directory
root@DietPi:~#

yu13740000, 2023-02-13:
/var/log/squeezeboxserver/server.log:
[23-02-12 23:50:08.8951] main::init (377) Starting Logitech Media Server (v8.3.0, 1667251155, Fri 04 Nov 2022 09:16:39 AM CET) perl 5.032001 - aarch64-linux-gnu-thread-multi
[23-02-13 07:55:59.1146] Slim::Networking::IO::Select::__ANON__ (130) Error: Select task failed calling Slim::Networking::Async::HTTP::_http_read_body: Can't call method "scanUrl" on an undefined value at /usr/share/perl5/Slim/Utils/Scanner/Remote.pm line 1294.
; fh=Slim::Networking::Async::Socket::HTTPS=GLOB(0xaaaad31ae418)
[23-02-13 07:57:27.0697] Slim::Formats::readTags (176) File missing: /ls-jd.qingting.fm/live/5022308/64k/20230213/07/5606-363-13280.aac
[23-02-13 07:57:27.0883] Slim::Formats::readTags (176) File missing: /ls-jd.qingting.fm/live/5022308/64k/20230213/07/5613-446-13281.aac
[23-02-13 07:57:27.1061] Slim::Formats::readTags (176) File missing: /ls-jd.qingting.fm/live/5022308/64k/20230213/07/5620-494-13282.aac
[23-02-13 07:57:27.1220] Slim::Formats::readTags (176) File missing: /ls-jd.qingting.fm/live/5022308/64k/20230213/07/5627-553-13283.aac
[23-02-13 07:57:27.1380] Slim::Formats::readTags (176) File missing: /ls-jd.qingting.fm/live/5022308/64k/20230213/07/5634-598-13284.aac
[23-02-13 07:57:27.1539] Slim::Formats::readTags (176) File missing: /ls-jd.qingting.fm/live/5022308/64k/20230213/07/5641-659-13285.aac
[23-02-13 07:57:27.1699] Slim::Formats::readTags (176) File missing: /ls-jd.qingting.fm/live/5022308/64k/20230213/07/5648-737-13286.aac
[23-02-13 07:57:27.1862] Slim::Formats::readTags (176) File missing: /ls-jd.qingting.fm/live/5022308/64k/20230213/07/5655-787-13287.aac
[23-02-13 07:57:27.2132] Slim::Networking::IO::Select::__ANON__ (130) Error: Select task failed calling Slim::Networking::Async::HTTP::_http_read_body: Can't locate object method "scanUrl" via package "Slim::Player::Protocols::File" at /usr/share/perl5/Slim/Utils/Scanner/Remote.pm line 1294.
; fh=Slim::Networking::Async::Socket::HTTP=GLOB(0xaaaad31a09a0)

/var/log/squeezeboxserver/perfmon.log:
null

/var/log/squeezeboxserver/scanner.log：
null

yu13740000, 2023-02-13:
https://cnlive.cnr.cn/hls/hjyyzs.m3u8
/var/log/squeezeboxserver/server.log:
[23-02-13 08:41:21.8335] Slim::Networking::IO::Select::__ANON__ (130) Error: Select task failed calling Slim::Networking::Async::HTTP::_http_read_body: Can't call method "scanUrl" on an undefined value at /usr/share/perl5/Slim/Utils/Scanner/Remote.pm line 1294.
; fh=Slim::Networking::Async::Socket::HTTPS=GLOB(0xaaaad325f720)

http://live.xmcdn.com/live/1205/64.m3u8
/var/log/squeezeboxserver/server.log:
[23-02-13 08:45:04.3965] Slim::Formats::readTags (176) File missing: /ls-jd.qingting.fm/live/5022308/64k/20230213/08/4346-431-158.aac
[23-02-13 08:45:04.4130] Slim::Formats::readTags (176) File missing: /ls-jd.qingting.fm/live/5022308/64k/20230213/08/4353-511-159.aac
[23-02-13 08:45:04.4310] Slim::Formats::readTags (176) File missing: /ls-jd.qingting.fm/live/5022308/64k/20230213/08/4400-563-160.aac
[23-02-13 08:45:04.4470] Slim::Formats::readTags (176) File missing: /ls-jd.qingting.fm/live/5022308/64k/20230213/08/4407-617-161.aac
[23-02-13 08:45:04.4626] Slim::Formats::readTags (176) File missing: /ls-jd.qingting.fm/live/5022308/64k/20230213/08/4414-748-162.aac
[23-02-13 08:45:04.4777] Slim::Formats::readTags (176) File missing: /ls-jd.qingting.fm/live/5022308/64k/20230213/08/4421-719-163.aac
[23-02-13 08:45:04.4924] Slim::Formats::readTags (176) File missing: /ls-jd.qingting.fm/live/5022308/64k/20230213/08/4428-799-164.aac
[23-02-13 08:45:04.5070] Slim::Formats::readTags (176) File missing: /ls-jd.qingting.fm/live/5022308/64k/20230213/08/4435-854-165.aac

 http://ls.qingting.fm/live/3412131.m3u8?bitrate=64
/var/log/squeezeboxserver/server.log:
[23-02-13 08:45:04.5156] Slim::Networking::IO::Select::__ANON__ (130) Error: Select task failed calling Slim::Networking::Async::HTTP::_http_read_body: Can't locate object method "scanUrl" via package "Slim::Player::Protocols::File" at /usr/share/perl5/Slim/Utils/Scanner/Remote.pm line 1294.
; fh=Slim::Networking::Async::Socket::HTTP=GLOB(0xaaaad324d198)

MichaIng, 2023-02-14:
You use the Radio > Tune In URL feature? This is what I tested it with, but not sure if there are other ways.

Your first URL returns a 403 (forbidden) here, so I cannot test.

The second for a some seconds, then it suddenly stops. I think I know what the problem is with it: This URL always returns 5 audio files with just a few seconds length each. When those have been played, the URL needs to be re-fetched to get the new list of files. Basically it needs to continuously be re-fetched, which Logitech Media Server doesn't seem to do. And of course you cannot download this playlist since then you likely have a list of URLs which do not exist anymore, which is most likely the reason for your "File missing" errors.

The 3rd is currently silent, but that may be as its currently night in East Asia. But otherwise it "plays" for > 8 minutes now. It however also a list of 8 audio files, and I guess is generally underlying the same issue than the second.

I'm actually not familiar with how radio streams are usually working, probably Logitech Media Server has a way to deal with this kind of dynamic playlists. But this is better asked in an audio community or Logitech Media Server forum.

yu13740000, 2023-02-15:
Tunein needs a VPN proxy in China, which is inconvenient,And none of these three.

After testing, the three URLs, MPD, Daphile and moOde, play normally. x86 Daphile is the LMS.

I will test later whether the LMS on armbian is normal.

MichaIng, 2023-02-15:
> I will test later whether the LMS on armbian is normal.

Good idea. Because I think this is a limitation of LMS, or we missed how to play such playlists with it. I forgot to mention that I found the "Dynamic Playlists" plugin in LMS, but failed to setup anything with it. The LMS forum (first) or GitHub repo (second) would be a good place to ask about this, if it fails the same way on Armbian.

yu13740000, 2023-02-16:
Armbian also cannot play these three

michaelherger, 2023-02-16:
Please try again with the PlayHLS plugin installed.

yu13740000, 2023-02-16:
> Please try again with the PlayHLS plugin installed.

Thank you, this is indeed the problem, I found a solution in 2015.

2015-02-16, 01:32
Linux Installation instructions

PlayHLS V2.* Instructions.

1. On the LMS WebGui; go to 'Settings', and select the 'Plugins' tab. Go to the bottom.
You may need to tick a box saying something like 'allow 3rd party plugins', and press Apply.
2. At the very bottom you should have a section called 'Additional Repositories'
Add the line
http://bpa-code.github.io/bpaplugins/repo-playhls-v2.xml
and Press Apply.
3. LMS will need to be restarted either as part of the above steps or manually.

----------------------------------------------------------------------------------------------------------------

PlayHLS V1.* installation instructions. Suggestion: only use V1.* if V2.* cannot be made to work.

1. Install ffmpeg. It is in a package called ffmpeg so you may use something like "apt-get install ffmpeg"
2. On the LMS WebGui; go to 'Settings', and select the 'Plugins' tab. Go to the bottom.
You may need to tick a box saying something like 'allow 3rd party plugins', and press Apply.
3. Find the line with PlayHLS and check the box and Press Apply.
4. Restart LMS.
5. After LMS has restarted, on the LMS WebGui; go to 'Settings', and select the 'Plugins' tab. Check the PlyHLS plugis is enabled - if not check the enable box and click Apply.
6. Check that Plugin has been installed correctly by looking LMS WebGui; go to 'Settings", and then select "Advanced".
Select "File Type" from the drop down box on Top Right. Scroll down until you see "Apple HLS m3u8" there should be a entries with "ffmpeg". They should not be greyed out or disabled.
7. LMS should now be able to play the new format streams. BBC URL to test will depend whether you are UK or not in UK.

This is a PlayHLS V1.* issue only.
There is a significant issue about ffmpeg version. If the version is too old then either stream won't play or ffmpeg process won't die after playing a stream so that there is a build up of ffmpeg processes. These problems seem to happen with ffmpeg with versions before 1.0. Ideally install a recent build of ffmpeg - this may require building your own.

Additional notes to use avconv instead of ffmpeg.
Last edited by [bpa](https://forums.slimdevices.com/member/1806-bpa); 2020-04-07, 10:51. Reason: Add V2.* instructions andupdate repo URL for V1.*

closed 2023-02-16 by yu13740000

nehasjains, 2023-08-11:
I m using HLS plugin v2.9. If I pause the m3u8 stream and resume it, the stream starts from the beginning. How do I fix this ?

michaelherger, 2023-08-11:
You should try to get in touch with the PlayHLS author on forums.slimdevices.com.

nehasjains, 2023-08-11:
Thank you... will post it there...