MichaIng/DietPi #6155: Portainer / Docker not accessible after update to 8.14.2

bbird8, 2023-02-12:
After update from 8.13.2 to 8.14.2 Docker / Portainer is not accessible anymore via http ("connection refused").
Update run thru without any error and services are up after update.
Apache itself is running fine and accessible. 

I reinstalled docker composer (134) as recommended in the change log. No change, still no access.

As I never had any issues with update to far, not sure where to investigate further. Thanks for any advice.

Joulinar, 2023-02-12:
This has nothing to do with Docker Compose. Probably you are hit by this Docker bug https://github.com/MichaIng/DietPi/issues/6126

bbird8, 2023-02-12:
Thanks. However, I tried

`apt install apparmor
systemctl stop apparmor`

without success. 
But yes, I'm on a Proxmox VM x86 here which seems to be hit by this issue.

Joulinar, 2023-02-12:
Can you share logs from Docker service?

bbird8, 2023-02-12:
I checked for portainer container log:

```
root@pKuma230:/var/lib/docker/containers# docker logs --since=10m 2071d9569812
2023/02/12 12:24:04 server: Reverse tunnelling enabled
2023/02/12 12:24:04 server: Fingerprint a9:80:5d:c9:3d:f9:77:93:6c:c6:f4:a8:a6:48:79:87
2023/02/12 12:24:04 server: Listening on 0.0.0.0:8000...
level=info msg="2023/02/12 12:24:04 [DEBUG] [chisel, monitoring] [check_interval_seconds: 10.000000] [message: starting tunnel management process]"
level=info msg="2023/02/12 12:24:04 [INFO] [cmd,main] Starting Portainer version 2.9.3"
level=info msg="2023/02/12 12:24:04 [DEBUG] [internal,init] [message: start initialization monitor ]"
level=info msg="2023/02/12 12:24:04 [INFO] [http,server] [message: starting HTTPS server on port :9443]"
level=info msg="2023/02/12 12:24:04 [INFO] [http,server] [message: starting HTTP server on port :9000]"
```

Not sure about the service as

```
root@pKuma230:~# docker service ls
Error response from daemon: This node is not a swarm manager. Use "docker swarm init" or "docker swarm join" to connect this node to swarm and try again.
```

Joulinar, 2023-02-12:
Check following

`journalctl -u docker.service`

bbird8, 2023-02-12:
Ok, here the related log after the update:
It seems to be linked to the apparmor issue.

```
Feb 12 13:44:10 pKuma230 systemd[1]: Stopping Docker Application Container Engine...
Feb 12 13:44:10 pKuma230 9082cf361f49[751]: s6-rc: info: service 99-ci-service-check: stopping
Feb 12 13:44:10 pKuma230 9082cf361f49[751]: s6-rc: info: service 99-ci-service-check successfully stopped
Feb 12 13:44:10 pKuma230 9082cf361f49[751]: s6-rc: info: service legacy-services: stopping
Feb 12 13:44:10 pKuma230 1065e3df9cc0[751]: 2023-02-12T13:44:10+01:00 [SERVER] INFO: Shutdown requested
Feb 12 13:44:10 pKuma230 1065e3df9cc0[751]: 2023-02-12T13:44:10+01:00 [SERVER] INFO: Called signal: SIGTERM
Feb 12 13:44:10 pKuma230 1065e3df9cc0[751]: 2023-02-12T13:44:10+01:00 [SERVER] INFO: Stopping all monitors
Feb 12 13:44:10 pKuma230 9082cf361f49[751]: s6-rc: info: service legacy-services successfully stopped
Feb 12 13:44:10 pKuma230 9082cf361f49[751]: s6-rc: info: service init-services: stopping
Feb 12 13:44:10 pKuma230 9082cf361f49[751]: s6-rc: info: service 00-legacy: stopping
Feb 12 13:44:10 pKuma230 9082cf361f49[751]: s6-rc: info: service 00-legacy successfully stopped
Feb 12 13:44:10 pKuma230 9082cf361f49[751]: s6-rc: info: service init-services successfully stopped
Feb 12 13:44:10 pKuma230 9082cf361f49[751]: s6-rc: info: service init-mods-end: stopping
Feb 12 13:44:10 pKuma230 9082cf361f49[751]: s6-rc: info: service init-mods-end successfully stopped
Feb 12 13:44:10 pKuma230 9082cf361f49[751]: s6-rc: info: service init-mods-package-install: stopping
Feb 12 13:44:10 pKuma230 9082cf361f49[751]: s6-rc: info: service init-mods-package-install successfully stopped
Feb 12 13:44:10 pKuma230 9082cf361f49[751]: s6-rc: info: service init-mods: stopping
Feb 12 13:44:10 pKuma230 9082cf361f49[751]: s6-rc: info: service init-mods successfully stopped
Feb 12 13:44:10 pKuma230 9082cf361f49[751]: s6-rc: info: service legacy-cont-init: stopping
Feb 12 13:44:10 pKuma230 9082cf361f49[751]: s6-rc: info: service legacy-cont-init successfully stopped
Feb 12 13:44:10 pKuma230 9082cf361f49[751]: s6-rc: info: service fix-attrs: stopping
Feb 12 13:44:10 pKuma230 9082cf361f49[751]: s6-rc: info: service fix-attrs successfully stopped
Feb 12 13:44:10 pKuma230 9082cf361f49[751]: s6-rc: info: service s6rc-oneshot-runner: stopping
Feb 12 13:44:10 pKuma230 9082cf361f49[751]: s6-rc: info: service s6rc-oneshot-runner successfully stopped
Feb 12 13:44:12 pKuma230 1065e3df9cc0[751]: 2023-02-12T13:44:12+01:00 [DB] INFO: Closing the database
Feb 12 13:44:14 pKuma230 1065e3df9cc0[751]: 2023-02-12T13:44:14+01:00 [DB] INFO: SQLite closed
Feb 12 13:44:14 pKuma230 1065e3df9cc0[751]: 2023-02-12T13:44:14+01:00 [CLOUDFLARED] INFO: Stop cloudflared
Feb 12 13:44:14 pKuma230 1065e3df9cc0[751]: 2023-02-12T13:44:14+01:00 [SERVER] INFO: Graceful shutdown successful!
Feb 12 13:44:14 pKuma230 systemd[1]: docker.service: Succeeded.
Feb 12 13:44:14 pKuma230 systemd[1]: Stopped Docker Application Container Engine.
Feb 12 13:46:39 pKuma230 systemd[1]: Started Docker Application Container Engine.
Feb 12 13:46:39 pKuma230 dockerd[6896]: time="2023-02-12T13:46:39.660011670+01:00" level=warning msg="AppArmor enabled on system but \"apparmor_parser\" binary is missing, so profile can't b
e loaded"
Feb 12 13:46:42 pKuma230 dockerd[6896]: time="2023-02-12T13:46:42.395521336+01:00" level=error msg="stream copy error: reading from a closed fifo"
Feb 12 13:46:42 pKuma230 dockerd[6896]: time="2023-02-12T13:46:42.395752993+01:00" level=error msg="stream copy error: reading from a closed fifo"
Feb 12 13:46:42 pKuma230 dockerd[6896]: time="2023-02-12T13:46:42.607874185+01:00" level=error msg="stream copy error: reading from a closed fifo"
Feb 12 13:46:42 pKuma230 dockerd[6896]: time="2023-02-12T13:46:42.608681569+01:00" level=error msg="stream copy error: reading from a closed fifo"
Feb 12 13:46:42 pKuma230 dockerd[6896]: time="2023-02-12T13:46:42.724018666+01:00" level=error msg="1065e3df9cc0371f801a7bafeef4f818c40f3405b3bf2b77eadff3ab30b4abb8 cleanup: failed to delete
 container from containerd: no such container"
Feb 12 13:46:42 pKuma230 dockerd[6896]: time="2023-02-12T13:46:42.724085769+01:00" level=error msg="failed to start container" container=1065e3df9cc0371f801a7bafeef4f818c40f3405b3bf2b77eadff
3ab30b4abb8 error="failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: unable to apply apparmor profile
: apparmor failed to apply profile: write /proc/self/attr/exec: no such file or directory: unknown"
Feb 12 13:46:42 pKuma230 dockerd[6896]: time="2023-02-12T13:46:42.729050059+01:00" level=error msg="stream copy error: reading from a closed fifo"
Feb 12 13:46:42 pKuma230 dockerd[6896]: time="2023-02-12T13:46:42.730323901+01:00" level=error msg="stream copy error: reading from a closed fifo"
Feb 12 13:46:42 pKuma230 dockerd[6896]: time="2023-02-12T13:46:42.984305445+01:00" level=error msg="2071d9569812bd37dfabd2ced806392e85239242e2aa62d2b5c2ac7690beb027 cleanup: failed to delete
 container from containerd: no such container"
Feb 12 13:46:42 pKuma230 dockerd[6896]: time="2023-02-12T13:46:42.984769995+01:00" level=error msg="failed to start container" container=2071d9569812bd37dfabd2ced806392e85239242e2aa62d2b5c2a
c7690beb027 error="failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: unable to apply apparmor profile
[... cut ...]
Feb 12 13:46:43 pKuma230 dockerd[6896]: time="2023-02-12T13:46:43.049078169+01:00" level=error msg="failed to start container" container=9082cf361f49fc3f2cc93a8fbce572e3c49c1699b1e93446de73f
7a823511925 error="failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: unable to apply apparmor profile
: apparmor failed to apply profile: write /proc/self/attr/exec: no such file or directory: unknown"
Feb 12 13:46:43 pKuma230 dockerd[6896]: time="2023-02-12T13:46:43.118437994+01:00" level=warning msg="WARNING: No swap limit support"
```

bbird8, 2023-02-12:
Ahhh, not it seems to work after:

```
apt install apparmor
systemctl disable --now apparmor
systemctl restart docker
```

Joulinar, 2023-02-12:
I'm going to close the issue for now, feel free to reopen if needed.

closed 2023-02-12 by Joulinar