MichaIng/DietPi #6156: Error in dietpi-update (Orange PI 4 LTS)

metrolan, 2023-02-12:
#### Details:
- Date           | Sun Feb 12 13:23:58 CET 2023
- DietPi version | v8.13.2 (MichaIng/master)
- Image creator  | Raspbian
- Pre-image      | Raspbian
- Hardware       | Generic Rockchip RK3399 (aarch64) (ID=24)
- Kernel version | `Linux OrangePi 5.10.43 #3.0.6 SMP Mon Aug 1 14:28:13 CST 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `getent hosts dns9.quad9.net`
- Exit code      | 2
- Software title | DietPi-Update
#### Steps to reproduce:

1. executing dietpi-update

#### Expected behaviour:

Normal version update

#### Actual behaviour:

get this error

#### Extra details:

- Orange Pi 4 LTS

```

### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SD card used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?
<!-- If you sent a "dietpi-bugreport", please paste the ID here -->
- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce
<!-- Explain how to reproduce the issue -->
1. ...
3. ...

#### Expected behaviour
<!-- What SHOULD be happening? -->
- ...

#### Actual behaviour
<!-- What IS happening? -->
- ...

#### Extra details
<!-- Please post any extra details that might help solve the issue -->
- ...

Joulinar, 2023-02-12:
That looks like a DNS issue. Can you check DNS resolution is working fine.

`getent hosts dns9.quad9.net`

metrolan, 2023-02-12:
Yes, you are right.

I was usin dns0. Tried quad9 and it works fine

Thankyou!

Joulinar, 2023-02-12:
Ok I'm going to close the issue for now, feel free to reopen if needed.

closed 2023-02-12 by Joulinar