MichaIng/DietPi #6157: [Feature request] Add option to periodically 'reinstall' (=update) dietpi-software packages from custom repo's.

GvY85, 2023-02-12:
Hi,

I just found out that I never updated Rclone in about 4 years because I forgot it is from a custo repo and I thus have to run `dietpi-software reinstall 202` once in a while.

Perhaps it would be nice to add an option to Dietpi-Software to reinstall all software with custom repo's once in a while (selectable time)?
Also, a notification could be added if it has been a while since any of these packages have been updated?

For things like Pi-hole I guess people watch the WebUI once in a while but other packages might not get so lucky and ts easy to forget you are using something from a custom repo via `dietpi-software` that is not updateable via `apt`.

MichaIng, 2023-02-12:
> in about 4 years

Rclone was added to `dietpi-software` less than a year ago, so this cannot be true 😄: #5560

However, yes the general issue is clear, an update check would be nice, but implementing this for all non-APT installs is a large task. For now, a little self-responsibility is required.

Joulinar, 2023-02-12:
Even for some software title we are not able to fetch current version. This would be quite time consuming to have such functionality implemented. We could print a notification at the end of `dietpi-update` and remind people to take care on non-apt software packages.

MichaIng, 2023-02-12:
We could add related links to check for new versions to our docs, to the update tab or so. They are listed here: https://github.com/MichaIng/DietPi/wiki/DietPi-Software-list

Bonus is that one can often find release notes there as well.

Joulinar, 2023-02-12:
still users would need to check manually on new version. Probably majority is even not aware on our docs at all 🤣

GvY85, 2023-02-13:
Perhaps it can be as simple as giving the user, at the end of `dietpi-update`, a screen where he can choose from the installed packages that do not come from official repo's/.deb's and then automatically `dietpi software reinstall` the packages selected?

Looking at the documentation (yes I did ;)) it seems you have a good idea what packages would be suited for this correct?

GvY85, 2023-02-13:
Even better :)!