MichaIng/DietPi #6159: Failure:  Restoring a Backup

bccarbone, 2023-02-12:
**Hardware Info**

DietPi version | v8.13.2 (MichaIng/master)
Image creator | DietPi Core Team
Pre-image | from scratch
Hardware | RPi 4 Model B (aarch64) (ID=4)
Kernel version | Linux DietPi 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux
Distro | bullseye (ID=6,RASPBIAN=0)

**Setup**

Based on the advice of this thread here, this is how I setup my backup routine:  https://dietpi.com/forum/t/dietpi-backup-formatting-usb-flash-drive-backup/15406

**Restoring Backup**

- I used dietpi-backup and chose the 'Restore' option.
- After doing so, it failed and I've attached the log here:  [dietpi-backup.failure.txt](https://github.com/MichaIng/DietPi/files/10716743/dietpi-backup.failure.txt)

**Latest Backup Log Files**

From what I can see, the backups seem to be running ok since I started them.

[dietpi-backup_stats.txt](https://github.com/MichaIng/DietPi/files/10716747/dietpi-backup_stats.txt)
[dietpi-backup.txt](https://github.com/MichaIng/DietPi/files/10716751/dietpi-backup.txt)

Let me know what else I can provide.  Thank you!

MichaIng, 2023-02-12:
Thanks for reporting.

> some files/attrs were not transferred (see previous errors)

Uff, how to find an error between ~48k lines of log 😄. I cannot find it, also not sure about a good pattern, "failure", "failed", "error" all do work. Probably an immutable flag or something.

It shouldn't be a problem as everything else was transferred, but still good to search through those lines and find the failed one. Also good to then check and remember the way rsync logs this inline so it can be searched for with `grep` and similar tools and a regular expression.

bccarbone, 2023-02-12:
I had the same thought when I saw this log too.... :-).  Thanks for the feedback.

I figured it was worth noting to you guys.  My setup is pretty stock and standard thanks to all of your great automated tools.  I don't deviate beyond that.

MichaIng, 2023-04-03:
Does this still show up? Probably when we remove the verbose flag from `rsync` and see whether then only the error is shown:
```sh
sed -i 's/-v --log-file/--log-file/' /boot/dietpi/dietpi-backup
```
then repeat the backup.

bccarbone, 2023-04-03:
Just so I understand:

1) You want me to run: sed -i 's/-v --log-file/--log-file/' /boot/dietpi/dietpi-backup
2) Repeat the backup

And then what would you like...?

MichaIng, 2023-04-03:
If the issue with the failure buried somewhere among the 48k lines is still present, yes. The first command removes the `-v` (verbose) flag which should mute the output for each synced file so that hopefully only the error and summary are left. Then we should see what causes the issue.

bccarbone, 2023-04-03:
Ok, will do and will let you know.  Appreciate you keeping this on your radar.

bccarbone, 2023-04-05:
Took a look at the log and I'm not seeing any errors with the backup.  As I recall, this occurred when I did a restore of the backup.

MichaIng, 2023-04-05:
The command removed the verbose flag as well for restoring. So you could restore a fresh backup then to check for the error.

bccarbone, 2023-04-06:
Restored a fresh backup and received no errors.  Hmm...odd.

MichaIng, 2023-04-06:
Hmm, probably the new backup somehow solved it. Maybe some filesystem corruption which was solved on reboot or so. I'll close the issue for now. If the same happens again, feel free to reopen and try the above command again to hopefully extract the error message.

closed 2023-04-06 by MichaIng

bccarbone, 2023-04-06:
Will do, thank you!