MichaIng/DietPi #6160: SSL error with dietpi-update

mews-se, 2023-02-12:
#### Details:
- Date           | Sun Feb 12 21:00:04 CET 2023
- Bug report     | b49c7931-76de-4a09-8605-18a6955fb855
- DietPi version | v8.13.2 (MichaIng/master)
- Image creator  | mews
- Pre-image      | d11-alexhost
- Hardware       | Virtual Machine (x86_64) (ID=20)
- Kernel version | `Linux moldova2.pissnisse.se 5.10.0-21-amd64 #1 SMP Debian 5.10.162-1 (2023-01-21) x86_64 GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `curl -sSfLO https://github.com/MichaIng/DietPi/archive/master.tar.gz`
- Exit code      | 35
- Software title | DietPi-Update
- Bug report ID | b49c7931-76de-4a09-8605-18a6955fb855

#### Additional logs:
```
curl: (35) OpenSSL SSL_connect: Connection reset by peer in connection to github.com:443
```

Basically I'm testing out DietPi compability on a lot of VPS providers. In just this instance (I have two from the same one) it fails due to an ssl error. Any tips?

Joulinar, 2023-02-12:
maybe a temp. issue connecting to GitHub. Can you try following manually? 

```
curl -sSfLO https://github.com/MichaIng/DietPi/archive/master.tar.gz
```

mews-se, 2023-02-12:
Yep. Same issue. The other VPS worked flawlessly. I'm trying to add SSH-keys to try to please github :)

Joulinar, 2023-02-12:
I don't think any SSH-keys needed. This should be plain HTTPS.

MichaIng, 2023-02-12:
Many thanks for reporting.

Probably a dumb firewall or proxy of the VPS provide or its ISP, some countries seem to lock down GitHub connections:
- https://stackoverflow.com/questions/71571965/openssl-ssl-connect-connection-was-reset-in-connection-to-github-com443-while
- https://github.com/composer/composer/issues/10905
- https://github.com/desktop/desktop/issues/11287

SSH keys to connect via Git to GitHub are a workaround for this problem when you are trying to clone a repo. But here we are explicitly accessing via HTTPS and not Git.

mews-se, 2023-02-12:
Yup most likely. I even tried `curl -v github.com:443`
and got
```
*   Trying 140.82.121.3:443...
* Connected to github.com (140.82.121.3) port 443 (#0)
> GET / HTTP/1.1
> Host: github.com:443
> User-Agent: curl/7.74.0
> Accept: */*
>
* Empty reply from server
* Connection #0 to host github.com left intact
curl: (52) Empty reply from server
```

MichaIng, 2023-02-12:
You just hit the HTTPS redirect. Try:
```sh
curl -I https://github.com/
```

mews-se, 2023-02-12:
Same unfortunatly. Wierd since both VPS are on the same network.

Well, perhaps it's just a temporary thing until something resets...

Joulinar, 2023-02-12:
or check if the certificate is correctly shown from your VPS

```
openssl s_client -connect github.com:443 -servername github.com
```

mews-se, 2023-02-12:
Well yes, now we are on to something.

```
CONNECTED(00000003)
write:errno=104
---
no peer certificate available
---
No client certificate CA names sent
---
SSL handshake has read 0 bytes and written 302 bytes
Verification: OK
---
New, (NONE), Cipher is (NONE)
Secure Renegotiation IS NOT supported
Compression: NONE
Expansion: NONE
No ALPN negotiated
Early data was not sent
Verify return code: 0 (ok)
---

```

Joulinar, 2023-02-12:
you could try to reinstall `ca-certificates` package.

``
apt install --reinstall ca-certificates
```

mews-se, 2023-02-12:
I'm adding this to the "intermittent"-status and adding them to my own not-recommended list :) I'll mark this as closed due to provider error.

Thanks guys!

closed 2023-02-12 by mews-se

mews-se, 2023-02-13:
Just a small note. Adding a -l to the curl command worked....

MichaIng, 2023-02-14:
The `https://` was what solved the empty reply (given that the request otherwise goes through), the `-I` option is to do a HEAD request instead of a GET request, so the HTML content is omitted but only the response headers and status are shown 🙂.

piyushaswani55, 2023-03-11:
@MichaIng @mews-se how did you fix it, I am facing the same issue.

mews-se, 2023-03-11:
> @MichaIng @mews-se how did you fix it, I am facing the same issue.

I didn’t. The IP was placed on a few blocklists (TOR exit) and when removed all was fine!

piyushaswani55, 2023-03-11:
> > @MichaIng @mews-se how did you fix it, I am facing the same issue.
> 
> I didn’t. The IP was placed on a few blocklists (TOR exit) and when removed all was fine!

For past couple of months I am unable to update dietpi. Earlier it was working and I see sometimes I get response for `curl https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/version` if done manually but `dietpi-update` always fails for me. 
[image]

Joulinar, 2023-03-11:
If you have the possibility, connect your device to a different internet provider or try using a mobile phone hotspot. Just for testing.

piyushaswani55, 2023-03-12:
I was wonder how `dietpi-banner` knew what update version was.
[image]

Anyhoo after a lot of research and hit & trial, I found out that my ISP has blocked github raw content and it was inconsistent, 1 out of 10 times randomly it worked.
So I modified the `dietpi-update` script and replaced all `curl` with `curl --retry 3 --retry-all-errors`, after which I was able to update.
Still I think there should be a better way to update in such cases, just a suggestion, if it helps a major section of audience.

MichaIng, 2023-03-12:
> I found out that my ISP has blocked github raw content

I would be highly interested in the reasons for this, if you have a chance to ask your ISP? This is highly unreasonable and harmful, unless it was the intention of the ISP/state to block access to one of the largest source of valuable free data for their customers/citizens, which of course would be the worst scenario. But then, not blocking `github.com` at the same time, doesn't fit.

piyushaswani55, 2023-03-12:
I found these below references and after going through them it seems multiple ISPs in India have done this.
- https://www.gadgets360.com/internet/news/github-raw-user-content-domain-url-blocked-act-fibernet-jio-fiber-india-3657407
- https://reddit.com/r/india/comments/zd7ykr/jio_has_blocked_rawgithubusercontentcom_again/
- https://github.com/orgs/community/discussions/42655

One speculation is that due to `raw` in the URL, they consider it as porn 😂 
Whatever the reason, I totally agree that this is dumb of them to block github raw content while github is working fine, it has apparently affected a lot of services which were relying on github raw content. 

The are mostly 3 ways which work
- Using a VPN works like a charm all the time (I am sure that it will work and is the safest as well as simplest way of all but I couldn't do this for personal reasons).
- Using DNS over HTTPS, and although my `nameserver` is already set to use Quad9 IPs, not sure why it didn't work for me. (NOTE: I have DNS over HTTPS configured in my browser and never faced any issue opening github raw links in my browser).
- Adding an entry for Github IP in `/etc/hosts`, but that too didn't work for me not sure why.
- Since I noticed it was inconsistent, adding retries to `curl` worked for me. (NOTE: I had to click on retry prompt 3-4 times over and over).

In the end it's going to be a pain for me to update dietpi every time now because of my dumb ISP 🤦🏼

MichaIng, 2023-03-12:
> In the end it's going to be a pain for me to update dietpi every time now because of my dumb ISP 🤦🏼

Well, why not just asking the ISP to whitelist the domain? It was successful in another case: https://github.com/orgs/community/discussions/42655#discussioncomment-4648395
> Will create a ticket with ACT tomorrow and see what they say.
> ...
> The domain is now whitelisted and accessible.

after just 3 days.

piyushaswani55, 2023-03-12:
Thanks @MichaIng , I really appreciate you for `dietpi` and helping me and many others. ❤️ 

Actually my ISP is different, I will definitely talk to them but I have very little hopes from them. 
Even if they whitelist it now, there is no telling if they will block again in future.

Recently they blocked VLC website because of some cases where the installer had malware which people downloaded from random website, making ISPs block official source for months 😂 
- https://techcrunch.com/2022/08/17/vlc-india/
- https://arstechnica.com/tech-policy/2022/10/videolan-gets-no-answers-on-why-website-is-blocked-in-india-threatens-to-sue-govt/

MichaIng, 2023-03-12:
> Even if they whitelist it now, there is no telling if they will block again in future.

You will never know without trying it. Nothing will change it everything is believing that nothing can change 😉.

piyushaswani55, 2023-03-12:
Yes I will definitely talk to them. 
I have to talk to them anyway to open ports to router and get public IP to expose my service. Currently I am stuck with `cloudflared` tunnels and can't seem to find any other better option.

Also, I will also continue to find alternatives and update the thread if I come across any good solution.