MichaIng/DietPi #6164: Cannot connect to ssh.dietpi.com to purge survey / submit bug report

kullachartp, 2023-02-15:
#### Details:
- Date           | Wed Feb 15 07:24:05 +07 2023
- DietPi version | v8.14.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux RPi4B2 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `curl --key  --connect-timeout 8 --retry 1 --retry-delay 4 -sSvT fa04ea05-ff17-4732-8c32-7fb2c569fef7.7z sftp://dietpi-survey:upload2dietpi@ssh.dietpi.com:29248/bugreport/`
- Exit code      | 28
- Software title | DietPi-BugReport
#### Steps to reproduce:

# /boot/dietpi/dietpi-survey 1
[FAILED] DietPi-Survey | Purging survey data
 - Command: curl --key  -m 20 -sT fa04ea05-ff17-4732-8c32-7fb2c569fef7.txt sftp://dietpi-survey:upload2dietpi@ssh.dietpi.com:29248/survey/
 
so I try to submit a bug report but the following issue occurs:
#### Additional logs:
```
*   Trying 185.101.93.93:29248...
*   Trying 2a06:1c40:3::142:29248...
* Immediate connect fail for 2a06:1c40:3::142: Network is unreachable
* After 2441ms connect time, move on!
* connect to 185.101.93.93 port 29248 failed: Connection timed out
* Failed to connect to ssh.dietpi.com port 29248: Connection timed out
* Closing connection 0
curl: (28) Failed to connect to ssh.dietpi.com port 29248: Connection timed out
* Hostname ssh.dietpi.com was found in DNS cache
*   Trying 185.101.93.93:29248...
*   Trying 2a06:1c40:3::142:29248...
* Immediate connect fail for 2a06:1c40:3::142: Network is unreachable
* After 4000ms connect time, move on!
* connect to 185.101.93.93 port 29248 failed: Connection timed out
* Failed to connect to ssh.dietpi.com port 29248: Connection timed out
* Closing connection 1
curl: (28) Failed to connect to ssh.dietpi.com port 29248: Connection timed out

Joulinar, 2023-02-15:
> * Immediate connect fail for 2a06:1c40:3::142: Network is unreachable

Looks like IPv6 is not working or configured correctly. Can you try to disable IPv6 and check following?

```
> "$G_HW_UUID.txt"
curl --key '' -m 20 -T "$G_HW_UUID.txt" 'sftp://dietpi-survey:upload2dietpi@ssh.dietpi.com:29248/survey/'
```

kullachartp, 2023-02-15:
> Looks like IPv6 is not working or configured correctly. Can you try to disable IPv6 and check following?
>
> ```
> curl --key '' -m 20 -T "$G_HW_UUID.txt" [...]

I am not sure I do it correctly. Please advise.

root@pi# "$G_HW_UUID.txt"
-bash: fa04ea05-ff17-4732-8c32-7fb2c569fef7.txt: command not found
root@pi# curl --key '' -m 20 -T "$G_HW_UUID.txt" 'sftp://dietpi-survey:upload2dietpi@ssh.dietpi.com:29248/survey/'
curl: Can't open 'fa04ea05-ff17-4732-8c32-7fb2c569fef7.txt'!
curl: try 'curl --help' or 'curl --manual' for more information
curl: (26) Failed to open/read local data from file/application
root@RPi4B2:/#

Joulinar, 2023-02-15:
You need to copy the entire command. You missed a `>` at the beginning.

kullachartp, 2023-02-15:
> curl --key '' -m 20 -T "$G_HW_UUID.txt" 'sftp://dietpi-survey:upload2dietpi@ssh.dietpi.com:29248/survey/'

root@pi:~#> "$G_HW_UUID.txt"
root@pi:~# curl --key '' -m 20 -T "$G_HW_UUID.txt" 'sftp://dietpi-survey:upload2dietpi@ssh.dietpi.com:29248/survey/'
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:15 --:--:--     0curl: (6) Could not resolve host: ssh.dietpi.com

Joulinar, 2023-02-15:
Now it looks like you have DNS issue.

> Could not resolve host: ssh.dietpi.com

What DNS server you are using?

kullachartp, 2023-02-15:
> What DNS server you are using?

I use Google DNS 8.8.8.8. Anyway, from the first log, I saw that it can translate dietpi.com to 185.101.93.93.
I think it's a network issue like your thought, because my DietPi in another network works properly. Maybe I try changing DNS and see the issue be solved.

kullachartp, 2023-02-15:
Changing to Cloudflare DNS but the problem still persists:
'''
cat /etc/resolv.conf
nameserver 1.1.1.1
nameserver 1.0.0.1
'''

'''
/boot/dietpi/dietpi-survey 1
[FAILED] DietPi-Survey | Purging survey data
 - Command: curl --key  -m 20 -sT fa04ea05-ff17-4732-8c32-7fb2c569fef7.txt sftp://dietpi-survey:upload2dietpi@ssh.dietpi.com:29248/survey/
'''

Joulinar, 2023-02-15:
Can you share the test command I shared above.

kullachartp, 2023-02-15:
> Can you share the test command I shared above.

Which one?

Joulinar, 2023-02-15:
```
> "$G_HW_UUID.txt"
curl --key '' -m 20 -T "$G_HW_UUID.txt" 'sftp://dietpi-survey:upload2dietpi@ssh.dietpi.com:29248/survey/'
```

kullachartp, 2023-02-15:
> curl --key '' -m 20 -T "$G_HW_UUID.txt" 'sftp://dietpi-survey:upload2dietpi@ssh.dietpi.com:29248/survey/'
`
curl --key '' -m 20 -T "$G_HW_UUID.txt" 'sftp://dietpi-survey:upload2dietpi@ssh.dietpi.com:29248/survey/'
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:10 --:--:--     0
  0     0    0     0    0     0      0      0 --:--:--  0:00:10 --:--:--     0
curl: (28) Failed to connect to ssh.dietpi.com port 29248: Connection timed out
`

MichaIng, 2023-02-15:
Probably your network or ISP restricts connections. DNS via Cloudflare works now but the connection to your SSH server still times out. Could you try it on port 22?
```sh
curl --key '' -m 20 -T "$G_HW_UUID.txt" 'sftp://dietpi-survey:upload2dietpi@ssh.dietpi.com:22/survey/'
```

kullachartp, 2023-02-15:
> ```shell
> curl --key '' -m 20 -T "$G_HW_UUID.txt" 'sftp://dietpi-survey:upload2dietpi@ssh.dietpi.com:22/survey/'
> ```

```
curl --key '' -m 20 -T "$G_HW_UUID.txt" 'sftp://dietpi-survey:upload2dietpi@ssh.dietpi.com:22/survey/'
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:10 --:--:--     0
  0     0    0     0    0     0      0      0 --:--:--  0:00:10 --:--:--     0
curl: (28) Failed to connect to ssh.dietpi.com port 22: Connection timed out
```
From your hint I also think it's probably about my ISP restrictions. I tried ping 185.101.93.93 (ssh.dietpi.com) but received nothing. I cannot use traceroute in DietPi to trace where packets were dropped. Please advise if there is any other point that should be tested.

Besides, as I am new to GitHub, I am sorry for this disorganized thread and thank you to arrange it.

MichaIng, 2023-02-15:
> Besides, as I am new to GitHub, I am sorry for this disorganized thread and thank you to arrange it.

All good, my edits (if this is what you mean) were mostly about removing the links, so bots are not too easily made aware of this address, which is, other than dietpi.com, not protected/anonymised by Cloudflare. This is also the reason we switched to the custom port to mute the random login attempts (once the number of old DietPi systems using the old port are sufficiently low) that happen on our sever every second 😉.

> I cannot use traceroute in DietPi

You mean because it is not installed?
```sh
apt install traceroute
```

kullachartp, 2023-02-16:
```
~# traceroute ssh.dietpi.com
traceroute to ssh.dietpi.com (185.101.93.93), 30 hops max, 60 byte packets
 1  192.168.1.1 (192.168.1.1)  9.931 ms  9.837 ms  9.770 ms
 2  100.93.40.1 (100.93.40.1)  24.780 ms  24.693 ms  24.630 ms
 3  10.168.113.77 (10.168.113.77)  24.837 ms  24.783 ms 10.168.113.73 (10.168.113.73)  24.723 ms
 4  10.185.173.53 (10.185.173.53)  24.680 ms  24.566 ms  31.364 ms
 5  171-102-253-41.static.asianet.co.th (171.102.253.41)  25.727 ms  26.557 ms  26.420 ms
 6  119-46-49-16.static.asianet.co.th (119.46.49.16)  25.723 ms 119-46-49-56.static.asianet.co.th (119.46.49.56)  23.197 ms 171-102-249-120.static.asianet.co.th (171.102.249.120)  23.789 ms
 7  171-102-250-1.static.asianet.co.th (171.102.250.1)  23.634 ms  24.726 ms 171-102-254-65.static.asianet.co.th (171.102.254.65)  26.554 ms
 8  171-102-254-232.static.asianet.co.th (171.102.254.232)  25.256 ms  25.147 ms  24.901 ms
 9  171-102-251-244.static.asianet.co.th (171.102.251.244)  25.605 ms  27.381 ms  25.425 ms
10  TIG-Net28-8.trueintergateway.com (122.144.28.8)  25.475 ms  27.645 ms TIG-Net28-10.trueintergateway.com (122.144.28.10)  27.964 ms
11  TIG-Net245-53.trueintergateway.com (113.21.245.53)  49.996 ms  49.817 ms TIG-Net245-241.trueintergateway.com (113.21.245.241)  49.168 ms
12  ae-17.edge3.Singapore3.Level3.net (4.68.71.9)  51.617 ms  51.992 ms 113.21.245.41 (113.21.245.41)  57.169 ms
13  4.69.218.90 (4.69.218.90)  77.303 ms 4.69.218.94 (4.69.218.94)  77.012 ms 4.69.218.90 (4.69.218.90)  76.876 ms
14  8.245.6.70 (8.245.6.70)  55.648 ms  57.665 ms  57.404 ms
15  sin-eqx-02t.voxility.net (103.31.184.6)  57.288 ms  56.250 ms  56.641 ms
16  fra-eqx-02t.voxility.net (93.115.91.27)  235.296 ms  229.219 ms  237.112 ms
17  fra-eq5-02c.voxility.net (5.254.61.221)  227.418 ms  233.662 ms  226.523 ms
18  fra-eq5-04sw.voxility.net (5.254.68.150)  228.178 ms  227.841 ms  227.637 ms
19  5.254.14.94 (5.254.14.94)  224.979 ms  224.847 ms  226.552 ms
20  lag6.core2-dus3.bb.as24961.net (62.141.47.86)  229.342 ms lag6.core3-dus1.bb.as24961.net (62.141.47.20)  230.224 ms  230.383 ms
21  ae0-1014.r01-a1-7.dus.onecorp.eu (89.163.136.201)  231.976 ms ae0-1013.r01-a1-7.dus.onecorp.eu (89.163.136.203)  232.722 ms  232.233 ms
22  * * *
23  * * *
24  * * *
25  * * *
26  * * *
27  * * *
28  * * *
29  * * *
30  * * *
```
From above traceroute log, my address is in Thailand, the local ISP works properly as it can transfer packets to Singapore. However, the packets were dropped somewhere in Europe, interesting.

MichaIng, 2023-02-17:
oneCorp GmbH is the company behind our VPS provider. So this means it is dropped just within our providers network. Strange as we have no firewall enabled, only a DDoS protection on plain HTTP requests (which cannot be disabled, and are redirected to HTTPS anyway).

I'll inform our VPS provider.

MichaIng, 2023-02-19:
@kullachartp 
You you test and paste a current traceroute again? I'm in contact with our VPS provider. I did a route from our server back to `171-102-253-41.static.asianet.co.th`, the last public host onto your location, and that was successful, but also went through China instead of through Singapore for some reason. Best would be if you could send me your public IP, so I can better test the back route, but better send it via mail to me (see my GitHub profile).

kullachartp, 2023-02-20:
Many thanks for your kind support. I have tested today and it works fine now as follows:
```
# traceroute ssh.dietpi.com
traceroute to ssh.dietpi.com (185.101.93.93), 30 hops max, 60 byte packets
 1  192.168.1.1 (192.168.1.1)  1.846 ms  1.820 ms  1.817 ms
 2  100.93.48.1 (100.93.48.1)  19.029 ms  19.635 ms  19.588 ms
 3  10.168.113.85 (10.168.113.85)  20.752 ms 10.168.113.81 (10.168.113.81)  20.842 ms  20.013 ms
 4  10.185.173.77 (10.185.173.77)  20.668 ms  20.938 ms  20.745 ms
 5  103-3-176-109.static.asianet.co.th (103.3.176.109)  22.106 ms  22.299 ms 171-102-253-41.static.asianet.co.th (171.102.253.41)  22.362 ms
 6  171-102-253-80.static.asianet.co.th (171.102.253.80)  23.918 ms 119-46-49-16.static.asianet.co.th (119.46.49.16)  21.798 ms 171-102-249-136.static.asianet.co.th (171.102.249.136)  22.048 ms
 7  171-102-254-65.static.asianet.co.th (171.102.254.65)  22.736 ms  23.104 ms 171-102-250-1.static.asianet.co.th (171.102.250.1)  22.039 ms
 8  171-102-254-232.static.asianet.co.th (171.102.254.232)  23.729 ms  24.063 ms  23.928 ms
 9  171-102-251-244.static.asianet.co.th (171.102.251.244)  24.027 ms  23.884 ms  28.587 ms
10  TIG-Net28-8.trueintergateway.com (122.144.28.8)  26.193 ms TIG-Net28-6.trueintergateway.com (122.144.28.6)  25.841 ms TIG-Net28-10.trueintergateway.com (122.144.28.10)  30.358 ms
11  TIG-Net245-247.trueintergateway.com (113.21.245.247)  49.638 ms TIG-Net245-53.trueintergateway.com (113.21.245.53)  52.713 ms TIG-Net245-247.trueintergateway.com (113.21.245.247)  43.769 ms
12  ae-17.edge3.Singapore3.Level3.net (4.68.71.9)  57.626 ms TIG-Net245-41.trueintergateway.com (113.21.245.41)  45.981 ms  45.012 ms
13  4.69.218.94 (4.69.218.94)  54.031 ms  53.815 ms 4.69.218.90 (4.69.218.90)  53.665 ms
14  8.245.6.70 (8.245.6.70)  48.999 ms  49.576 ms  49.011 ms
15  sin-eqx-02t.voxility.net (103.31.184.6)  49.152 ms  49.006 ms  52.226 ms
16  fra-eqx-02t.voxility.net (93.115.91.27)  225.931 ms  231.489 ms  230.461 ms
17  fra-eq5-02c.voxility.net (5.254.61.221)  213.901 ms  219.151 ms  218.611 ms
18  fra-eq5-04sw.voxility.net (5.254.68.150)  230.159 ms  227.505 ms  227.316 ms
19  5.254.14.94 (5.254.14.94)  217.575 ms  217.310 ms  216.852 ms
20  lag6.core2-dus3.bb.as24961.net (62.141.47.86)  226.948 ms lag6.core3-dus1.bb.as24961.net (62.141.47.20)  219.172 ms  226.768 ms
21  ae0-1014.r01-a1-7.dus.onecorp.eu (89.163.136.201)  225.902 ms ae0-1013.r01-a1-7.dus.onecorp.eu (89.163.136.203)  220.687 ms  226.133 ms
22  185.101.93.93 (185.101.93.93)  226.420 ms  218.476 ms  218.035 ms
```

However,  the Raspberry Pi remains in trouble but with different error message.
```
# > "$G_HW_UUID.txt"
curl --key '' -m 20 -T "$G_HW_UUID.txt" 'sftp://dietpi-survey:upload2dietpi@ssh.dietpi.com:29248/survey/'
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (79) Error in the SSH layer
```

I have another Diet Pi in the same network, so I upgraded to the same version (8.14.2) and tested. Surprisingly, it works properly.
```
# traceroute ssh.dietpi.com
traceroute to ssh.dietpi.com (185.101.93.93), 30 hops max, 60 byte packets
 1  192.168.1.1 (192.168.1.1)  10.825 ms  10.599 ms  10.489 ms
 2  100.93.48.1 (100.93.48.1)  50.590 ms  50.520 ms  50.452 ms
 3  10.168.113.85 (10.168.113.85)  50.179 ms 10.168.113.81 (10.168.113.81)  50.087 ms  50.527 ms
 4  10.185.173.77 (10.185.173.77)  51.065 ms  60.671 ms  50.647 ms
 5  103-3-176-109.static.asianet.co.th (103.3.176.109)  51.566 ms  51.417 ms 171-102-253-41.static.asianet.co.th (171.102.253.41)  55.331 ms
 6  119-46-49-88.static.asianet.co.th (119.46.49.88)  52.763 ms 171-102-249-136.static.asianet.co.th (171.102.249.136)  47.782 ms 119-46-49-16.static.asianet.co.th (119.46.49.16)  47.430 ms
 7  171-102-254-65.static.asianet.co.th (171.102.254.65)  47.711 ms 171-102-250-1.static.asianet.co.th (171.102.250.1)  26.616 ms  26.378 ms
 8  171-102-254-232.static.asianet.co.th (171.102.254.232)  26.290 ms *  26.024 ms
 9  171-102-251-244.static.asianet.co.th (171.102.251.244)  25.915 ms  30.807 ms  31.305 ms
10  TIG-Net28-4.trueintergateway.com (122.144.28.4)  31.175 ms TIG-Net28-6.trueintergateway.com (122.144.28.6)  30.520 ms TIG-Net28-4.trueintergateway.com (122.144.28.4)  30.896 ms
11  TIG-Net241-175.trueintergateway.com (113.21.241.175)  53.172 ms TIG-Net245-241.trueintergateway.com (113.21.245.241)  54.869 ms TIG-Net245-247.trueintergateway.com (113.21.245.247)  52.913 ms
12  ae-17.edge3.Singapore3.Level3.net (4.68.71.9)  54.781 ms  54.426 ms TIG-Net245-41.trueintergateway.com (113.21.245.41)  51.152 ms
13  4.69.218.94 (4.69.218.94)  49.407 ms 4.69.218.90 (4.69.218.90)  50.651 ms  53.639 ms
14  8.245.6.70 (8.245.6.70)  54.191 ms  53.283 ms  53.073 ms
15  sin-eqx-02t.voxility.net (103.31.184.6)  52.960 ms  53.467 ms  53.395 ms
16  fra-eqx-02t.voxility.net (93.115.91.27)  233.734 ms  241.262 ms  233.973 ms
17  fra-eq5-02c.voxility.net (5.254.61.221)  232.065 ms  224.400 ms  218.803 ms
18  fra-eq5-04sw.voxility.net (5.254.68.150)  236.258 ms  235.979 ms  242.183 ms
19  5.254.14.94 (5.254.14.94)  226.264 ms  219.402 ms  223.865 ms
20  lag6.core2-dus3.bb.as24961.net (62.141.47.86)  234.124 ms  234.307 ms lag6.core3-dus1.bb.as24961.net (62.141.47.20)  223.646 ms
[... cut ...]
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:03 --:--:--     0
  0     0    0     0    0     0      0      0 --:--:--  0:00:03 --:--:--     0
#
```

One of the difference between my both Raspberry Pi is the the one that has the problem use OpenSSH Server, while the others that works use the default Dropbear. I am not sure this relates to the problem (just my guess from the newest log).

MichaIng, 2023-02-20:
> curl: (79) Error in the SSH layer

Can you try to reinstall curl, and use verbose flag:
```sh
apt install --reinstall curl libcurl4 libssh2-1
curl --key '' -v -T "$G_HW_UUID.txt" 'sftp://dietpi-survey:upload2dietpi@ssh.dietpi.com:29248/survey/'
```

kullachartp, 2023-02-20:
After reinstalling `curl` using your guidance, the problem still persists
```
# curl --key '' -v -T "$G_HW_UUID.txt" 'sftp://dietpi-survey:upload2dietpi@ssh.dietpi.com:29248/survey/'
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0*   Trying 185.101.93.93:29248...
*   Trying 2a06:1c40:3::142:29248...
* Immediate connect fail for 2a06:1c40:3::142: Cannot assign requested address
* Connected to ssh.dietpi.com (185.101.93.93) port 29248 (#0)
* Found host ssh.dietpi.com in /root/.ssh/known_hosts
* Set "ecdsa-sha2-nistp256" as SSH hostkey type
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
* Closing connection 0
curl: (79) Error in the SSH layer
```

MichaIng, 2023-02-20:
Can you try again along with with a
```sh
date -u
```
I want to check whether I see a related failure in our logs. Also if you could send me your public IP (via email) that would be great. I want to rule out that it is not that a fail2ban block our end. Although `Connected to ssh.dietpi.com` looks like a success.

kullachartp, 2023-02-20:
```
# date -u
Mon Feb 20 21:31:39 UTC 2023
```
I also replied my internet public IP address to you in GitHub notified mail.

MichaIng, 2023-02-20:
I answered your email, many thanks.

Does it actually work to connect via HTTP to that server?
```sh
curl -v ssh.dietpi.com
```

kullachartp, 2023-02-21:
> I answered your email, many thanks.
> 
> Does it actually work to connect via HTTP to that server?
> 
> ```shell
> curl -v ssh.dietpi.com
> ```
```
# curl -v ssh.dietpi.com
*   Trying 185.101.93.93:80...
* Connected to ssh.dietpi.com (185.101.93.93) port 80 (#0)
> GET / HTTP/1.1
> Host: ssh.dietpi.com
> User-Agent: curl/7.74.0
> Accept: */*
>
* Mark bundle as not supporting multiuse
< HTTP/1.1 301 Moved Permanently
< Date: Tue, 21 Feb 2023 14:37:45 GMT
< Server: Apache
< X-Content-Type-Options: nosniff
< X-Frame-Options: SAMEORIGIN
< X-XSS-Protection: 1; mode=block
< X-Robots-Tag: all
< X-Permitted-Cross-Domain-Policies: none
< Referrer-Policy: no-referrer-when-downgrade
< Content-Security-Policy: upgrade-insecure-requests; frame-ancestors 'self'; base-uri 'self' https://dietpi.com/matomo/index.php https://dietpi.com/grafana/; default-src 'none'; object-src 'none'; style-src 'self' 'unsafe-inline'; font-src 'self' data:; img-src * data:; script-src 'self' 'unsafe-inline' 'unsafe-eval' https://unpkg.com; frame-src 'self'; manifest-src 'self'; connect-src 'self' https://api.github.com
< Permissions-Policy: accelerometer=(), autoplay=(), browsing-topics=(), camera=(), encrypted-media=(), fullscreen=(), geolocation=(), gyroscope=(), magnetometer=(), microphone=(), midi=(), payment=(), usb=(), screen-wake-lock=()
< Cache-Control: public, max-age=86399
< Location: https://dietpi.com/
< Content-Length: 227
< Content-Type: text/html; charset=iso-8859-1
<
<!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
<html><head>
<title>301 Moved Permanently</title>
</head><body>
<h1>Moved Permanently</h1>
<p>The document has moved <a href="https://dietpi.com/">here</a>.</p>
</body></html>
* Connection #0 to host ssh.dietpi.com left intact
```

Sorry for misunderstanding about `date -u`. Really funny :)

MichaIng, 2023-02-21:
Okay, so HTTP works, SSH does still not. And indeed I see
```
error: kex_exchange_identification: Connection closed by remote host
```
on our server, so your `curl` closed the connection before even trying to verify our host key. Based on your earlier debug output, it chose the correct host key algorithm `ecdsa-sha2-nistp256`...
.... ah wait, it actually selects the wrong host key type. We have an ECDSA key on our server, but on DietPi systems until Bullseye, the RSA key is stored in known hosts, as tests showed that libssh on Bullseye does not support ECDSA, which contradicts the selection of your curl. Can you show the following:
```sh
which curl
dpkg -l curl libcurl4 libssh2-1 libgcrypt20
```

kullachartp, 2023-02-21:
> Okay, so HTTP works, SSH does still not. And indeed I see
> 
> ```
> error: kex_exchange_identification: Connection closed by remote host
> ```
> 
> on our server, so your `curl` closed the connection before even trying to verify our host key. Based on your earlier debug output, it chose the correct host key algorithm `ecdsa-sha2-nistp256`... .... ah wait, it actually selects the wrong host key type. We have an ECDSA key on our server, but on DietPi systems until Bullseye, the RSA key is stored in known hosts, as tests showed that libssh on Bullseye does not support ECDSA, which contradicts the selection of your curl. Can you show the following:
> 
> ```shell
> which curl
> dpkg -l curl libcurl4 libssh2-1 libgcrypt20
> ```
```
# which curl
/usr/bin/curl
# dpkg -l curl libcurl4 libssh2-1 libgcrypt20
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name              Version            Architecture Description
+++-=================-==================-============-============================================================
==
ii  curl              7.74.0-1.3+deb11u5 arm64        command line tool for transferring data with URL syntax
ii  libcurl4:arm64    7.74.0-1.3+deb11u5 arm64        easy-to-use client-side URL transfer library (OpenSSL flavou
r)
ii  libgcrypt20:arm64 1.8.7-6            arm64        LGPL Crypto library - runtime library
ii  libssh2-1:arm64   1.9.0-2            arm64        SSH2 client-side library
```

MichaIng, 2023-02-21:
Okay libssh2 v1.9.0 does support ECDSA keys:
- https://github.com/libssh2/libssh2/commit/aba34f5
- https://github.com/libssh2/libssh2/commit/62b825c

There was some bug about it in curl, but this has been fixed as well:
- https://github.com/libssh2/libssh2/commit/62b825c

So generally it should work on Bullseye, shown by your earlier debug log, and I can replicate. But still, using ECDSA pubkey authentication with curl on Bullseye fails on that very early stage, for unknown reasons.

However, since our host is added with RSA key to known hosts, curl should pick RSA as authentication algorithm. In my case if that key is missing, it fails the same way as in your case. Can you check:
```sh
grep ssh.dietpi.com /root/.ssh/known_hosts
```
If this does not return any output, you can fix the issue with:
```sh
G_CONFIG_INJECT '\[?ssh.dietpi.com(]:29248)?[[:blank:]]' '[ssh.dietpi.com]:29248 ssh-rsa AAAAB3NzaC1yc2EAAAADAQABAAABAQDE6aw3r6aOEqendNu376iiCHr9tGBIWPgfrLkzjXjEsHGyVSUFNnZt6pftrDeK7UX+qX4FxOwQlugG4fymOHbimRCFiv6cf7VpYg1Ednquq9TLb7/cIIbX8a6AuRmX4fjdGuqwmBq3OG7ZksFcYEFKt5U4mAJIaL8hXiM2iXjgY02LqiQY/QWATsHI4ie9ZOnwrQE+Rr6mASN1BVFuIgyHIbwX54jsFSnZ/7CdBMkuAd9B8JkxppWVYpYIFHE9oWNfjh/epdK8yv9Oo6r0w5Rb+4qaAc5g+RAaknHeV6Gp75d2lxBdCm5XknKKbGma2+/DfoE8WZTSgzXrYcRlStYN' /root/.ssh/known_hosts
```
Now this should succeed:
```sh
dietpi-survey 1
```

kullachartp, 2023-02-21:
> Okay libssh2 v1.9.0 does support ECDSA keys:
> 
> * [libssh2/libssh2@aba34f5](https://github.com/libssh2/libssh2/commit/aba34f5)
> * [libssh2/libssh2@62b825c](https://github.com/libssh2/libssh2/commit/62b825c)
> 
> There was some bug about it in curl, but this has been fixed as well:
> 
> * [libssh2/libssh2@62b825c](https://github.com/libssh2/libssh2/commit/62b825c)
> 
> So generally it should work on Bullseye, shown by your earlier debug log, and I can replicate. But still, using ECDSA pubkey authentication with curl on Bullseye fails on that very early stage, for unknown reasons.
> 
> However, since our host is added with RSA key to known hosts, curl should pick RSA as authentication algorithm. In my case if that key is missing, it fails the same way as in your case. Can you check:
> 
> ```shell
> grep ssh.dietpi.com /root/.ssh/known_hosts
> ```
> 
> If this does not return any output, you can fix the issue with:
> 
> ```shell
> G_CONFIG_INJECT '\[?ssh.dietpi.com(]:29248)?[[:blank:]]' '[ssh.dietpi.com]:29248 ssh-rsa AAAAB3NzaC1yc2EAAAADAQABAAABAQDE6aw3r6aOEqendNu376iiCHr9tGBIWPgfrLkzjXjEsHGyVSUFNnZt6pftrDeK7UX+qX4FxOwQlugG4fymOHbimRCFiv6cf7VpYg1Ednquq9TLb7/cIIbX8a6AuRmX4fjdGuqwmBq3OG7ZksFcYEFKt5U4mAJIaL8hXiM2iXjgY02LqiQY/QWATsHI4ie9ZOnwrQE+Rr6mASN1BVFuIgyHIbwX54jsFSnZ/7CdBMkuAd9B8JkxppWVYpYIFHE9oWNfjh/epdK8yv9Oo6r0w5Rb+4qaAc5g+RAaknHeV6Gp75d2lxBdCm5XknKKbGma2+/DfoE8WZTSgzXrYcRlStYN' /root/.ssh/known_hosts
> ```
> 
> Now this should succeed:
> 
> ```shell
> dietpi-survey 1
> ```

Unfortunately, the key has been existed in known_hosts file, yet dietpi-survey remains error.
```
# grep ssh.dietpi.com /root/.ssh/known_hosts
[ssh.dietpi.com]:29248 ssh-rsa AAAAB3NzaC1yc2EAAAADAQABAAABAQDE6aw3r6aOEqendNu376iiCHr9tGBIWPgfrLkzjXjEsHGyVSUFNnZt6pftrDeK7UX+qX4FxOwQlugG4fymOHbimRCFiv6cf7VpYg1Ednquq9TLb7/cIIbX8a6AuRmX4fjdGuqwmBq3OG7ZksFcYEFKt5U4mAJIaL8hXiM2iXjgY02LqiQY/QWATsHI4ie9ZOnwrQE+Rr6mASN1BVFuIgyHIbwX54jsFSnZ/7CdBMkuAd9B8JkxppWVYpYIFHE9oWNfjh/epdK8yv9Oo6r0w5Rb+4qaAc5g+RAaknHeV6Gp75d2lxBdCm5XknKKbGma2+/DfoE8WZTSgzXrYcRlStYN
# dietpi-survey 1
[FAILED] DietPi-Survey | Sending survey data
 - Command: curl --key  -m 20 -sT fa04ea05-ff17-4732-8c32-7fb2c569fef7.txt sftp://dietpi-survey:upload2dietpi@ssh.dietpi.com:29248/survey/
```

kullachartp, 2023-02-21:
> Hmm, running the `C_CONFIG_INJECT` regardless does not change anything (e.g. typo inside)?
> 
> Mysterious why does curl/libssh in 99,9% of cases pick RSA and the defined key without issues and in your particular case it just picks ECDSA... Is there any other content in that file?
> 
> ```shell
> cat /root/.ssh/known_hosts
> ```

I have already sent my known_hosts to you via email.

MichaIng, 2023-02-21:
I just tested it with a freshly created ECDSA host key on our server. The size was doubled, so the current key is just 128 bits. However, with 256 bits, trying to connect from Bullseye fails the same way. When building it with 364 bits, the algorithm in the public key is called `ecdsa-sha2-nistp386`, but strangely with our 128 bits key it is still called `ecdsa-sha2-nistp128`. I'm confused. However, those keys were created in 2018 and Bullseye's curl isn't able to work with neither old 128 bits nor new 256 bits key...

Found it:
- In your `/root/.ssh/known_hosts` are hashed trusted host key entries, generated by `ssh-keyscan -H <host>`.
- The last two of them are from our server, as the key matches our ECDSA key. Since ECDSA is newer, they are preferred, overriding and that way breaking the intended RSA host key entry. So remove the last two lines and try again.
- If it still fails, probably the first 3 lines are as well from our server, causing the same issue. I cannot know this from the hashes since those are salted. The public key however does not match.

kullachartp, 2023-02-21:
I just removed all entries in known_hosts file, and add a new one with your suggestion `C_CONFIG_INJECT`. Eventually, the dietpi-survey works correctly.

```
# G_CONFIG_INJECT '\[?ssh.dietpi.com(]:29248)?[[:blank:]]' '[ssh.dietpi.com]:29248 ssh-rsa AAAAB3NzaC1yc2EAAAADAQABAAABAQDE6aw3r6aOEqendNu376iiCHr9tGBIWPgfrLkzjXjEsHGyVSUFNnZt6pftrDeK7UX+qX4FxOwQlugG4fymOHbimRCFiv6cf7VpYg1Ednquq9TLb7/cIIbX8a6AuRmX4fjdGuqwmBq3OG7ZksFcYEFKt5U4mAJIaL8hXiM2iXjgY02LqiQY/QWATsHI4ie9ZOnwrQE+Rr6mASN1BVFuIgyHIbwX54jsFSnZ/7CdBMkuAd9B8JkxppWVYpYIFHE9oWNfjh/epdK8yv9Oo6r0w5Rb+4qaAc5g+RAaknHeV6Gp75d2lxBdCm5XknKKbGma2+/DfoE8WZTSgzXrYcRlStYN' /root/.ssh/known_hosts
[  OK  ] G_CONFIG_INJECT | Added setting [ssh.dietpi.com]:29248 ssh-rsa AAAAB3NzaC1yc2EAAAADAQABAAABAQDE6aw3r6aOEqendNu376iiCHr9tGBIWPgfrLkzjXjEsHGyVSUFNnZt6pftrDeK7UX+qX4FxOwQlugG4fymOHbimRCFiv6cf7VpYg1Ednquq9TLb7/cIIbX8a6AuRmX4fjdGuqwmBq3OG7ZksFcYEFKt5U4mAJIaL8hXiM2iXjgY02LqiQY/QWATsHI4ie9ZOnwrQE+Rr6mASN1BVFuIgyHIbwX54jsFSnZ/7CdBMkuAd9B8JkxppWVYpYIFHE9oWNfjh/epdK8yv9Oo6r0w5Rb+4qaAc5g+RAaknHeV6Gp75d2lxBdCm5XknKKbGma2+/DfoE8WZTSgzXrYcRlStYN to end of file /root/.ssh/known_hosts

# cat /root/.ssh/known_hosts
[ssh.dietpi.com]:29248 ssh-rsa AAAAB3NzaC1yc2EAAAADAQABAAABAQDE6aw3r6aOEqendNu376iiCHr9tGBIWPgfrLkzjXjEsHGyVSUFNnZt6pftrDeK7UX+qX4FxOwQlugG4fymOHbimRCFiv6cf7VpYg1Ednquq9TLb7/cIIbX8a6AuRmX4fjdGuqwmBq3OG7ZksFcYEFKt5U4mAJIaL8hXiM2iXjgY02LqiQY/QWATsHI4ie9ZOnwrQE+Rr6mASN1BVFuIgyHIbwX54jsFSnZ/7CdBMkuAd9B8JkxppWVYpYIFHE9oWNfjh/epdK8yv9Oo6r0w5Rb+4qaAc5g+RAaknHeV6Gp75d2lxBdCm5XknKKbGma2+/DfoE8WZTSgzXrYcRlStYN
# dietpi-survey 1
[  OK  ] DietPi-Survey | Sending survey data
```

I have SSH connections to another Raspberry Pis, which use some of the line I removed from the known_hosts file. I will try to  reconfig them again.

MichaIng, 2023-02-21:
Finally. The question remains how those hashed entries went into your `known_hosts` file 😄. I'm thinking about removing the ECDSA key from our server to prevent such incidence.

> another Raspberry Pis, which use some of the line

When you use the `ssh` command, you are offered to accept the host key, which is then added to the file automatically. Otherwise with `ssh-keyscan`. E.g. to add all host keys of that host again in hashed from:
```sh
ssh-keyscan -H your.other.pi >> /root/.ssh/known_hosts
```

kullachartp, 2023-02-21:
Thank you for your kind help along the way. It is such a great experience discussing with you and follow your expert guidance.

MichaIng, 2023-02-21:
Great, I'll close this issue now. Feel free to reopen if the issue reappears.

I checked and thought again about the `100.64.0.0/10` IP range for ISPs. I don't know exactly how this is organised, but possibly a lot of people in your area share the same global public IP, or a small number of public IPs of your ISP, at least as seen from our servers. In such case, DDoS/rate limiting filters and firewall blocks may kick in quicker and affect a lot of other users which are not involved into any attack, or none is, but its just a large number of users naturally possibly doing a large number of concurrent requests to the same host, which then looks like a DDoS attack, but isn't. So I know now to check for `171.*` IPs among our blocks next time.

closed 2023-02-21 by MichaIng