MichaIng/DietPi #6171: Abillity to choose OpenSSH when making own distro from Debian

mews-se, 2023-02-16:
#### Is your feature request related to a problem? Please describe:
- Not a problem per se, but when "converting" a debian install to dietpi - there is no ability to preselect OpenSSH, which means that is has to be done manually every time.

#### Describe the solution you'd like:

- A choice of which version of ssh is preferred, as with which release of debian

Joulinar, 2023-02-16:
Once completed to DietPi install script, you should be able to configure `/boot/dietpi.txt` and select OpenSSH before rebooting system first time.

mews-se, 2023-02-16:
True, I will give it a try next time. Although - automatition is nice ;)

Joulinar, 2023-02-16:
within `/boot/dietpi.txt` you can configure a E2E fully automated installation.

mews-se, 2023-02-16:
Tried that, but still had to choose "install" from config, also none of the prechosen software entries was installed and just a bunch of different errors

Joulinar, 2023-02-16:
did you set the flag for automatic install? and what kind of errors?

mews-se, 2023-02-16:
For SSH after I "converted" the debian installation, I changed dietpi.txt to:

`AUTO_SETUP_SSH_SERVER_INDEX=-2`

However the only thing that changed was that OpenSSH was "pre-selected" in the menu, not installed.

For the non-installs I have:

`AUTO_SETUP_AUTOMATED=1`
and just tested the following to try git and pivpn:
```
AUTO_SETUP_INSTALL_SOFTWARE_ID=105
AUTO_SETUP_INSTALL_SOFTWARE_ID=17
```

which seems to have been ignored.

A few other changes such as passwords and so fort just resulted in a "failed first run" that reran interactively with no change.

Since it does quite a bunch during the conversion, perhaps it would have worked for a fresh sd-install, but not when building from scratch?

Joulinar, 2023-02-18:
> AUTO_SETUP_INSTALL_SOFTWARE_ID=105
AUTO_SETUP_INSTALL_SOFTWARE_ID=17

But ID of PiVPN would be 117 and not 105 (OpenSSH). There is no need to select OpenSSH again. It should switch automatically using `AUTO_SETUP_SSH_SERVER_INDEX=-2`

MichaIng, 2023-02-18:
I won't add an option to select the SSH server in `dietpi-installer`. So far there is a clear distinction between things required for a fresh system to boot with all capabilities to be fully and automatically set, and the actual setup options done in `dietpi.txt` or `dietpi-software` during first run setup. If we lift this, we just introduce a lot of doubled code, inconsistencies, trigger further requests like about choosing the log level, a desktop environment, a default password and what not.

But instead, a larger task, I aim to allow pre-installing software generally in `dietpi-installer` via `dietpi-software` directly, hence a generic approach for all kind of such ideas. So basically one would be able to run `dietpi-software` to install/uninstall anything near the end of the `dietpi-installer` run. Afterwards traces/logs of this will be removed, so one faces a common first run setup on reboot, but with previously selected software titles preinstalled. This also allows us to offer desktop images and to align the Allo GUI images better with regular ones.

closed 2023-02-24 by mews-se