MichaIng/DietPi #6172: Error when apt update, server down how do I change

jonasradke, 2023-02-17:
#### Details:
- Date           | Fri Feb 17 12:32:01 CET 2023
- DietPi version | v8.10.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | `Linux DietPi 5.15.61-v7+ #1579 SMP Fri Aug 26 11:10:59 BST 2022 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `apt-get -y update`
- Exit code      | 255
- Software title | DietPi-Update

#### Extra details:

I can't find where this entry is. How do I change it
It's not in any of the repository lists that are usually used
- ...
#### Additional logs:
```
Hit:1 http://debian.obspm.fr/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye InRelease
Hit:3 https://deb.debian.org/debian bullseye-updates InRelease
Hit:4 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:5 https://deb.debian.org/debian bullseye-backports InRelease
Hit:7 http://deb.volian.org/volian scar InRelease
Hit:6 http://debian.uvigo.es/debian bullseye InRelease
Hit:8 https://archive.raspberrypi.org/debian bullseye InRelease
Err:9 http://debian.uni-duisburg-essen.de/debian bullseye InRelease
  Could not connect to debian.uni-duisburg-essen.de:80 (132.252.181.48), connection timed out
Reading package lists...
W: Failed to fetch http://debian.uni-duisburg-essen.de/debian/dists/bullseye/InRelease  Could not connect to debian.uni-duisburg-essen.de:80 (132.252.181.48), connection timed out
W: Some index files failed to download. They have been ignored, or old ones used instead.
```

Joulinar, 2023-02-17:
looks like you added a 3rd party repository yourself. 

```
Err:9 http://debian.uni-duisburg-essen.de/debian bullseye InRelease
  Could not connect to debian.uni-duisburg-essen.de:80 (132.252.181.48), connection timed out
```

If not needed you should remove the configuration

jonasradke, 2023-02-17:
Yeah idk where it comes from I don't think I added it 
How would I about removing it 
It's not in the normal sources file

Joulinar, 2023-02-17:
can you create a list and check as follow

```
for i in /etc/apt/sources.list{,.d/*.list}; do echo "$i:"; cat "$i"; done
```

jonasradke, 2023-02-17:
Bingo that fixed it thanks!

closed 2023-02-17 by jonasradke